ap--/python-seabreeze #86: using `list_devices()` closes currently opened devices

jonathanvanschenck, 2020-05-30:
## System Info

- model: USB2000

- operating system: Debian 10

- python version: 3.7.6

- python-seabreeze version: 1.0.1

- installed-via: pip

## Caveat
I'm not sure if this is a `python-seabreeze` issue/bug/feature or an underlying `seabreeze` issue/bug/feature.

## The Issue
I am using the `seabreeze.cseabreeze` backend and I am running into some anomalous behavior with the `list_devices()` function. The main issue seems to be that if I already have an opened `seabreeze.cseabreeze.SeaBreezeDevice` object, and then interact with the `SeaBreezeAPI().list_devices()` list, my original device gets closed. Here's some example code:
```python
from seabreeze.cseabreeze import SeaBreezeAPI as API
api = API()
api.list_devices() # returns [<SeaBreezeDevice USB2000:USB2E7922>]
dev = api.list_devices()[0]
dev.is_open # returns False
dev.open()
dev.is_open # returns True
api.list_devices() # only call function, returns [<SeaBreezeDevice USB2000:USB2E7922>]
dev.is_open # returns True
API().list_devices()[0] # interact with list, returns <SeaBreezeDevice USB2000:USB2E7922>
dev.is_open # returns False ?
```
(Note, I get the same behavior if I substitute `API()` for `api` in the above code).

Curiously, the problem is worse (well, consistent) for the front end:
```python
import seabreeze.spectrometers as sbs
dev = sbs.list_devices()[0]
dev.open()
dev.is_open # returns True
sbs.list_devices() # simply call list, returns [<SeaBreezeDevice USB2000:USB2E7922>]
dev.is_open # returns False?
```

Ideally, I would love to be able to check the current list of available devices without closing all the devices that are currently open.

Thanks in advance for the help!

jonathanvanschenck, 2020-05-30:
Further investigation reveals that the [line 232 in `src/seabreeze/cseabreeze/c_seabreeze_wrapper.pyx`](https://github.com/ap--/python-seabreeze/blob/2756e094b4277e54d454cef748bef2de1278b770/src/seabreeze/cseabreeze/c_seabreeze_wrapper.pyx#L232) all previously opened devices are closed when `.list_devices()` is called. So, this seems like this is partially a feature? Strange then that it is only when the list is interacted with that the device actually gets closed.

Anyway, I'm going to put together a pull request where I add an optional function argument to `.list_devices()` which will allow listed devices to remain open. Give me a few minutes.

ap--, 2020-06-01:
Hi @jonathanvanschenck,

I'll try to reproduce it here and will have a look.

Just in general: which usecase are you trying to support?
Spectrometer hot plugging?

Cheers,
Andreas

jonathanvanschenck, 2020-06-01:
Andreas,
     Thanks for the response! Yes, hot plugging is my eventual goal. I'm developing a GUI that will run on top of seabreeze, and the hope would be to have (potentially multiple) devices connected and removed with the GUI continually running.
-J

mhagdorn, 2020-06-02:
I'd be interested in that too. Hot-plugging is on my todo list for our piccolo system: https://bitbucket.org/teampiccolo/workspace/projects/PP3

ap--, 2020-06-02:
Okay, so I have a reproducible test case :smile: :tada:.
Will work on this whenever I have a bit of time in the coming days.

Currently it works for none of the backends.
In case of `cseabreeze`, which you are using, it's a problem with garbage collection of the temporary `SeaBreezeDevices` that `list_devices` creates.

If you change your code to:

```python
import seabreeze.spectrometers as sbs
dev = sbs.list_devices()[0]
dev.open()
keep_references_list = sbs.list_devices()
dev.is_open  # returns True
```
The device stays open.
But as soon as you delete the references, it closes the original device:
```python
del keep_references_list
dev.is_open  # returns False
```

Cheers :blush:,
Andreas

ap--, 2020-06-02:
> I'd be interested in that too. Hot-plugging is on my todo list for our piccolo system: https://bitbucket.org/teampiccolo/workspace/projects/PP3

@mhagdorn very cool project. let me know if you want it to be added to the list of projects that use `python-seabreeze` :smiley:

mhagdorn, 2020-06-02:
yes, that would be nice. we really need to sort out a website for the project where we will obviously mention python-seabreeze as well. Your project has been very helpful. Somehow creating a website has been very low on my list of things todo.

Very much higher up on my TODO list is to figure out how to reset the current measurement on the QEPros.

ap--, 2020-06-03:
> Very much higher up on my TODO list is to figure out how to reset the current measurement on the QEPros.

It's best to open a new issue for that. If you have the developer manual, I can help.

closed 2020-06-09 by ap--

jonathanvanschenck, 2020-06-18:
I was able to build the current development version on my system and everything is seeming to work! Thanks!

commit 4957f175 (2020-06-04): [psb] keep instance registry of seabreeze devices

pyseabreeze fix for #86.
list_devices should now behave correctly when SeaBreezeDevices
are already opened.

Close #86.
commit fd319b04 (2020-06-04): [csb] keep instance registry of seabreeze devices

list_devices() does not longer created temporary SeaBreezeDevices
which fixes #86 for the cseabreeze backend.