ap--/python-seabreeze #87: spec.intensities() does not work  (pyseabreeze with FLAME-T / USB4000)

np-8, 2020-06-02:
### spectrometer and system information ###

- model: FLAME-T (recognized as USB4000)
- operating system: Win10 64bit
- python version: 3.7.7 32-bit
- python-seabreeze version: 1.1.0
- installed-via: pip

### current problem ###

_detailed description of what doesn't work_

### steps to reproduce ###

Getting spectrum with pyseabreeze backend does not work. The reading of the wavelengths _does_ work. Also reading spectrum (intensities) with cseabreeze backend works.

What happens after running the example code is that the program "hangs". I suppose it just does not give Timeout exceptions since the timeout is 653 seconds and I never waited so long. 

### minimal code example and error (very helpful if available) ###

```python
import seabreeze
seabreeze.use('pyseabreeze')

from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
print(spec.wavelengths())
print(spec.intensities())
```
### Additional details

If played around with the code and found out the place where it actually does hang and what is the problem. In spectrometer.py:

- First, the program goes to `SeaBreezeSpectrometerFeatureOOI.get_intensities()` and calls the `tmp = self._get_spectrum_raw()` there. 

- Then, the program goes to `SeaBreezeSpectrometerFeatureOOIFPGA4K._get_spectrum_raw()` and after sending `self.protocol.send(0x09)` it checks `self.protocol.transport._default_read_endpoint` which is `"low_speed"`. Then it hangs while waiting 
```
            tmp[:] = self.protocol.receive(size=self._spectrum_raw_length,
                                           timeout_ms=timeout)
```
to answer.

- Running the contents of `SeaBreezeSpectrometerFeatureOOI._get_spectrum_raw()` will result into:

```python
      1 tmp[:] = self.protocol.receive(size=self._spectrum_raw_length,
      2                                timeout_ms=5000, # my temporary hard coded value
----> 3                                mode="high_speed")

ValueError: could not broadcast input array from shape (5633) into shape (7681)
```
- Running the following (modified `SeaBreezeSpectrometerFeatureOOIFPGA4K._get_spectrum_raw()`)
```
        tmp = numpy.zeros((self._spectrum_raw_length, ), dtype=numpy.uint8)
        timeout = int(self._integration_time_max * 1e-3 +
                      self.protocol.transport.default_timeout_ms)
        self.protocol.send(0x09)
            tmp[:2048] = self.protocol.receive(size=2048,
                                               timeout_ms=timeout,
                                               mode="high_speed_alt")
            tmp[2048:] = self.protocol.receive(
                size=self._spectrum_raw_length - 2048,
                timeout_ms=timeout,
                mode="high_speed",
            )
```
 will result into

```
In [10]: len(tmp)
Out[10]: 7681

In [11]: tmp
Out[11]: array([ 49,  24,  73, ...,  83,  24, 105], dtype=uint8)
```
note that I put initially the array to be all zeroes, so it filled the whole array.  I don't understand why the `self._spectrum_raw_length` is 7681, since it has only 3648 (or 3840) pixels. Maybe there is an explanation for that, too.

ap--, 2020-06-03:
Hi @np-8 

Thanks for the very detailed bug report! :blush: 
To narrow down why it's not working I would try the following:

First, try using the `cseabreeze` backend. If that works we know that it's an issue with the pyseabreeze implementation:
```python
import seabreeze
# seabreeze.use('pyseabreeze') <-- just don't call this. cseabreeze is the default backend
# you can also call
# seabreeze.use('cseabreeze', force=True)
# to make sure that cseabreeze is used

from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
print(spec.wavelengths())
print(spec.intensities())
```

If that doesn't work either, we have to figure out if the Flame-T changed the spectrum acquisition communication protocol. If you could search online and try to find a Developer Manual for the Flame-T spectrometer that describes the low level USB communication layer, we can work out what's going on. (It might be hard to find one. If you can't find anything, I'd try contacting OceanOptics.)

_Some extra info:_
The raw spectrum length is often something like i.e. `num_pixels * (sizeof uint16) + num_bytes_metadata + 1` for a spectrometer with 16bit resolution. A raw spectrum in the pyseabreeze code is a bytestring that binary encodes a spectrum.

Let me know how things go,
Cheers :smile: 
-Andreas

ap--, 2020-06-03:
And one more thing:
The USB4000 needs to access two different usb endpoints to read one spectrum,
the fact that you call returned `5633` bytes looks like we might just get it to work, because:
The first `2048` bytes need to be read from somewhere else and then `2048 + 5633 = 7681`

Please post a complete and reproducible script that get to the 5633 error.

Cheers,
-Andreas

np-8, 2020-06-03:
Sorry, one might notice that I was writing the "Additional details" with a time limit 😅.

> First, try using the cseabreeze backend. If that works we know that it's an issue with the pyseabreeze implementation:

I did this already (see the "current problem" section) in the original post. To recap:

- cseabreeze backend: Everything works
- pyseabreeze backend: `spec.wavelengths()` works, `spec.intensities()` does not.

### Complete and reproducible script 
Writing a complete copy-pasteable script is a bit difficult, since I was trying different solutions using `IPython.embed()` (having a breakpoint in the pyseabreeze library code). But, I can write again the contents of the original post in more copyable form.

 #### Example 1: "Original code"
- This is the original code, i.e. identical as calling `spec.intensities()`. One difference: Timeout changed (from 650 sec)
```
import numpy
import seabreeze
seabreeze.use('pyseabreeze')

from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
s = spec.features['spectrometer'][0]

# from SeaBreezeSpectrometerFeatureOOIFPGA4K._get_spectrum_raw
tmp = numpy.empty((s._spectrum_raw_length, ), dtype=numpy.uint8)
timeout = 2000  # ms. Not needed more than that. Integration time is 100ms.

s.protocol.send(0x09)
if s.protocol.transport._default_read_endpoint == "low_speed":
    tmp[:] = s.protocol.receive(size=s._spectrum_raw_length,
                                timeout_ms=timeout)
else:  # high_speed
    tmp[:2048] = s.protocol.receive(size=2048,
                                    timeout_ms=timeout,
                                    mode="high_speed_alt")
    tmp[2048:] = s.protocol.receive(
        size=s._spectrum_raw_length - 2048,
        timeout_ms=timeout,
        mode="high_speed",
    )
```
This will output long exception chain with `usb.core.USBError: [Errno 10060] Operation timed out`.

#### Example 2:  "script that gets to the 5633 error"
```python
import numpy
import seabreeze
seabreeze.use('pyseabreeze')

from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
s = spec.features['spectrometer'][0]

# from SeaBreezeSpectrometerFeatureOOIFPGA4K._get_spectrum_raw
tmp = numpy.empty((s._spectrum_raw_length, ), dtype=numpy.uint8)
timeout = 2000  # ms. Not needed more than that. Integration time is 100ms.

s.protocol.send(0x09)

tmp[:] = s.protocol.receive(size=s._spectrum_raw_length,
                            timeout_ms=timeout,
                            mode='high_speed')
```
which will output the "desired"
```
Traceback (most recent call last):
  File ".\tests\test_spectrometer_raw.py", line 21, in <module>
    mode='high_speed')
ValueError: could not broadcast input array from shape (5633) into shape (7681)
```

#### Example 3: Code that seems to work
- Note that the `tmp` is now initiated with all zeroes (to verify that it is actually filled with real data)
```python
import numpy
import seabreeze
seabreeze.use('pyseabreeze')

from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
s = spec.features['spectrometer'][0]

# from SeaBreezeSpectrometerFeatureOOIFPGA4K._get_spectrum_raw
tmp = numpy.zeros((s._spectrum_raw_length, ), dtype=numpy.uint8)
timeout = 2000  # ms. Not needed more than that. Integration time is 100ms.

s.protocol.send(0x09)

tmp[:2048] = s.protocol.receive(size=2048,
                                timeout_ms=timeout,
                                mode="high_speed_alt")
tmp[2048:] = s.protocol.receive(
    size=s._spectrum_raw_length - 2048,
    timeout_ms=timeout,
    mode="high_speed",
)
```
this outputs
```
len(tmp) 7681 tmp: [211   5 203 ... 235   5 105]
```

#### Example 4: Code that also works
- This is the same thing as `Example 3`, just shorter

```python
import numpy
import seabreeze
seabreeze.use('pyseabreeze')

from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
s = spec.features['spectrometer'][0]
s.protocol.transport._default_read_endpoint = 'high_speed'
print(spec.intensities())
```
this outputs
```
[2983.89046875 3041.23359375 3014.61       ... 3090.38484375 3090.38484375
 3090.38484375]
```

### Guesses
- The `_default_read_endpoint` for USB4000 / FLAME-T should be set to `'high_speed'`
- The `integration_time_max` should be set to 65 seconds (to get timeout faster). See also: https://github.com/ap--/python-seabreeze/issues/89

ap--, 2020-06-03:
:see_no_evil: oh I missed the `cseabreeze` part, sorry :sweat_smile: 

So your Example 4 is actually what should already happen internally. (the `__init__` method of the SpectrometerFeature class is overridden for the FPGA spectrometers and should handle this)
Let's try and figure out why it currently doesn't!
Awesome bug hunting btw :+1: :1st_place_medal:
And thanks for the thorough description!

```python
In [1]: seabreeze.use('pyseabreeze')                                                                     
In [2]: from seabreeze.spectrometers import Spectrometer                                                 
In [3]: spec = Spectrometer.from_first_available()                                                       
In [4]: spec.f.spectrometer.protocol.send(0xFE)                                                          
Out[4]: 1
In [5]: spec.f.spectrometer.protocol.receive(size=16)                                                    
Out[5]: array('B', [0, 8, 112, 23, 0, 0, 0, 0, 0, 8, 0, 0, 0, 0, 128, 85])
```

What's the return value of line 5 when you run the code above?

Cheers,
Andreas

np-8, 2020-06-04:
Sure let's hunt the bug down! 🐱‍👤🧐

```python
In [1]: seabreeze.use('pyseabreeze')
   ...: from seabreeze.spectrometers import Spectrometer
   ...: spec = Spectrometer.from_first_available()
   ...: spec.f.spectrometer.protocol.send(0xFE)
Out[1]: 1

In [2]: spec.f.spectrometer.protocol.receive(size=16)
Out[2]: array('B', [0, 15, 160, 134, 1, 0, 0, 0, 0, 15, 0, 15, 0, 0, 128, 255])
```

ap--, 2020-06-09:
Okay, so it turns out that the USB full speed vs USB high speed behavior for all FPGA based OOI spectrometers is implemented incorrectly in `pyseabreeze`.
Interestingly, it's implemented in a way that allows it to work for all but the `USB4000` and `HR4000` when connected to USB2.0 and higher USB ports. And my guess is that there's probably very very few host PCs with USB1.0 ports anymore which caused this bug to never come up.

I'll try to push a fix to a separate branch soon, and would then ask you to install `seabreeze` from that branch and run the tests.
When all of them pass the fix will be added to the next release.

Cheers,
Andreas

ap--, 2020-06-09:
@np-8 please run all tests on the develop branch

```
py.test ./tests
```

If there's any questions let me know :smiley:

np-8, 2020-06-09:
I don't know what version the USB ports are, but there are also usb 3.0 ports listed. I was using a USB 2.0 hub, but the problem is the same when connecting the spectrometer directly to an USB port.

I installed the develop branch and ran `py.test ./tests`. First observations: missing `pytest` and `mock` ( maybe could be added to requirements-dev.txt?). Also missing the dll for pyusb, since got:

```
________________________________________________________________ ERROR collecting tests/test_spectrometers.py ________________________________________________________________
tests\test_spectrometers.py:115: in <module>
    @pytest.fixture(scope="function", **_retr())  # request all spectrometers on module load
tests\test_spectrometers.py:48: in _retr
    api = backend.SeaBreezeAPI()
src\seabreeze\pyseabreeze\api.py:56: in __init__
    self.initialize()
src\seabreeze\pyseabreeze\api.py:64: in initialize
    USBTransport.initialize(**self._kwargs)
src\seabreeze\pyseabreeze\transport.py:314: in initialize
    for device in cls.list_devices(**_kwargs):
src\seabreeze\pyseabreeze\transport.py:272: in list_devices
    backend=get_pyusb_backend_from_name(name=_pyusb_backend),
venv\lib\site-packages\usb\core.py:1263: in find
    raise NoBackendError('No backend available')
E   usb.core.NoBackendError: No backend available
```

After copying `libusb-1.0.dll` to the `/venv/Scripts/` I started to see some output.  Here it is:

```
(venv) PS C:\external-git-repos\python-seabreeze> python -m pytest .\tests\
============================================================================ test session starts =============================================================================
platform win32 -- Python 3.7.3, pytest-5.4.3, py-1.8.1, pluggy-0.13.1 -- C:\external-git-repos\python-seabreeze\venv\Scripts\python.exe
cachedir: .pytest_cache
rootdir: C:\external-git-repos\python-seabreeze, inifile: pytest.ini
collected 38 items                                                                                                                                                            

tests/test_backends.py::test_backend_features_interface PASSED                                                                                                          [  2%]
tests/test_backends.py::test_cseabreeze_seabreezeapi PASSED                                                                                                             [  5%]
tests/test_backends.py::test_pyseabreeze_seabreezeapi PASSED                                                                                                            [  7%]
tests/test_protocol.py::test_obp_protocol_messages PASSED                                                                                                               [ 10%]
tests/test_spectrometers.py::test_cant_find_serial[pyseabreeze(openusb)] SKIPPED                                                                                        [ 13%]
tests/test_spectrometers.py::test_cant_find_serial[pyseabreeze(libusb0)] SKIPPED                                                                                        [ 15%]
tests/test_spectrometers.py::test_cant_find_serial[pyseabreeze(libusb1)] PASSED                                                                                         [ 18%]
tests/test_spectrometers.py::test_cant_find_serial[cseabreeze] PASSED                                                                                                   [ 21%]
tests/test_spectrometers.py::test_device_cleanup_on_exit[pyseabreeze(openusb)] SKIPPED                                                                                  [ 23%]
tests/test_spectrometers.py::test_device_cleanup_on_exit[pyseabreeze(libusb0)] SKIPPED                                                                                  [ 26%]
tests/test_spectrometers.py::test_device_cleanup_on_exit[pyseabreeze(libusb1)] FAILED                                                                                   [ 28%]
tests/test_spectrometers.py::test_device_cleanup_on_exit[cseabreeze] PASSED                                                                                             [ 31%]
tests/test_spectrometers.py::test_read_model[cseabreeze:USB4000:FLMT02803] PASSED                                                                                       [ 34%]
tests/test_spectrometers.py::test_read_model[pyseabreeze:USB4000:FLMT02803] PASSED                                                                                      [ 36%]
tests/test_spectrometers.py::test_read_serial_number[cseabreeze:USB4000:FLMT02803] PASSED                                                                               [ 39%]
tests/test_spectrometers.py::test_read_serial_number[pyseabreeze:USB4000:FLMT02803] PASSED                                                                              [ 42%]
tests/test_spectrometers.py::test_crash_may_not_influence_following_tests[cseabreeze:USB4000:FLMT02803] XFAIL                                                           [ 44%]
tests/test_spectrometers.py::test_crash_may_not_influence_following_tests[pyseabreeze:USB4000:FLMT02803] XFAIL                                                          [ 47%]
tests/test_spectrometers.py::test_read_intensities[cseabreeze:USB4000:FLMT02803] SKIPPED                                                                                [ 50%]
[... cut ...]
FAILED tests/test_spectrometers.py::test_read_wavelengths[pyseabreeze:USB4000:FLMT02803] - usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
FAILED tests/test_spectrometers.py::test_read_spectrum[pyseabreeze:USB4000:FLMT02803] - usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
=========================================================== 5 failed, 21 passed, 10 skipped, 2 xfailed in 18.03s ============================================================
```
These tests I ran with Python 3.7.3 32-bit.

ap--, 2020-06-10:
Okay, so I just tried running the test on windows10 with `libusb0` and `libusb1` and I need to refactor the tests to run for each backend independently. Getting dynamic backend switching for tests to work on windows (which is the way it's done currently) is going to be a nightmare :disappointed: and I don't want to spend my free time on it. So basically, for now on windows ignore the tests.

Anyways, regarding the actual issue: In your current virtualenv can you run:
```python
import seabreeze
seabreeze.use('pyseabreeze')
from seabreeze.spectrometers import Spectrometer

spec = Spectrometer.from_first_available()
spec.intensities()
```

and report back if it works? If it does that means we can close this issue.

Cheers,
Andreas :smiley:

np-8, 2020-06-10:
Hi,

I ran the following code:
```python
import seabreeze
seabreeze.use('pyseabreeze')
print(seabreeze.__version__)

from seabreeze.spectrometers import Spectrometer

spec = Spectrometer.from_first_available()
print(spec.intensities())
```

With `libusb-1.0.dll` version `1.0.23` I get errors, which _I assume are related to libusb (or pyusb)_

<details>
<summary>Output with `libusb-1.0.dll` version `1.0.23`</summary>

```
1.1.1.dev53+g964f0f1
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x000001D3459CAEA0>
Traceback (most recent call last):
  File "C:\external-git-repos\python-seabreeze\venv\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\external-git-repos\python-seabreeze\venv\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\Python\Python37\lib\weakref.py", line 552, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\external-git-repos\python-seabreeze\venv\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\external-git-repos\python-seabreeze\venv\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\external-git-repos\python-seabreeze\venv\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
[3039.185625   3092.4328125  3092.4328125  ... 2959.31484375 2959.31484375
 2959.31484375]
```
</details>

With `libusb-1.0.dll` version `1.0.21` (MinGW64)  all good:
<details>
<summary>Output with `libusb-1.0.dll` version `1.0.21`</summary>

```
1.1.1.dev53+g964f0f1
[3035.0896875  3020.75390625 2969.5546875  ... 2961.3628125  2961.3628125
 2961.3628125 ]
```
</details>

So I think this issue can be closed! Thank you! 🎉

ap--, 2020-06-10:
Awesome :+1: :tada: 
Happy that it works :smile: 

It'd be great if I can ping you when I have refactored the tests.
Your bug reports are very informative, so it'll be easier to track down remaining issues :sparkling_heart: 
I'll release version 1.2.0 once that is done.

Cheers,
Andreas

commit 964f0f15 (2020-06-09): [psb] fix USB full speed / high speed handling

This should fix spectrum acquisition for USB4000 and HR4000.
Note that in general USB1.0 full speed mode is likely to be broken
for all FPGA OOI spectrometers. This adds a warning in case someone
runs into the issue. Implementing full speed mode for those
spectrometers requires someone with the hardware to test this.

The warning is emitted in case a spectrometer is forced to run in
full speed mode and recommends reporting it to upstream. My guess
is that there are not many USB1.0 hosts out there anymore, but
who knows, maybe there's some old lab [... cut]
closed 2020-06-12 by ap--