ap--/python-seabreeze #91: Refactor tests to run for each backend independently

ap--, 2020-06-10:
The way this is currently done in the pytests is a bit esoteric.
No one is going to try to use both backends from the same process, and if they do, there's some serious development effort to be spent on cross-platform support for windows.

Proposed workaround:
add pytest parameters to allow running tests for each backend individually
i.e.
```
pytest -m cseabreeze ./tests
pytest -m pyseabreeze_libusb1 ./tests
etc...
```

ap--, 2020-06-12:
Hi @np-8 

I finished refactoring the test suite.
It's now easier to test specific backends for seabreeze.
could you run the pytests on the newest develop branch?

It would be great if you could run:
```sh
# without a spectrometer attached
pytest

# with one or more spectrometers attached
pytest --seabreeze-backend cseabreeze

# with one or more spectrometers attached
pytest --seabreeze-backend pyseabreeze
```
and post the output here :smiley: 

(fyi, possible options for `--seabreeze-backend` are `any`, `cseabreeze`, `pyseabreeze`, `pyseabreeze:libusb0`, `pyseabreeze:libusb1`, `pyseabreeze:openusb` with `any` being the default)

Thanks!
-Andreas

np-8, 2020-06-18:
Hi,

I pulled the latest development branch. Here are the test results

<details>
<summary>
`pytest` (with no spectrometer attached)
</summary>

```python

============================= test session starts =============================
platform win32 -- Python 3.7.3, pytest-5.4.3, py-1.8.2, pluggy-0.13.1 -- c:\external-git-repos\python-seabreeze\venv\scripts\python.exe
cachedir: .pytest_cache
rootdir: C:\external-git-repos\python-seabreeze, inifile: pytest.ini
collecting ... collected 28 items

tests/test_backends.py::test_seabreeze_installed PASSED                  [  3%]
tests/test_backends.py::test_seabreeze_wrong_backend_requested PASSED    [  7%]
tests/test_backends.py::test_seabreeze_any_backend_available[cseabreeze] PASSED [ 10%]
tests/test_backends.py::test_seabreeze_any_backend_available[pyseabreeze] PASSED [ 14%]
tests/test_backends.py::test_seabreeze_cseabreeze_backend_available PASSED [ 17%]
tests/test_backends.py::test_seabreeze_pyseabreeze_backend_available PASSED [ 21%]
tests/test_backends.py::test_seabreeze_cseabreeze_api_init PASSED        [ 25%]
tests/test_backends.py::test_seabreeze_pyseabreeze_api_init[any] PASSED  [ 28%]
tests/test_backends.py::test_seabreeze_pyseabreeze_api_init[openusb] SKIPPED [ 32%]
tests/test_backends.py::test_seabreeze_pyseabreeze_api_init[libusb0] SKIPPED [ 35%]
tests/test_backends.py::test_seabreeze_pyseabreeze_api_init[libusb1] SKIPPED [ 39%]
tests/test_backends.py::test_seabreeze_compare_backend_feature_interfaces PASSED [ 42%]
tests/test_protocol.py::test_pyseabreeze_protocol_messages PASSED        [ 46%]
tests/test_spectrometers.py::TestHardware::test_cant_find_serial[backend(any)] FAILED [ 50%]
tests/test_spectrometers.py::TestHardware::test_device_cleanup_on_exit[backend(any)] SKIPPED [ 53%]
tests/test_spectrometers.py::TestHardware::test_read_model[no-spectrometer-backend(any)] SKIPPED [ 57%]
tests/test_spectrometers.py::TestHardware::test_read_serial_number[no-spectrometer-backend(any)] SKIPPED [ 60%]
tests/test_spectrometers.py::TestHardware::test_crash_may_not_influence_following_tests[no-spectrometer-backend(any)] SKIPPED [ 64%]
tests/test_spectrometers.py::TestHardware::test_read_intensities[no-spectrometer-backend(any)] SKIPPED [ 67%]
tests/test_spectrometers.py::TestHardware::test_correct_dark_pixels[no-spectrometer-backend(any)] SKIPPED [ 71%]
tests/test_spectrometers.py::TestHardware::test_read_wavelengths[no-spectrometer-backend(any)] SKIPPED [ 75%]
tests/test_spectrometers.py::TestHardware::test_read_spectrum[no-spectrometer-backend(any)] SKIPPED [ 78%]
tests/test_spectrometers.py::TestHardware::test_max_intensity[no-spectrometer-backend(any)] SKIPPED [ 82%]
tests/test_spectrometers.py::TestHardware::test_integration_time_limits[no-spectrometer-backend(any)] SKIPPED [ 85%]
tests/test_spectrometers.py::TestHardware::test_integration_time[no-spectrometer-backend(any)] SKIPPED [ 89%]
tests/test_spectrometers.py::TestHardware::test_trigger_mode[no-spectrometer-backend(any)] SKIPPED [ 92%]
tests/test_spectrometers.py::TestHardware::test_trigger_mode_wrong[no-spectrometer-backend(any)] SKIPPED [ 96%]
tests/test_spectrometers.py::TestHardware::test_list_devices_dont_close_opened_devices[backend(any)] SKIPPED [100%]

================================== FAILURES ===================================
______________ TestHardware.test_cant_find_serial[backend(any)] _______________

cls = <class 'seabreeze.pyseabreeze.transport.USBTransport'>, kwargs = {}
_pyusb_backend = None

    @classmethod
    def list_devices(cls, **kwargs):
        """list pyusb devices for all available spectrometers
    
        Note: this includes spectrometers that are currently opened in other
        processes on the machine.
    
        Yields
        ------
        devices : USBTransportHandle
            unique pyusb devices for each available spectrometer
        """
        # check if a specific pyusb backend is requested
        _pyusb_backend = kwargs.get("_pyusb_backend", None)
        # get all matching devices
        try:
            pyusb_devices = usb.core.find(
                find_all=True,
[... 2764 lines cut ...]
                custom_match=lambda dev: (
                    dev.idVendor == cls.vendor_id and dev.idProduct in cls.product_ids
                ),
                backend=get_pyusb_backend_from_name(name=_pyusb_backend),
            )
        except usb.core.NoBackendError:
>           raise RuntimeError("No pyusb backend found")
E           RuntimeError: No pyusb backend found

src\seabreeze\pyseabreeze\transport.py:283: RuntimeError
============================== warnings summary ===============================
tests\test_spectrometers.py:62
  C:\external-git-repos\python-seabreeze\tests\test_spectrometers.py:62: PytestUnknownMarkWarning: Unknown pytest.mark.cseabreeze - is this a typo?  You can register custom marks to avoid this warning - for details, see https://docs.pytest.org/en/latest/mark.html
    marks.append(pytest.mark.cseabreeze)

-- Docs: https://docs.pytest.org/en/latest/warnings.html
=========================== short test summary info ===========================
SKIPPED [1] C:\external-git-repos\python-seabreeze\tests\test_backends.py:61: pyusb_backend openusb not available
SKIPPED [1] C:\external-git-repos\python-seabreeze\tests\test_backends.py:61: pyusb_backend libusb0 not available
SKIPPED [1] C:\external-git-repos\python-seabreeze\tests\test_backends.py:61: pyusb_backend libusb1 not available
SKIPPED [1] tests\test_spectrometers.py:178: FIXME: TEST BREAKS OTHER TESTS ON WINDOWS
============= 14 failed, 10 passed, 4 skipped, 1 warning in 3.04s =============

```
</details>

ap--, 2020-06-25:
Thanks for the test output :+1: :sparkling_heart: 

Hello ~darkness~ windows my old friend... :sweat_smile: 

Okay, so it looks like you are missing a `libusb` DLL. can you copy the one that worked for you in the virtual env that you use to run the tests, and post the output again?

### Notes for me:
- [ ] The tests should output some additional info to deal with common environment setup issues.
- [ ] The failing `cseabreeze` test suite is a real issue: `Windows fatal exception: access violation` Need to see if I can reproduce this on my machine
- [ ] the pyseabreeze test suite is failing because of a missing DLL. This should be made obvious in the test output
- [ ] maybe: many windows env issues could be circumvented by making `seabreeze_os_setup` download the correct DLLs

np-8, 2020-06-29:
Hi,

You're welcome! I used `libusb-1.0.dll` (v.1.0.21 64-bit) and python-seabreeze version with hash `a547848`. Here's the output:

<details>
<summary>
python -m pytest
</summary>

```python
======================================================================== test session starts ========================================================================
platform win32 -- Python 3.7.3, pytest-5.4.3, py-1.8.2, pluggy-0.13.1 -- C:\external-git-repos\python-seabreeze\venv\Scripts\python.exe
cachedir: .pytest_cache
rootdir: C:\external-git-repos\python-seabreeze, inifile: pytest.ini
collected 28 items                                                                                                                                                   

tests/test_backends.py::test_seabreeze_installed PASSED                                                                                                        [  3%]
tests/test_backends.py::test_seabreeze_wrong_backend_requested PASSED                                                                                          [  7%]
tests/test_backends.py::test_seabreeze_any_backend_available[cseabreeze] PASSED                                                                                [ 10%]
tests/test_backends.py::test_seabreeze_any_backend_available[pyseabreeze] PASSED                                                                               [ 14%]
tests/test_backends.py::test_seabreeze_cseabreeze_backend_available PASSED                                                                                     [ 17%]
tests/test_backends.py::test_seabreeze_pyseabreeze_backend_available PASSED                                                                                    [ 21%]
tests/test_backends.py::test_seabreeze_cseabreeze_api_init PASSED                                                                                              [ 25%]
tests/test_backends.py::test_seabreeze_pyseabreeze_api_init[any] PASSED                                                                                        [ 28%]
tests/test_backends.py::test_seabreeze_pyseabreeze_api_init[openusb] SKIPPED                                                                                   [ 32%]
tests/test_backends.py::test_seabreeze_pyseabreeze_api_init[libusb0] SKIPPED                                                                                   [ 35%]
tests/test_backends.py::test_seabreeze_pyseabreeze_api_init[libusb1] PASSED                                                                                    [ 39%]
tests/test_backends.py::test_seabreeze_compare_backend_feature_interfaces PASSED                                                                               [ 42%]
tests/test_protocol.py::test_pyseabreeze_protocol_messages PASSED                                                                                              [ 46%]
tests/test_spectrometers.py::TestHardware::test_cant_find_serial[backend(any)] PASSED                                                                          [ 50%]
tests/test_spectrometers.py::TestHardware::test_device_cleanup_on_exit[backend(any)] SKIPPED                                                                   [ 53%]
tests/test_spectrometers.py::TestHardware::test_read_model[no-spectrometer-backend(any)] SKIPPED                                                               [ 57%]
tests/test_spectrometers.py::TestHardware::test_read_serial_number[no-spectrometer-backend(any)] SKIPPED                                                       [ 60%]
tests/test_spectrometers.py::TestHardware::test_crash_may_not_influence_following_tests[no-spectrometer-backend(any)] SKIPPED                                  [ 64%]
tests/test_spectrometers.py::TestHardware::test_read_intensities[no-spectrometer-backend(any)] SKIPPED                                                         [ 67%]
tests/test_spectrometers.py::TestHardware::test_correct_dark_pixels[no-spectrometer-backend(any)] SKIPPED                                                      [ 71%]
tests/test_spectrometers.py::TestHardware::test_read_wavelengths[no-spectrometer-backend(any)] SKIPPED                                                         [ 75%]
tests/test_spectrometers.py::TestHardware::test_read_spectrum[no-spectrometer-backend(any)] SKIPPED                                                            [ 78%]
tests/test_spectrometers.py::TestHardware::test_max_intensity[no-spectrometer-backend(any)] SKIPPED                                                            [ 82%]
tests/test_spectrometers.py::TestHardware::test_integration_time_limits[no-spectrometer-backend(any)] SKIPPED                                                  [ 85%]
tests/test_spectrometers.py::TestHardware::test_integration_time[no-spectrometer-backend(any)] SKIPPED                                                         [ 89%]
tests/test_spectrometers.py::TestHardware::test_trigger_mode[no-spectrometer-backend(any)] SKIPPED                                                             [ 92%]
tests/test_spectrometers.py::TestHardware::test_trigger_mode_wrong[no-spectrometer-backend(any)] SKIPPED                                                       [ 96%]
tests/test_spectrometers.py::TestHardware::test_list_devices_dont_close_opened_devices[backend(any)] SKIPPED                                                   [100%]

[... cut ...]  or 64 bit dll based on python version and the just try some simple function with all of the dll's, and the first dll with which everything works is the "correct one". :D  I would think that a helpful error message saying "Probably missing libusb-x.x.dll" would be ok, too. And maybe some link where it can be downloaded and instructions where to put it (in the Scripts-folder). There could be some simple test to perform of the dll's are found correctly or not ("try to connect to a spectrometer"?).

ap--, 2020-08-10:
With version v1.2.0 the test have been completely refactored.

I'll close this issue in favour of a more specific issue.

ap--, 2020-08-10:
Thanks again for you help @np-8

closed 2020-08-10 by ap--