ap--/python-seabreeze #92: Flame-S spectrometer is seen but doesn't report data

waldowda, 2020-06-24:
### spectrometer and system information ###

- model: Flame-S supposed to be the same as USB-2000PLUS I believe 
- operating system: MacOS 15.5.5
- python version: 3.7.7 (via Anaconda)
- python-seabreeze version: v1.1.0 I believe
- installed-via: conda  and python for anaconda 

### current problem ###

The python code only appears to communicate and give the name / serial number.  No data appears to be return for  .wavelengths() or  .intensities(). I could easily have made a mistake somewhere in the install perhaps.

### minimal code example and error (very helpful if available) ###

Here is example code:

1. run code example

import seabreeze
seabreeze.use('pyseabreeze')
from seabreeze.spectrometers import list_devices, Spectrometer

devices = list_devices()

print (devices)
 
spec = Spectrometer.from_serial_number("FLMS16723")

spec.open()

spec.wavelengths()
data = spec.intensities(correct_dark_counts=True, correct_nonlinearity=True)

print ("Dim: ", data.ndim)
print ("Shape: ", data.shape)
print ("size: ", data.size)

data[0]
data
spec.close()

2. Output:
[<SeaBreezeDevice USB2000PLUS:FLMS16723>]
Dim:  1
Shape:  (2048,)
size:  2048

ap--, 2020-06-24:
Hi @waldowda 

It looks to me like everything is working.
`spec.wavelengths()` as well as `spec.intensities()` return a numpy array.

I think you're just looking at the print output of your program and you're assuming it doesn't return data.

Try:
```python
import seabreeze
seabreeze.use('pyseabreeze')
from seabreeze.spectrometers import list_devices, Spectrometer

spec = Spectrometer.from_serial_number("FLMS16723")

wavelengths = spec.wavelengths()
print('Wavelengths', wavelengths)
data = spec.intensities(correct_dark_counts=True, correct_nonlinearity=True)
print('Intensities', data)
```

P.S.: Don't call the `Spectrometer.open` and `Spectrometer.close` methods yourself, unless you now what you're doing. Under almost all normal use cases you do not have to interact with them.

Let me know if this solves your issue, or if you need further help.
Also please close this issue if it's resolved.

Cheers, :smiley: 
Andreas

waldowda, 2020-06-24:
Hi Andreas, Thanks for the reply! The code is now sort of working. I'll explain in a moment. I was blindly following the QuickStart.html and that simply enters the numpy array on a single line and I expected similar output. I also may have been confusing panda vs numpy in this regard. So, the basic code works now but 1) not every execution and 2) it 'sometimes' errors when setting int time, and always with boxcar and scans to average. I would not be surprised if again I was doing something silly.  See below for the output of the above code. Apologies for the length. Thanks! Dean

**1) First example with a sequence of times running the simple code from above: First one is ok, then two errors and finally the it runs again. I removed the open()/close() calls that I had added those when I confused by the issues.**

runcell(0, '/Users/waldow/.spyder-py3/untitled2.py')
Wavelengths [ 339.8952      340.27820458  340.66116785 ... 1026.47833653 1026.76169767
 1027.04500266]
Intensities [1.81250494e+01 3.01859751e+04 2.21902390e+03 ... 2.71098953e+03
 2.71098953e+03 2.71098953e+03]

runcell(0, '/Users/waldow/.spyder-py3/untitled2.py')
Traceback (most recent call last):

  File "/Users/waldow/.spyder-py3/untitled2.py", line 13, in <module>
    spec = Spectrometer.from_serial_number("FLMS16723")

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/spectrometers.py", line 114, in from_serial_number
    for dev in list_devices():

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/spectrometers.py", line 35, in list_devices
    return list_devices._api.list_devices()

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/api.py", line 98, in list_devices
    dev.open()

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/devices.py", line 326, in open
    self._transport.open_device(self._raw_device)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/transport.py", line 142, in open_device
    self._protocol = self._protocol_cls(self)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/protocol.py", line 66, in __init__
    self.send(0x01)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/protocol.py", line 93, in send
    return self.transport.write(data, timeout_ms=timeout_ms)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/transport.py", line 160, in write
    return self._device.write(self._endpoint_map.ep_out, data, timeout=timeout_ms)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 940, in write
    intf, ep = self._ctx.setup_request(self, endpoint)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])

USBError: [Errno 13] Access denied (insufficient permissions)

runcell(0, '/Users/waldow/.spyder-py3/untitled2.py')
Wavelengths [ 339.8952      340.27820458  340.66116785 ... 1026.47833653 1026.76169767
 1027.04500266]
Traceback (most recent call last):

  File "/Users/waldow/.spyder-py3/untitled2.py", line 17, in <module>
    data = spec.intensities()
[... 104 lines cut ...]
[<SeaBreezeDevice USB2000PLUS:FLMS16723>]
Serial Number:  FLMS16723
Max Intensity:  65535.0
Traceback (most recent call last):

  File "/Users/waldow/.spyder-py3/flame_test.py", line 28, in <module>
    spec.boxcar_width(4)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/compat.py", line 42, in boxcar_width
    self.f.spectrum_processing.set_boxcar_width(boxcar_width)

AttributeError: 'NoneType' object has no attribute 'set_boxcar_width'`

**Snippet from scans_to_average:**

Added this code to the end:

`
  File "/Users/waldow/.spyder-py3/flame_test.py", line 64, in <module>
    ave_spec = spec.scans_to_average(4)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/compat.py", line 48, in scans_to_average
    self.f.spectrum_processing.set_scans_to_average(self, scans_to_average)

AttributeError: 'NoneType' object has no attribute 'set_scans_to_average'`

ap--, 2020-06-25:
Hi Dean,

Glad I could help! Let's try to sort out the remaining issues:

> I was blindly following the QuickStart.html and that simply enters the numpy array on a single line and I expected similar output.

Could you provide the url of the page you looked at? And if you think that the documentation is confusing, I would be grateful if you could suggest changes to make it more accessible.

----

> I also may have been confusing panda vs numpy in this regard. So, the basic code works now but 1) not every execution and 

I am not entirely sure how you are using the the module yet. From the info you provided so far I assume you're using Spyder and you're executing the same cell over and over again.
I suggest you put the code that imports seabreeze and opens the spectrometer in one cell and execute that only once.
you can then execute the cell that calls intensities() over and over again and should not run into issues.

The `USBError: [Errno 13] Access denied (insufficient permissions)` error definitely needs some investigation. I might be able to test more OSX compatibility soon. Maybe something regarding USB permissions changed with OSX 10.15 and newer...

I would suggest the following. Try using:
```python
seabreeze.use('cseabreeze')
```
instead of `seabreeze.use('pyseabreeze')`

and run your code again. Maybe that fixes it.

----

> 2) it 'sometimes' errors when setting int time, and always with boxcar and scans to average. I would not be surprised if again I was doing something silly. See below for the output of the above code. Apologies for the length. Thanks! Dean

Boxcar and scans to average is not supported by this spectometer

----

> 2) Here is another example were I follow the QuickStart.html for boxcar and its error. Sometimes I get similar behavior for int. time.

Again, please post the link to the documentation you are referring to. Also the code you posted in this issue never accesses `spec.f.spectrum_processing` (which is not supported by your spectrometer and therefore returns `None`)
To summarize: Don't use `Spectrometer.scans_to_average()` or `Spectrometer.boxcar_width`

Let me know how things go.

Cheers, :smiley: 
Andreas

##### P.S. Please use triple backticks around your codeblocks in the issues to make it more readable:

\`\`\`python
a = 2
\`\`\`

becomes:

```python
a = 2
```

waldowda, 2020-06-25:
Hi Andreas, Thanks for the detailed reply and the tip about triple back-ticks and for correcting the previous posts.  I was just going to do that. That is so much more readable!  The quickstart url is: [https://python-seabreeze.readthedocs.io/en/latest/quickstart.html](url) and under the acquire spectrum heading. 

```
# get intensities
>>> intensities = spec.intensities()
>>> intensities
array([1485., 1493., 1487., ..., 1496., 1497., 1493.])
```

So, I simply followed the above exactly. There might be a python 2.7 and 3.x difference as I am quite new to Python and am using 3.7 in spyder or jupyter as part of anaconda. 

I now see that the behavior in spyder vs jupyter is different by testing a simple numpy array. In jupyter, just having the array on a single line displays the array but not in spyder - at least for me. That is likely what initially confused me. I thought they would behave the same. Perhaps changing that line and similar ones above to use the `print ("",)` format (python 3.x I guess) when then work for both or stating in a cell in jupyter would help some like me. But, I suspect a more seasoned python / jupyter / anaconda user would not have an issue with the formatting as it is now.
 
I have mostly been running from spyder where I execute the *.py file via shift-return or the icon. I did try it in jupyter and had issues. I split up the calls as suggested in jupyter and when I call intensities in subsequent cells its works as I would expect. It seems that switching back and forth from spyder to jupyter is not a good thing without releasing (so to speak) the driver. In spyder, when I execute that simple code you initially suggested then I also can't execute the *.py script multiple times and have it work each time. But if I get it to execute once in one python script and then only include the intensity call in another I can keep calling  and calling intensities without errors. So, I am wondering if I do need to have some sort of open / close portion to test the same script when coding? I hope I described that well enough.  

In terms of the USB error, yes I really wonder if that has to do with the increased security put into MacOS 15.x?  I checked in the privacy and security system preferences but didn't see anything waiting to be checked. However, that may not explain the cycling behavior. I changed from pyseabreeze to the cseabreeeze and got basically the same thing. 

> I am not entirely sure how you are using the the module yet. From the info you provided so far I assume you're using Spyder and you're executing the same cell over and over again.
> I suggest you put [...]

Yes, I think you understood what I was doing. Shouldn't the *.py code be able to executed multiply time?  The following three outputs keep cycling through each execution of my trial code in spyder with and `Errno 13` then `Errno 19` then runs (albeit with an exception). Here is a copy.

```
runcell(0, '/Users/waldow/.spyder-py3/flame_test.py')
Traceback (most recent call last):

  File "/Users/waldow/.spyder-py3/flame_test.py", line 17, in <module>
    devices = list_devices()

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/spectrometers.py", line 35, in list_devices
    return list_devices._api.list_devices()

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/api.py", line 98, in list_devices
    dev.open()

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/devices.py", line 326, in open
    self._transport.open_device(self._raw_device)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/transport.py", line 142, in open_device
    self._protocol = self._protocol_cls(self)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/protocol.py", line 66, in __init__
    self.send(0x01)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/protocol.py", line 93, in send
    return self.transport.write(data, timeout_ms=timeout_ms)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/seabreeze/pyseabreeze/transport.py", line 160, in write
    return self._device.write(self._endpoint_map.ep_out, data, timeout=timeout_ms)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 940, in write
    intf, ep = self._ctx.setup_request(self, endpoint)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))

  File "/Users/waldow/opt/anaconda3/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
[... 105 lines cut ...]
[... cut ...] ure how much an improvement that might be. There might not be a lot of improvement in the overhead of the USB calls?
2) Another thing that confused me was that apparently now a Flame-S spectrometer is basically a USB2000+ and it took a long time to figure that out. I had only previously used USB650s in my teaching with OI or Vernier software. I finally was able to piece together when trying to develop for this new spectrometer. 

Thanks for the help and I hope my meandering comments are helpful.

ap--, 2020-06-28:
Hi Dean,

> Perhaps changing that line and similar ones above to use the print ("",) format (python 3.x I guess) when then work for both or stating in a cell in jupyter would help some like me. But, I suspect a [...]

Yes. Jupyter notebooks automatically print the last returned output. The code in the documentation is an example of how the output would look like if you run it in an interactive session in a python interpreter. I'll try to make that more clear.

> It seems that switching back and forth from spyder to jupyter is not a good thing without releasing (so to speak) the driver. In spyder, when I execute that simple code you initially suggested then I also can't execute the *.py script multiple times and have it work each time.

That's the problem. You're basically trying to access the spectrometer from two independent processes at the same time, and that causes the issues. It's a very unusual use case, although I know understand why it caused confusion. What I would recommend is: Use the spectrometer from either spyder oder the jupyter notebook. But don't use it at the same time from both.

> I have mostly been running from spyder where I execute the *.py file via shift-return or the icon. I did try it in jupyter and had issues. I split up the calls as suggested in jupyter and when I [...]

I'll try to rephrase in my words and will add a few comments. Please confirm if I understand you correctly.

1. I think you are using spyder as your python editor and when you wrote a script in spyder you copy it into a cell of a jupyter notebook? When you execute code in spyder via shift-return, the python interpreter running in spyder is opening a handle to the usb spectrometer and keeping the connection alive until the python garbage collector can clean up the spectrometer instance. So when you run the code in a jupyter notebook while spyder is still running a second python interpreter run by jupyter tries to open the handle to the usb spectrometer and it can't. I would suggest moving your entire workflow to either spyder or jupyter instead of doing it in parallel. If you insist (which I don't recommend) you should be able to move from one to the other by deleting the reference to your spectrometer instance and calling SeaBreezeAPI().shutdown(). (Please note, that I am not providing an explicit example of how to do this, because then I will receive emails with people copy pasting this code and running into issues trying exactly this thing.)

2. When you're developing interactively in either spyder or jupyter you see the error pop up when you execute all of the code multiple times. What this does is that it basically has to close the spectrometer connection and reopen it because you're trying to overwrite your reference to the currently opened spectrometer with a new instance of the same spectrometer. I understand how this occurs now, and will add it to my list of things to improve about spectrometer connection handling. But there's a very simple workaround, as you already tested, and it's to simply not execute the code that creates the Spectrometer instance more than once.

> In terms of the USB error, yes I really wonder if that has to do with the increased security put into MacOS 15.x? I checked in the privacy and security system preferences but didn't see anything [...]

I think I understand under which circumstances this error occurs now. There's a chance that this is fixed in the current development branch, but I'll try to add another test. I might be able to test this on a newer Mac in the future.

> Here, I am again referring to same URL I quoted above but now I can't get "spec.integration_time_micros(1000)" to not work. 

Okay, as discussed further above I think the misunderstanding was that the code example was from an interactive python interpreter session. And `now I can't get "spec.integration_time_micros(1000)" to not work` is a success :smiley: 

> Since, you indicate that boxcar and scans to average are not supported for this spectrometer, how would I know that? Where might I have missed that information? I have been looking into the [...]

The OmniDriver is OceanOptics' (OO) main driver suite which is not used by python-seabreeze. From the hardware documentation I don't see that the Flame-S would be able to average on the spectrometer itself, so my guess would be that the OO spectrometer software is just doing the averaging on the PC. There are other spectrometers that have the capability to average N spectra on the hardware itself and only transfer the average spectrum over usb.  

<br>

#### regarding the timing issues:

> I did some timing tests and am getting about 3.4 to 3.7 ms per call (with a 1 ms int time) to get intensities. Is that would you might expect likely depending on the number of usb hubs one has to go [...]

**Disclaimer: all of this info is me trying to make sense of all the developer/oem datasheets and infomaterial I read over the years. You could probably verify all of this via directly contacting OO and talking to an engineer**

By default most of the OO spectrometers run in a `free running` trigger mode. That means they are constantly acquiring spectra and everytime you request a new spectrum via usb it will wait until the next acquired spectrum is available. My guess is that below ~1600us you're skipping exactly one recorded spectrum. Summing the data in your loop should only take on the order of 10us. So this is either overhead in the python calls (which coul [... cut ...] th the spectrometer. (Not too complicated, but I'm lacking the required information and hardware, and I am not being payed for any of this :sweat_smile: )

<br><br>
So, I hope I could answer your questions! And thanks again for the very detailed report :smiley: :+1: 
I'll try to improve the documentation a bit and will see if I can add another test for rerunning the spectrometer instance creation.

Let me know how things go, and if there's anything else that is unclear.

Cheers,
Andreas :smiley:

waldowda, 2020-06-28:
Hi Andreas, 
Thanks for the nice reply.  Your last comment first... You are doing a great job and don't worry about delving into the firmware to decipher the flame-s / usb2000+ part.  I suspect a simple note in the doc.s might be handy for those buying current flame-s's versus your time to play with firmware. I just had no idea the flame-s was viewed as a usb2000+. LOL. I finally pieced that together from labview and their NI drivers (where it returned a product ID of 4126 which concurs with the manual as it states in hex that `Ocean Optics Vendor ... the Product ID is 0x101E.`) but I didn't know that USB-2000+ also is that Product ID until I looked at the labview code.

**Regarding your point 1 above:** I think the initial reason I tried jupyter after spyder was because of the initial errors and not being sure if / what the cause was coupled with my learning curve.  One environment is far better. I will cautiously look into that shutdown command. But, I think developing in Jupyter with one cell for the start up and then playing with code in subsequent cells would be a good approach for now. I am also looking into ScopeFoundry.

**Regarding your point 2 above:** OK. I think I follow. If I do split the code up into the initial parts and testing portions, then I don't see these warning / notes. 

> I think I understand under which circumstances this error occurs now. There's a chance that this is fixed in the current development branch, but I'll try to add another test. I might be able to test this on a newer Mac in the future.

I am happy to test from my mac if that is helpful. I cloned my anaconda environment and will try to test your development branch if I can get that working and if it would be helpful.

Also, I think you are likely correct in that the OI OmniDriver must be off loading spectra and then averaging or boxcaring via code in their driver.

Timing wise, I think I understood your notes. Also, I realized that the specs indicates a 400 Hz scan rate (which it suggests is processor / computer dependent) which is 2.5 ms. That is about the minimum response rate I see for int times less then 1600 µs but then I don't understand why I could get close to the real int. time at int. times > 1600 µs. I am not sure I need to understand this. 😄 Leaving as many Hubs our of the connection path is possible is likely best, yes. 

I also might not be reading correctly, but the paragraph you reference seems to state that in Normal mode that it waits for a request and then will acquire one spectrum and store it. Then run two more int time acquisitions automatically and 'if' another request is made it will store that one during either the second or third scan. Lastly, it will go idle waiting for a new request. Well, I am not sure exactly what it is doing in normal mode. However, I just tested in triggering with mode 1 and going back to mode 0 using an arduino to trigger it. So, at least it seems I can take one scan triggered then switch back to normal to acquire more scans to effectively signal average. I am trying to integrate the Flame with a Gamry potentiostat for spectroelectrochemistry experiments and there is TTL sync pulse per acquisition on the Gamry so I think this has potential for syncing them together well and still doing signal averaging as I only get one pulse per acquisition from the Gramy.

Did I miss any of your questions / points? Anyway, I think I am at a point I can get on with the testing / coding. Let me know if I can test anything to help out via my Flame.   Thanks, Dean

ap--, 2020-06-29:
Hi Dean,

> Thanks for the nice reply. Your last comment first... You are doing a great job and don't worry about delving into the firmware to decipher the flame-s / usb2000+ part

Thank you :smile: And thank you so much for your contribution :sparkling_heart: 

<br>

> But, I think developing in Jupyter with one cell for the start up and then playing with code in subsequent cells would be a good approach for now. I am also looking into ScopeFoundry.

There's a few projects that have already integrated python-seabreeze with user interfaces. A browser based example is https://dash-gallery.plotly.host/dash-ocean-optics/ and it's worth checking out other projects listed here: https://github.com/ap--/python-seabreeze/network/dependents

Regarding development: I do most of my python development with JetBrains PyCharm. There's a free community version of it that is definitely worth giving a try.

<br>

> I am happy to test from my mac if that is helpful. I cloned my anaconda environment and will try to test your development branch if I can get that working and if it would be helpful.

That would be great. If you could install pytest, clone the repository and checkout and pip install  the `develop` branch and run
```
pytest
```
in the repository root. That should run all the tests. (You can run them with and without a spectrometer connected)

<br>

> However, I just tested in triggering with mode 1 and going back to mode 0 using an arduino to trigger it. So, at least it seems I can take one scan triggered then switch back to normal to acquire [...]

Great! Yes, I think your use case sounds like it would best work with the Hardware level trigger mode.
Reading the docs mode 1 is external hardware level trigger mode. :+1: 

```
External Hardware Level Trigger Mode
In the External Hardware Level Trigger mode, a rising edge detected by the spectrometer from
the External Trigger input starts the integration period specified through the software interface.
After the integration period, the spectrum is retrieved and is ready to be read by the user. As
long as the trigger level remains active in a logic one state, back-to-back acquisitions can occur,
as in the Normal mode, until the trigger transitions to an inactive level. As in normal mode, no
further spectra are acquired until the original spectrum is read by the user.
```

So If you're using an arduino anyways I would try the following (Note: I think this should work. I haven't tested this yet):

Switch the spectrometer to trigger mode 1 and then start your acquisition loop:

```python
num_spectra = 100

# in trigger mode 1 the intensities() call should block until a trigger signal
# was received by the spectrometer (or a timeout occurs)
first = spec.intensities()
specta = [first]

# if the arduino now keeps the trigger signal active the spectrometer will continue to acquire spectra
for _ in range(num_spectra - 1):
    spectra.append( spec.intensities() )
    
# now the arduino can deactivate the trigger line
```

Cheers,
Andreas

waldowda, 2020-06-30:
Greetings, I'll take a look at those projects. I'll also look at `JetBrains PyCharm` as well as try out `pytest` (and I will add the results to this issue) in the next few days. Too many Zoom meetings today... The arduino is simply a place holder for now though it is pretty cool and your code is very succinct. The Gamry Ref 600+ puts out a very short TTL pulse per data point and so I would use that for the first point - hence my thought of returning to trigger mode (0) to acquire the rest to average before the next data point. (I could also use edges detection I presume to the same effect.) I anticipate a 100 ms delta time per electrochemical data point so I would need to wrap up any calculations before the next point. Is there an advantage to appending vs adding to the array perhaps then averaging after the `num_spectra` in your example? Perhaps one then can get a stand. dev. versus only an average? Thanks, Dean

waldowda, 2020-07-03:
Sorry. I have been running out of time lately and had some issues getting the develop version of python-seabreeze cloned and checkedout in my test environment. It is likely due to me still learning anaconda and path issues to know where it gets cloned to and then how to use checkout. I hope to make progress this weekend.

ap--, 2020-07-05:
No worries :blush: I'm quite busy currently too, and we're not at all in a rush.
I'll see if I can put together a guide on how to setup everything needed for running tests on macos :smiley: 

Cheers,
Andreas :smiley:

waldowda, 2020-07-15:
Additional question but maybe should be a separate issue... In `devices.py`, USB2000PLUS has four trigger modes: `trigger_modes = TriggerMode.supported("NORMAL", "LEVEL", "SYNCHRONIZATION", "EDGE")` and this seems consistent with the Flame manual (Document: 225-00000-000-11-201503b). However, in another PDF specific to triggering for firmware 3.0 and above (external-triggering-options_firmware3.0andabove.pdf), it lists 5 modes for the USB2000+ and Flame inserting a software trigger at data value 1 and continuing up to data value 4. (page 18)  Is one of these a typo?

```
HR2000+, USB2000+ and Flame-S Set Trigger Mode
Data Value = 0 -> Normal (Free running) Mode
Data Value = 1 -> Software Trigger Mode
Data Value = 2  ->  External Hardware Level Trigger Mode 
Data Value = 3 -> External Synchronization Trigger Mode 
Data Value = 4 -> External Hardware Edge Trigger Mode
```
So, if I want hardware edge triggering, does one set trigger mode to 3 or 4? Or is firmware 3.0 old? Sorry, I am puzzled by two different suggestions for edge triggering. Maybe I am missing something obvious.

Here is the link to the firmware 3.0 pdf on triggering I found: [https://www.oceaninsight.com/globalassets/catalog-blocks-and-images/manuals--instruction-old-logo/electronic-accessories/external-triggering-options_firmware3.0andabove.pdf](url)

I will also give the pytest a try when I can get the develop version installed. Thanks!

ap--, 2020-07-15:
Good find!
The datasheets seem to be inconsistent, sometimes even within the same pdf...

This is from the USB2000+ datasheet:
**page 21** (but it says 'five' modes)
```
Data Value = 0 -> Normal (Free running) Mode
Data Value = 1 -> External Hardware Level Trigger Mode
Data Value = 2 -> External Synchronization Trigger Mode
Data Value = 3 -> External Hardware Edge Trigger Mode
```

**page 34**
```
0 = Normal (Free running) Mode
1 = Software Trigger Mode
2 = External Hardware Level Trigger Mode
3 = External Synchronization Trigger Mode
4 = External Hardware Edge Trigger Mode
```

So with the document you found I'd say 5 modes is correct. But you should ask the ocean optics tech support if you want to be sure. But I think that this might need fixing in python-seabreeze too. We should open a new issue regarding trigger support for USB2000+ and Flame-S.

Firmware version 3.0 is definitely new!
(I my current guess is, that a USB2000+ has firmware version = 0.x 1.x 2.x and the Flame-S is just a USB2000+ with firmware version 3.x)

I might be able to write a guide for mac on the weekend (if not this then definitely the following weekend)

Cheers,
Andreas :smiley:

waldowda, 2020-07-15:
OK... in a day or two I'll start a new issue for triggering as well as see if I can get definitive info on triggering modes from OI. Thanks.

waldowda, 2020-07-30:
Andreas, I worked with the NI ocean optics VIs and dived into the trigger subVI and was able to add trigger mode '4'. Then I appear to be able to trigger via edge. However, I am not finding an obvious way to add trigger mode 4 to py-seabreeze.  Also the nomenclature of the modes I see in seabreeze doesn't match the info OI gave me (see above).  It does appear that the Flame-X and a Flame-NIR show up in seabreeze but Flame-S does not. I wonder if there are enough unique aspects to have a specific entry for the Flame-S distinct from USB2000+? Thoughts? Anyway, I am confident that the edge mode is not  able to be set currently. Thanks, Dean

ap--, 2020-08-18:
Closed. See: #97

closed 2020-08-18 by ap--