ap--/python-seabreeze #94: Ocean FX | Integration Time Error: Input was out of bounds

Roy4777, 2020-07-22:
### spectrometer and system information ###

- model: Ocean FX
- operating system: Windows 7 64bit
- python version: Python 3.7.6
- python-seabreeze version: Version: 1.1.0
- installed-via: pip 

### current problem ###

I get integration time out of bounds for 100μs even though Ocean FX can have integration time as low as 10μs. This problem doesn't occur if I set an integration time over 1000μs. I followed the documentation for adding a spectrometer and added the following class for Ocean FX to seabreeze/pyseabreeze/devices.py and implemented the specialized spectrometer feature class in seabreeze/pyseabreeze/features/spectrometer.py

P.S. I am new to coding so I might be missing out on something trivial and thank you for creating this library. :)

`
class OceanFX(SeaBreezeDevice):

    model_name = "OceanFX"

    # communication config
    transport = (USBTransport,)
    usb_product_id = 0x2001
    usb_endpoint_map = EndPointMap(
        ep_out=0x01, lowspeed_in=0x81
    )  # XXX: we'll ignore the alternative EPs
    usb_protocol = OBPProtocol

    # spectrometer config
    dark_pixel_indices = DarkPixelIndices.from_ranges()
    integration_time_min = 10
    integration_time_max = 85000000
    integration_time_base = 1
    spectrum_num_pixel = 1024
    spectrum_raw_length = (1024 * 2) + 64  # XXX: Metadata
    spectrum_max_value = 16383
    trigger_modes = TriggerMode.supported("OBP_NORMAL", "OBP_EXTERNAL", "OBP_INTERNAL")

    # features
    feature_classes = (
        sbf.spectrometer.SeaBreezeSpectrometerFeatureOceanFX,
        sbf.rawusb.SeaBreezeRawUSBBusAccessFeature,
    )
`
### steps to reproduce ###
`import seabreeze
seabreeze.use("pyseabreeze")
import numpy as np
from seabreeze.spectrometers import Spectrometer

spec = Spectrometer.from_first_available()

spec.integration_time_micros(100) 

spectrum = spec.spectrum()`

### minimal code example and error (very helpful if available) ###

```
---------------------------------------------------------------------------
SeaBreezeError                            Traceback (most recent call last)
<ipython-input-10-be2988c76f54> in <module>
     14 #Set Integration Time
     15 
---> 16 spec.integration_time_micros(100) 
     17 
     18 

~\AppData\Roaming\Python\Python37\site-packages\seabreeze\spectrometers.py in integration_time_micros(self, integration_time_micros)
    263                 )
    264             else:
--> 265                 raise e
    266 
    267     @property

~\AppData\Roaming\Python\Python37\site-packages\seabreeze\spectrometers.py in integration_time_micros(self, integration_time_micros)
    250         try:
    251             self._dev.f.spectrometer.set_integration_time_micros(
--> 252                 integration_time_micros
    253             )
    254         except OverflowError:

src\seabreeze\cseabreeze\c_seabreeze_wrapper.pyx in seabreeze.cseabreeze._wrapper.SeaBreezeSpectrometerFeature.set_integration_time_micros()

SeaBreezeError: Error: Input was out of bounds
```

Roy4777, 2020-07-23:
I realized I wasn't using the pyseabreeze backend and the error was due to different integration limits on the cseabreeze backend.

closed 2020-07-23 by Roy4777

ap--, 2020-07-23:
Hey @Roy4777 

happy that you found the solution to your problem 😃  
And feel free to create a pull request if you manage to get the OceanFX working with pyseabreeze.
I'm happy to review and give feedback and help where I can.

Cheers,
Andreas 😃

Roy4777, 2020-07-23:
Hi @ap-- ,

We are able to talk to OceanFX using the cseabreeze backend. We haven't been able to connect with the pyseabreeze yet. I'm looking into what I might be doing wrong.