ap--/python-seabreeze #96: minimum integration time

asenchristov, 2020-07-29:
### spectrometer and system information ###

- model: QE Pro
- operating system: Linux
- python version: Python 3.6.9
- python-seabreeze version: 'current master'
- installed-via: python setup.py install

### current problem ###

The QE Pro can go down to 8000 us integration time, while in the seabreeze there is a hard-coded limit of 10000 us

ap--, 2020-08-13:
Hi @asenchristov 

The datasheet agrees with you.
I'm a bit hesitant to change, because I don't know if there was a reason from OceanOptics to set it to 10ms in the orginal seabreeze C library.

That being said, can you confirm the 8ms in one of their other softwares? Spectrasuite or whatever the newest software is called? If yes, we'll change it in both backends.

Cheers,
Andreas 😃

asenchristov, 2020-08-18:
Hi,
I can confirm that 8ms works with this device (QE Pro). I was able to use it in their "oceanview" software and also in seabreeze (after mods to seabreeze (and the C lib) - change is necessary only in two places, one in the python code, one in the C lib).
cheers, Asen.

ap--, 2020-08-18:
It'd be great if you could make a pull request! ❤️ 

Thanks,
Andreas 😃

ap--, 2020-08-18:
Closed via 8567d662b6d98f13d2a097355ac71f7e7769196d

closed 2020-08-18 by ap--