ap--/python-seabreeze #97: Flame-S Edge Trigger Mode and Timing

waldowda, 2020-08-04:
### spectrometer and system information ###

- model: Flame-S supposed to be the same as USB-2000PLUS I believe
- operating system: MacOS 15.5.5 or Win10 current
- python version: 3.7.7 (via Anaconda) or 3.8.x
- python-seabreeze version: v1.1.0 I believe
- installed-via: conda and python for anaconda

### current problem ###

_This issue is branched from a previous issue regarding the Flame-S --> "Flame-S spectrometer is seen but doesn't report data #92"._

(Andreas: Sorry this is long but hopefully understandable and helpful. Dean )

**First Trigger / Timing Issue**

I suspect the settings for triggering the HR2000+, USB2000+, and Flame-S are missing edge triggering unless the trigger numbers one specifies are not the same as the manuals.

I heard back from OI regarding discrepancies in the manuals and per your suggestion I am starting a new issue.

Here is what tech support said which is consistent with page 34 above from the USB2000+ manual and from the Flame  for the Flame-S spectrometer: 

```
The Flame-S has 4 trigger modes:
	• On Demand (normal mode): mode 0
	• Edge: mode 4
	• Level: mode 2
	• Synchronous: mode 3
```

Note that they leave out '1' which is described as a software trigger...

and from the part of the Flame manual that is consistent with the note from tech support (though in other parts of the manual it appears to have contradictory settings). I suspect this issue has to do with the differences between the Flame-S and Flame-T models where the latter one seems have the other set of trigger values seen in the manual.

```
HR2000+, USB2000+ and Flame-S Set Trigger Mode
Data Value = 0 -> Normal (Free running) Mode
Data Value = 1 -> Software Trigger Mode
Data Value = 2  ->  External Hardware Level Trigger Mode 
Data Value = 3 -> External Synchronization Trigger Mode 
Data Value = 4 -> External Hardware Edge Trigger Mode
```

From my testing of both cseabreeze and py-seabreeze...

```
import seabreeze
seabreeze.use('cseabreeze')

#seabreeze.use('pyseabreeze')

from seabreeze.spectrometers import list_devices, Spectrometer

spec = Spectrometer.from_serial_number("FLMS16723")
spec.trigger_mode(0)
```

So, if the data value for the trigger info: "T(word)" in py/c-seabreeze goes from 0 to 3, then I suspect it will not be possible to setup edge trigger mode which is '4'. However, if there is an internal translation from 0...3 to 0, 4, 2, 3 then perhaps but it is not clear to me if that is the case.

I was able to add '4' to the NI Labview trigger subVI and was able to trigger. I can't figure out if I can modify the py-seabreeze code to add trigger mode 4 or even if that would be the correct thing to do.

I have found the following in spectrometers.py but I have not been able to go past this to see where trigger mode gets set.

```
 def trigger_mode(self, mode):
        """set the trigger mode of the device

        Parameters
        ----------
        mode : `int`
[... 225 lines cut ...]
ValueError                                Traceback (most recent call last)
<ipython-input-1-195d4932b562> in <module>
     30 print ("Wavelengths: ", wavelengths)
     31 
---> 32 data = spec.intensities(correct_dark_counts=True, correct_nonlinearity=True)
     33 #data = spec.intensities()
     34 

~\.conda\envs\SpecEchem Environment\lib\site-packages\seabreeze\spectrometers.py in intensities(self, correct_dark_counts, correct_nonlinearity)
    177             measured intensities in (a.u.)
    178         """
--> 179         if correct_dark_counts and not self._dp:
    180             raise self._backend.SeaBreezeError(
    181                 "This device does not support dark count correction."

ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```

But... the following works apparently:

```
spec.trigger_mode(4)
```

It doesn't error. I will add more if I can get it to trigger from the potentiostat but that is all I have time for now.

ap--, 2020-08-10:
Hi Dean,

Thank you for the really detailed error description! 🥇 This makes solving issues a lot easier.
Also with the newer version of `seabreeze` out, let's start debugging your issue.

**First:** the way you added the new `TriggerMode` is correct! 😃👍 Basically `pyseabreeze` just blocks using unsupported triggermodes from getting send to the spectrometer and when you added `SINGLE_SHOT` to the `TriggerModes.supported()` call you allowed it to be passed through.

**Second:** the darkcount error you get is definitely a bug. And I thought I fixed this a long time ago already... Can you show the output of:

```python
import seabreeze
print(seabreeze.__version__)
```

Let's solve the darkcount issue first, and then move towards the trigger issue.

Also, just to let you know: I have a work in progress branch where I implement that the Flame-S will be recognized correctly and will push that soon. But time is currently a scarce commodity so it's hard for me to predict when I can continue to work on it. Let's try to get out the dark count issue because that should make everything you need for now.
Then we'll add the trigger mode to a new version of seabreeze and then we'll investigate the constant time offset issue 😅  

Cheers,
Andreas 😃

waldowda, 2020-08-10:
Hi Andreas, I just saw your message. I think you are correct that if we can get the Dark to work then I can get the trigger '4' single shot to work and can actually start putting the potentiostat with the spectrometer. I tested some OI  py code but it is not ready for this application yet. I'll run your test shortly. Did I understand that there is a new py-seabreeze version? Well, I'll do the test first. Should be able to get the version number in a few moments. I'll update this entry when I get the info. Dean

ap--, 2020-08-10:
Hi Dean,

1.1.0 should already have the bugfix. I guess I made some mistake there.

And yes, the newest version is 1.2.0 and I released it yesterday. It fixed a few small issues with test and usb-transport handling. But should not change anything for you.

I'll try to reproduce the darkcount error on my machine tomorrow. Will have to sign off for today!

Have a nice day,
Andreas 😃

waldowda, 2020-08-10:
OK... I'll try the py vs c - seabreeze test again on both my surface pro 7 and MacBook pro... maybe it is an OS issue but doubt it and update if I find anything useful. I have many zoom meetings today so it might take a bit. Regards, Dean

waldowda, 2020-08-11:
I wanted to setup another test on the surface pro 7 from a new environment. I made a new environment with python 3.8 and did a pip install seabreeze (1.2.0) then realized that it didn't install py-seabreeze so then issued "pip install seabreeze[pyseabreeze]".  Since I had already installed drivers before via 1.1.0, I didn't worry about the following error for seabreeze_os_setup. Maybe this is a problem? 

```
Install windows drivers? [y/n] y
Downloading windows drivers from github
Error when installing drivers
Press [enter] to close.
```

To test, I using the following code first and in creating a spectrometer it erred in a  manner I have not seen. An eeprom error.

```
import numpy
import time
import matplotlib.pyplot as plt

# explicitly request pyseabreeze
import seabreeze
#seabreeze.use('cseabreeze')

seabreeze.use('pyseabreeze')

print (seabreeze.__version__)

from seabreeze.spectrometers import list_devices, Spectrometer

devices = list_devices()

print (devices)
```

and got the following back: 

```
1.2.0
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-6308e1042135> in <module>
     11 print (seabreeze.__version__)
     12 
---> 13 from seabreeze.spectrometers import list_devices, Spectrometer
     14 
     15 devices = list_devices()

~\.conda\envs\pyseabreeze\lib\site-packages\seabreeze\spectrometers.py in <module>
     13 
     14 # get the backend and add some functions/classes to this module
---> 15 _lib = seabreeze.backends.get_backend()
     16 
     17 SeaBreezeError = _lib.SeaBreezeError

~\.conda\envs\pyseabreeze\lib\site-packages\seabreeze\backends.py in get_backend()
     76     backends = _SeaBreezeConfig["available_backends"]
     77 
---> 78     backend = _use_backend(requested)  # trying to import requested backend
     79     if backend is None and fallback:
     80 

~\.conda\envs\pyseabreeze\lib\site-packages\seabreeze\backends.py in _use_backend(name)
     13             import seabreeze.cseabreeze as sbb
     14         elif name == "pyseabreeze":
---> 15             import seabreeze.pyseabreeze as sbb
     16         else:
     17             raise ValueError("unknown backend '{}'".format(name))

~\.conda\envs\pyseabreeze\lib\site-packages\seabreeze\pyseabreeze\__init__.py in <module>
      2 
      3 """
----> 4 from seabreeze.pyseabreeze.api import SeaBreezeAPI
      5 from seabreeze.pyseabreeze.devices import SeaBreezeDevice
      6 from seabreeze.pyseabreeze.exceptions import SeaBreezeError, SeaBreezeNumFeaturesError

~\.conda\envs\pyseabreeze\lib\site-packages\seabreeze\pyseabreeze\api.py in <module>
     10 import weakref
     11 
---> 12 from seabreeze.pyseabreeze.devices import SeaBreezeDevice, _model_class_registry
     13 from seabreeze.pyseabreeze.transport import (
     14     USBTransport,

~\.conda\envs\pyseabreeze\lib\site-packages\seabreeze\pyseabreeze\devices.py in <module>
    429 # ========================
    430 #
--> 431 class USB2000PLUS(SeaBreezeDevice):
    432 
    433     model_name = "USB2000PLUS"

~\.conda\envs\pyseabreeze\lib\site-packages\seabreeze\pyseabreeze\devices.py in USB2000PLUS()
    455     # features
    456     feature_classes = (
--> 457         sbf.eeprom.SeaBreezeEEPromFeatureOOI,
    458         sbf.spectrometer.SeaBreezeSpectrometerFeatureUSB2000PLUS,
    459         sbf.continuousstrobe.SeaBreezeContinuousStrobeFeatureOOI,

AttributeError: module 'seabreeze.pyseabreeze.features' has no attribute 'eeprom'
```

I doubt this is that helpful but thought I would pass it along. Maybe it is related or may be new per 1.2.0. I'll update if I learn more.

ap--, 2020-08-12:
Hi Dean,

So the `seabreeze_os_setup` windows error is something i've had troubles to reproduce lately. But I've definitely seen it too. I created a new issue #99 for that.

That the eeprom feature is missing is very strange. I'll put that on my list for trying to reproduce.

Cheers,
Andreas 😃

waldowda, 2020-08-12:
I am beginning to wonder if some of these errors might be connected to a particular anaconda environment or win10 / MacOS permissions. I am going to try two things: 1) not use anaconda on my win10 box but the chocolaty procedure (though I like anaconda's flexibility) and 2) on my MacOS box, try to install in a fresh environment. It sure seems like we are close to have the Flame-S working nicely. I may end up finishing my Labview version to get things running since my teaching schedule will get heavy shortly. BTW: I have not run that test you asked about for the development version. I was not confident I was able to download and install the dev version properly. Regards, Dean

ap--, 2020-08-13:
Okay, small update:
- [x] The dark pixel issue was still present in `1.1.0` and is fixed in `1.2.0`. I was wrong about that.
- [ ] I'll have a look at the `seabreeze_os_setup` issue next
- [ ] Then we'll investigate the TriggerMode issue

Regarding anaconda:
It's totally fine if you keep on using anaconda. For now make sure that you create a conda environment with python=3.8
that will rule out a few suspicions I have regarding failing SSL certs for `seabreeze_os_setup`

waldowda, 2020-08-13:
OK. If I understand your first point... the dark pixel issue is the reason I can't set True to Dark and nonlinearity correction in the get spectrum call and it is currently not fixed in version 1.2.0? Second, I'll transition to anaconda environments with python 3.8.  Thanks! I think if I can get the dark part fixed I can patch the '4' setting temporarily but having py-seabreeze recognize Flame-S and have the trigger settings match would be great.

ap--, 2020-08-13:
> OK. If I understand your first point... the dark pixel issue is the reason I can't set True to Dark and nonlinearity correction in the get spectrum call and it is currently not fixed in version 1.2.0?

It's fixed in version 1.2.0 So if you are running 1.2.0 you should be able to set it to True and not crash with the error you reported before :)

> Second, I'll transition to anaconda environments with python 3.8. Thanks!

Great. I have a suspicion that a possible download error will crash with a unspecific error message like you reported. Running it with python=3.8 should circumvent that. (It's totally possible that there's another bug though)

>  I think if I can get the dark part fixed I can patch the '4' setting temporarily but having py-seabreeze recognize Flame-S and have the trigger settings match would be great.

Yes. As soon as I have some spare time to continue and finish the work on "real" Flame-S support (not aliased as a USB2000plus) this will be supported out of the box.

Cheers,
Andreas 😃

waldowda, 2020-08-14:
I am not quite sure what happened but I stopped getting backend error on macOS. But then also discovered that the following seems to work after I modified devices.py to allow SINGLE_SHOT. I really doubt that did much for the backend USB error but...

```
print ("Stray Light Coef",spec.stray_light_coeffs)
print ("_nc: ", spec._nc)
print ("_dp: ", spec._dp)

data = spec.intensities(correct_dark_counts=True, correct_nonlinearity=False)
data

wavelengths = spec.wavelengths()
#print ("Wavelengths: ", wavelengths)

wavelengths1 = wavelengths[24:2000]

data0b = data[24:2000] 

plt.plot(wavelengths1,data0b)
plt.show()
```

and then get:

```
Stray Light Coef <bound method DeprecatedSpectrometerMixin.stray_light_coeffs of <Spectrometer USB2000PLUS:FLMS16723>>
_nc:  None
_dp:  [6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20]
```

With dark=True and nc=False:

[image]

With dark =False and nc=False:

[image]

So, it seems perhaps that the problem is with the nonlinearity portion. When I switch back to cseabreeze I can easily grab those coef.s

Still no idea why the backend USB issues stopped.

ap--, 2020-08-15:
the backend issues probably stopped because you installed `pyusb` which is required for the pyseabreeze backend to work.

I'm not sure why it wasn't installed in the first place, but debugging issues with your local environment is almost impossible in retrospect. You would need to show every single command including output that you used to create the environment for me to be able to MAYBE figure out what's wrong with your python environment.

The spectra plot you showed are great! that means it's working already :+1: :medal_sports: 

**In other news:**
I just pushed the code required to let the pyseabreeze dynamicllay detect a flameS spectrometer. That means it's one step closer to having full flame-s support.

I also added the nonlinearity coefficient feature back into pyseabreeze for the USB2000+ and by extension the FlameS.
Idk why it was missing there.

And the Triggermode 4 support is added to the FlameS for pyseabreeze, too.

So that means, the current master branch should make the FlameS work as expected.
If you could test this, and there's no more issues I'll cut a new release.

Cheers,
Andreas :smiley:

waldowda, 2020-08-15:
Cool... 👏  All I did to install seabreeze was pip install seabreeze via terminal in the anaconda python3.8 environment I setup fresh. No dependencies were invoked as far as I could tell. It was the error that led me to pyusb. Should pyusb get added when installing seabreeze that way? I'll have to learn more about GitHub so I make sure I install the main branch.  I'll try to test that out. For fun... here is a plot of an overlay of 53 absorbance spectra collected every 100 ms with 24 ave.s for a thin film of P3HT. Just testing. Without the non-linear correction... BTW I think trigger mode 4 should work for USB2000+ too ??? I don't have one though to test. I only have other USB650s. Will let you know. 

[image]

waldowda, 2020-08-15:
I installed what I think is the master branch. It didn't seem to be correct. The files have a current timestamp. I am still getting it assigned to USB2000+, don't see trigger 4, and nonlinear correction is not working. I must not have succeeded as I can see your changes on GitHub. Will see if I can figure out what I did wrong. 

```
1.2.0
[<SeaBreezeDevice USB2000PLUS:FLMS16723>]
Serial Number:  FLMS16723
Max Intensity:  65535.0
Min. Int. Time:  1000
Stray Light Coef <bound method DeprecatedSpectrometerMixin.stray_light_coeffs of <Spectrometer USB2000PLUS:FLMS16723>>
```

waldowda, 2020-08-16:
Also, I was trying to figure out why the serial number is not reported and I looked at the Flame S class. Why is nonlinearity in USB2000+ and not in Flame-S and the other Features are there?

```
class FLAMES(USB2000PLUS):

    model_name = "FLAMES"

    # spectrometer config
    dark_pixel_indices = DarkPixelIndices.from_ranges((6, 21))  # as in seabreeze-3.0.9
    integration_time_min = 1000
    integration_time_max = 655350000
    integration_time_base = 1
    spectrum_num_pixel = 2048
    spectrum_raw_length = (2048 * 2) + 1
    spectrum_max_value = 65535
    trigger_modes = TriggerMode.supported(
        "NORMAL", "SOFTWARE", "SYNCHRONIZATION", "HARDWARE", "SINGLE_SHOT"
    )

    # features
    feature_classes = (
        sbf.eeprom.SeaBreezeEEPromFeatureOOI,
        sbf.spectrometer.SeaBreezeSpectrometerFeatureUSB2000PLUS,
        sbf.continuousstrobe.SeaBreezeContinuousStrobeFeatureOOI,
        sbf.rawusb.SeaBreezeRawUSBBusAccessFeature,
    )
```

In USB2000+:

```
    # features
    feature_classes = (
        sbf.eeprom.SeaBreezeEEPromFeatureOOI,
        sbf.spectrometer.SeaBreezeSpectrometerFeatureUSB2000PLUS,
---> sbf.nonlinearity.NonlinearityCoefficientsEEPromFeatureOOI,
        sbf.continuousstrobe.SeaBreezeContinuousStrobeFeatureOOI,
        sbf.rawusb.SeaBreezeRawUSBBusAccessFeature,
    )
```

ap--, 2020-08-16:
Hi Dean,

the command you used is correct.
```console
 pip install git+git://github.com/ap--/python-seabreeze.git
```
If you run it again now, it your seabreeze version should show up as `1.2.0.post9+something`

As you correctly noticed, I forgot to add the nonlinearity coefficient feature to the FlameS. And I pushed to more commits, fixing the issues.

Which OS are you running this on right now? Windows or Mac?

Cheers,
Andreas 😃

waldowda, 2020-08-16:
Good morning (here) (or evening... 😄 )... Thanks. I am currently running MacOS 10.15.6 for now just because my mac my main box but when things are working will be transitioning to the Surface. Installing the new build. I can move to the surface if that is more helpful.

FYI: I still don't get a serial number.

```
1.2.0.post9+g98f56cd
[<SeaBreezeDevice USB2000PLUS:?>]
---------------------------------------------------------------------------
SeaBreezeError                            Traceback (most recent call last)
<ipython-input-2-ba6e06943569> in <module>
     20 #spec
     21 
---> 22 spec = Spectrometer.from_serial_number("FLMS16723")
     23 
     24 print ("Serial Number: ", spec.serial_number)

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/spectrometers.py in from_serial_number(cls, serial)
    120                     return cls(dev)
    121         else:
--> 122             raise cls._backend.SeaBreezeError(
    123                 "No device attached with serial number '%s'." % serial
    124             )

SeaBreezeError: No device attached with serial number 'FLMS16723'.
```

Tried it again after forcing kernel to restart and rebooting Flame and get a different error?

```
1.2.0.post9+g98f56cd
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-1-b7c8313d4847> in <module>
     12 from seabreeze.spectrometers import list_devices, Spectrometer
     13 
---> 14 devices = list_devices()
     15 
     16 print (devices)

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/spectrometers.py in list_devices()
     34         list_devices._api = _lib.SeaBreezeAPI(**_lib._api_kwargs)
     35     # noinspection PyProtectedMember
---> 36     return list_devices._api.list_devices()
     37 
     38 

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/api.py in list_devices(self)
    102                 # opening the device will populate its serial number
    103                 try:
--> 104                     dev.open()
    105                 except USBTransportDeviceInUse:
    106                     # device used by another thread? -> exclude

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py in open(self)
    346         self._transport.open_device(self._raw_device)
    347         # substitute subclass if needed
--> 348         self.__class__ = self.__class__._substitute_compatible_subclass(self._transport)
    349         # cache features
    350         self._cached_features = {}

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py in _substitute_compatible_subclass(cls, transport)
    498 
    499         fpga = _FPGARegisterFeatureOOI(transport.protocol)
--> 500         if fpga.get_firmware_version()[0] >= 3:
    501             return FLAMES
    502         else:

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/fpga.py in get_firmware_version(self)
     31 
     32     def get_firmware_version(self):
---> 33         data = self.read_register(self.Codes.FIRMWARE_VERSION)
     34         fw_version = (
     35             (data >> 12) & 0x0F,  # major

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/fpga.py in read_register(self, register)
     23         fw_raw = self.protocol.query(0x6B, register)
     24         command, data = struct.unpack("<BH", fw_raw)
---> 25         assert command == self.Codes.FIRMWARE_VERSION
     26         return data
     27 

AssertionError: 
```

If I leave things alone and redo the import.... to list devices I am more or less back to the original error.

ap--, 2020-08-16:
Hmmm... 
So the second error is interesting.
Maybe fpga commands are handled differently on the FlameS...
I just pushed a small update that add some more information to the AssertionError.

Please run the code that produces this error again, after you updated your seabreeze install. version should be `1.2.0.post10+something` now.

waldowda, 2020-08-16:
OK... Here is what happened:

```
1.2.0.post10+ge807a69
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-1-b7c8313d4847> in <module>
     12 from seabreeze.spectrometers import list_devices, Spectrometer
     13 
---> 14 devices = list_devices()
     15 
     16 print (devices)

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/spectrometers.py in list_devices()
     34         list_devices._api = _lib.SeaBreezeAPI(**_lib._api_kwargs)
     35     # noinspection PyProtectedMember
---> 36     return list_devices._api.list_devices()
     37 
     38 

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/api.py in list_devices(self)
    102                 # opening the device will populate its serial number
    103                 try:
--> 104                     dev.open()
    105                 except USBTransportDeviceInUse:
    106                     # device used by another thread? -> exclude

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py in open(self)
    346         self._transport.open_device(self._raw_device)
    347         # substitute subclass if needed
--> 348         self.__class__ = self.__class__._substitute_compatible_subclass(self._transport)
    349         # cache features
    350         self._cached_features = {}

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py in _substitute_compatible_subclass(cls, transport)
    498 
    499         fpga = _FPGARegisterFeatureOOI(transport.protocol)
--> 500         if fpga.get_firmware_version()[0] >= 3:
    501             return FLAMES
    502         else:

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/fpga.py in get_firmware_version(self)
     31 
     32     def get_firmware_version(self):
---> 33         data = self.read_register(self.Codes.FIRMWARE_VERSION)
     34         fw_version = (
     35             (data >> 12) & 0x0F,  # major

~/opt/anaconda3/envs/pyseabreeze/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/fpga.py in read_register(self, register)
     23         fw_raw = self.protocol.query(0x6B, register)
     24         command, data = struct.unpack("<BH", fw_raw)
---> 25         assert command == self.Codes.FIRMWARE_VERSION, str(list(map(hex, fw_raw)))
     26         return data
     27 

AssertionError: ['0xfe', '0x55', '0x40']
```

Tried to build in win10 but git fails at building wheels.

waldowda, 2020-08-16:
Here is what I have found which includes all of the register info around page 87 and the FPGA firmware version is at 0x04 read only though you likely have this.

[https://www.oceaninsight.com/globalassets/catalog-blocks-and-images/manuals--instruction-old-logo/spectrometer/flameio.pdf](url)

[image]

ap--, 2020-08-16:
Yes. Could you please connect your FlameS via the OceanOptics Software and check its firmware version? If it's `4.5.5` (which I sadly doubt) we're lucky 😅

waldowda, 2020-08-16:
OK... just came in from the garden. LOL. hold on. Was just looking up to see how to do that.

ap--, 2020-08-16:
Wow! Very nice!
That means for some reason the spectrometer is just not adhering to the protocol specified in the datasheet...

I need to fix the `seabreeze.pyseabreeze.features.fpga` submodule.
But in the meantime, you should be able to disable the assertion check (which was just added as a sanity check while implementing the functionality) by doing the following:

Run your python script that you're using to test the spectrometer not with
```console
python your-spectrometer-test-script.py
```

But with:
```console
python -O your-spectrometer-test-script.py
```
(That is a capital O for optimize, which ignores all assert statements in the python code at runtime)

waldowda, 2020-08-16:
I have been testing in jupyter notebooks and am not sure how to do that in anaconda/jupyter. I'll research that or I'll put the cells together as a  single .py and test from a python call. I think I can do that. I am still 'learning to really drive' anaconda. I won't likely get to that until tomorrow now.

I was able to go back to the plain version and added the line for nonlinearity. It then functions albeit without native acknowledgment of being a Flame-S but it 'appears' to function for now with dark correction, nonlinearity, and trigger mode 4.

You must have found out info about upgraded firmware. I could not find any info on that. Do you think I should ask OI for an update to the firmware? 

Will let you know what I find. Thanks!

ap--, 2020-08-17:
Hi Dean, could you please try with the newest version? `1.2.0.post11+something`

waldowda, 2020-08-17:
Sure... just got off a webinar. Maybe 15-30 minutes.

ap--, 2020-08-17:
No rush. Will only be able to have another look tomorrow evening.
Cheers 😴

ap--, 2020-08-18:
Awesome 🎉 🥳 
Thanks for your help and patience 🥇 😃 

Okay, I'll try to get a new version out on the weekend.
I'll close the two issues and will open one specifically for the timing issue.
But yes, this might be a spectrometer issue.

Cheers,
Andreas 😃

closed 2020-08-18 by ap--

waldowda, 2020-08-18:
Thanks Andreas! I'll look at the new issue and can maybe contribute the code I used with some tweaks. I am sure you can generate testing code that is better than mine. LOL. Best wishes. Dean