ap--/python-seabreeze #100: SeaBreezeError not caught while using pyseabreeze backend

Lavikjo, 2020-08-13:
### spectrometer and system information ###

- model:  HDX UV-VIS
- operating system: Windows 10 64bit
- python version:  Python 3.8.2
- python-seabreeze version:  current master
- installed-via: pip

### current problem ###

While using pyseabreeze backend the errors caused for example by no connection to device is not caught with SeaBreezeError imported from pyseabreeze `from seabreeze.pyseabreeze.exceptions import SeaBreezeError`.

However the error can be caught when SeaBreezeError is imported from cseabreeze backend `from seabreeze.cseabreeze import SeaBreezeError`

Seems like that changing the place where import of error is done doesn't affect the result.

### steps to reproduce ###

1. run code example with device unplugged

### minimal code example and error (very helpful if available) ###

```python
import seabreeze
from seabreeze.spectrometers import list_devices, Spectrometer
seabreeze.use('pyseabreeze')
from seabreeze.pyseabreeze.exceptions import SeaBreezeError

try:
    spec = Spectrometer.from_serial_number("HDX00767")
except SeaBreezeError:
    print("caught")

```

```
  File "C:\Users\joela\AppData\Local\Programs\Python\Python38\lib\site-packages\seabreeze\spectrometers.py", line 121, in from_serial_number
    raise cls._backend.SeaBreezeError(
seabreeze.cseabreeze._wrapper.SeaBreezeError: No device attached with serial number 'HDX00767'.

```

ap--, 2020-08-13:
Hi @Lavikjo 

please try:
```python

import seabreeze
seabreeze.use('pyseabreeze')
from seabreeze.spectrometers import list_devices, Spectrometer, SeaBreezeError
```

`seabreeze.use` needs to be called before `seabreeze.spectrometers` is imported. Because the latter triggers the whole backend import logic and will default to `cseabreeze` and fallback to `pyseabreeze` if not available. So if you want to specifically request, you need to call it before!

That being said. `seabreeze.use` should throw a big warning if called afterwards.

ap--, 2020-08-13:
I renamed the issue so that i know what i need to do.

Cheers,
Andreas 😃

Lavikjo, 2020-08-14:
Hi @ap-- that seems to do the trick! Warning would be beneficial to avoid these confusions in the future. :)

commit 08fe7217 (2020-08-23): [seabreeze] seabreeze.use warn if called after importing seabreeze.spectrometers

Close #100.
closed 2020-08-23 by ap--