ap--/python-seabreeze #104: intensities value from USB4000

buihoangkhang, 2020-09-08:
### spectrometer and system information ###

- model: USB4000
- operating system: Raspian GNU/Linux 9 (stretch)
Raspberry Pi 3B+
- python version: 3.6.1
- python-seabreeze version: 0.6.0
- installed-via: not sure, maybe pip

### current problem ###

The intensity values did not respond to actual spectrum values as detected in Ocean View.
It just gives some values above 1000 like in your demo.
Link: https://python-seabreeze.readthedocs.io/en/latest/quickstart.html
[image]

This is from my computer.
[image]

Which is not correct (I think)

This should be

[image]

### steps to reproduce ###

I also encountered this problem with other combination of Raspberry Pi models ( I have RPi 4) and Python version - numpy version.
I also tried on Windows computer - anaconda build.
Could you please recommend the best system you think for me? I cannot change my USB4000 but I can change RPi 3B+ or 4, or change to Windows.

This package means a lot to me, thank you so much.

### minimal code example and error (very helpful if available) ###

```python
import seabreeze
import seabreeze.spectrometers as sb
devices = sb.list_devices()
print(devices)
spec = sb.Spectrometer(devices[0])
spec.integration_time_micros(200)
x = spec.wavelengths()
y = spec.intensities()
print(y)

ap--, 2020-09-10:
Hello @buihoangkhang,

I'm not yet sure if there is a problem.
Which seabreeze version are you running?

```python
import seabreeze
print(seabreeze.__version__)
```

And please provide a plot of your measured spectrum 😊  Or the output of

```python
import numpy as np
import seabreeze.spectrometers as sb
spec = sb.Spectrometer.from_first_available()
spec.integration_time_micros(100000)

y = spec.intensities()
print(np.min(y), np.max(y), np.mean(y))
```

Cheers,
Andreas

buihoangkhang, 2020-09-11:
Hello, thank you for your reply,
I am using Seabreeze version 0.6.0
[image]

As you requested the output.
[image]

I also try with windows 10, anaconda with spyder.
This is the code.
[image]

and the result
[image]

I try again with plotting
[image]
and the result
[image]

This package means a lot to me.
Thank you again,
Bui Hoang Khang

ap--, 2020-09-11:
Hi @buihoangkhang 

What kind of lightsource are you using with your spectrometer?

For testing, I would suggest running your code again, but this time use a larger integration_time:
```python
spec.integration_time_micros(100000)
```
Note: the integration_time is in microseconds so, 100000 would still only be 0.1 seconds.

To me it looks as if everything is working and you are just using an intergration that is too small to show your signal.

Let me know how things go,
Cheers,
Andreas 😃

buihoangkhang, 2020-09-12:
Hi,
it works!
Thank you so much, sorry for my neglectance.
I will further testing it.
Bui Hoang Khang

ap--, 2020-09-12:
No worries 😊  
I'm happy we could solve your problem together.

I'm closing the issue, let me know if you run into other issues!

Cheers,
Andreas 😃

closed 2020-09-12 by ap--