ap--/python-seabreeze #107: Calibration

ColinBrosseauAlgolux, 2020-09-28:
### spectrometer ###

- model: _(i.e. STS, etc...)_

### Question ###

I intend to use spectra from a STS-Vis. This device is spectrally calibrated (in intensity). Where does the calibration is done? Inside the device? In OceanOptics's software?
This is important because I need to use the calibrated version of the spectrum and I then I want to know if python-seabreeze will meet my requirements.

Thanks

Colin Brosseau

ap--, 2020-09-28:
Hi @ColinBrosseauAlgolux 

I assume when you say that your device is callibrated you mean that an irradiance calibration is stored on your spectrometer?
`python-seabreeze`'s `cseabreeze` backend let's you read the calibration from the device. But you'll have to write the bit of code that applies the calibration to a measured raw spectrum yourself.
(If I recall correctly, it's just multipling or dividing by the calibration array, so it should be very easy.)

Cheers,
Andreas :smiley:

ColinBrosseauAlgolux, 2020-09-30:
Hi @ap-- 

Thanks for your answer. As long as I can get the calibration from the device it's OK.

Many thanks,
Colin Brosseau

closed 2020-09-30 by ColinBrosseauAlgolux