ap--/python-seabreeze #111: docs for installing python-seabreeze without cseabreeze need fixing

ap--, 2021-02-13:
Currently I recommend this:

```
# install only pyseabreeze backend
pip install seabreeze[pyseabreeze] --no-use-pep517 --no-build-isolation --install-option="--without-cseabreeze"
```

But because of https://github.com/pypa/pip/issues/1883 `--without-cseabreeze` gets passed down to other dependencies that have to be installed, which crashes for example with:

```
ERROR: Command errored out with exit status 1: /Users/poehlmann/Environments/envs/newest-pip/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/6w/qw4h_znn53zgxml9dzfk2kwc0000gn/T/pip-install-u2itrw4f/numpy_9d40d75b6b4e421190804fa90b3c2d53/setup.py'"'"'; __file__='"'"'/private/var/folders/6w/qw4h_znn53zgxml9dzfk2kwc0000gn/T/pip-install-u2itrw4f/numpy_9d40d75b6b4e421190804fa90b3c2d53/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/6w/qw4h_znn53zgxml9dzfk2kwc0000gn/T/pip-record-19bepf6q/install-record.txt --single-version-externally-managed --compile --install-headers /Users/poehlmann/Environments/envs/newest-pip/include/python3.9/numpy --without-cseabreeze Check the logs for full command output.
```

So the easy solution is to:

```
# install only pyseabreeze backend
pip install numpy future pyusb
pip install seabreeze[pyseabreeze] --no-use-pep517 --no-build-isolation --install-option="--without-cseabreeze"
```

But that's annoying... let's see if there's a way to not pass down install options...

drmcnelson, 2021-02-14:
That solution does not work for me.    The future pysub is already installed, and the error still occurs on the seabreeze install

The only way it installs is  
 pip install -I seabreeze[pyseabreeze]

After that, it fails to detect the ADC1000 USB/SD 2000.

ap--, 2021-02-14:
Hi @drmcnelson 

Could you post the output of `python --version`, `pip --version` and `pip freeze`

Cheers,
Andreas

drmcnelson, 2021-02-14:
python --version ; pip --version ; pip freeze
Python 3.8.7
pip 19.3.1 from /usr/lib/python3.8/site-packages/pip (python 3.8)
angler==0.0.15
argcomplete==1.10.0
asteval==0.9.19
autograd==1.3
beautifulsoup4==4.9.3
blivet==3.2.2
blivet-gui==2.1.15
Bottleneck==1.2.1
cached-property==1.5.1
cairocffi==1.1.0
CairoSVG==2.4.2
cffi==1.14.0
chardet==3.0.4
cssselect2==0.3.0
cupshelpers==1.0
cycler==0.10.0
dasbus==0.2
dbus-python==1.2.16
decorator==4.4.0
defusedxml==0.6.0
distro==1.4.0
future==0.18.2
gpg==1.14.0
humanize==0.5.1
idna==2.8
initial-setup==0.3.80
kiwisolver==1.3.1
langtable==0.0.53
libcomps==0.1.15
lmfit==1.0.1
lxml==4.4.1
matplotlib==3.2.2
meld==3.20.2
mock==3.0.5
nftables==0.1
notify2==0.3.1
ntplib==0.3.3
numexpr==2.7.1
numpy==1.20.1
olefile==0.46
onboard==1.4.1
ordered-set==3.1
pandas==0.25.3
pandas-datareader==0.8.0
pexpect==4.7.0
pid==2.2.3
Pillow==7.0.0
Pivy==0.6.5
ply==3.11
powerline-status==2.8
productmd==1.30
progressbar2==3.37.1
psutil==5.6.7
ptyprocess==0.6.0
pwquality==1.4.4
pycairo==1.18.2
pycollada==0.6
pycparser==2.19
pycups==2.0.1
pycurl==7.43.0.5
pydbus==0.6.0
pyenchant==2.0.0
pygit2==1.2.1
PyGObject==3.36.1
pyinotify==0.9.6
pykickstart==3.24
pyMKL==0.0.3
pyparsing==2.4.7
pyparted==3.11.4
pyserial==3.4
PySide2==5.15.0
PySocks==1.7.1
pystray==0.17.1
python-augeas==0.5.0
python-dateutil==2.8.0
python-gettext==4.0
python-manatools==0.0.3
python-meh==0.48
python-pam==1.8.4
python-utils==2.4.0
python-xapp==2.0.1
python-xlib==0.28
pytz==2020.5
pyudev==0.22.0
pyusb==1.1.1
pyxdg==0.26
PyYAML==5.3.1
requests==2.22.0
requests-file==1.4.3
requests-ftp==0.3.1
rpm==4.15.1
scipy==1.4.1
seabreeze==1.3.0
selinux==3.0
sepolicy==3.0
setools==4.3.0
setproctitle==1.1.10
setroubleshoot==1.1
shiboken2==5.15.0
simpleline==1.6
six==1.14.0
slip==0.6.4
slip.dbus==0.6.4
sos==3.9
soupsieve==1.9.2
systemd-python==234
tables==3.5.2
tinycss==0.3
tinycss2==1.0.2
tmm==0.1.8
uncertainties==3.1.4
urllib3==1.25.7
webencodings==0.5.1
xcffib==0.9.0
xlrd==1.2.0
xlwt==1.1.2

ap--, 2021-02-14:
Thanks for the detailed report ❤️ 
 
Interesting. I can't quite figure out why the suggested change to install command would not work for you, but the difference compared to my tests is the `pip` version. So maybe there's been a change in pip behavior from 19.3.1 to the newest (which would not be without precedent...) or fedora does some special patching of pip...

I'll have to investigate this further...

Cheers,
Andreas

ap--, 2021-09-20:
It seems I could recommend this option:
https://github.com/pypa/pip/issues/1883#issuecomment-919625949