ap--/python-seabreeze #112: ADC1000-USB with SD2000 not detected

drmcnelson, 2021-02-14:
### spectrometer and system information ###

- model:  ADC1000-USB, SD2000
- operating system:  Linux, Fedora 32, Cinnamon spin
- python version: Python 3.8.7
- python-seabreeze version:  1.3.0
- installed-via:  sudo pip install -I seabreeze[pyseabreeze]

### current problem ###
### steps to reproduce ###
### minimal code example and error (very helpful if available) ###

#```
!/usr/bin/python

import seabreeze
seabreeze.use('pyseabreeze')

from seabreeze.spectrometers import list_devices, Spectrometer

print( list_devices() )

spec = Spectrometer.from_first_available()
print(spec)
[
```

Output:
```

[]
Traceback (most recent call last):
  File "./SD200test.py", line 10, in <module>
    spec = Spectrometer.from_first_available()
  File "/usr/local/lib64/python3.8/site-packages/seabreeze/spectrometers.py", line 91, in from_first_available
    raise cls._backend.SeaBreezeError("No unopened device found.")
seabreeze.pyseabreeze.exceptions.SeaBreezeError: No unopened device found.

```

Output from lsusb

`ID 2457:1004 Ocean Optics Inc. 
`

```python
copy of minimal code example and error messages
```

ap--, 2021-02-14:
Hi @drmcnelson 

So I checked, and the ADC1000-USB is currently not supported by `python-seabreeze`.
I found an oem [datasheet](http://materiel-physique.ens-lyon.fr/Logiciels/CD%20N%C2%B0%201%20Logiciels%20ENS%20LYON%202018/Spectrasuite%20Oc%C3%A9an%20optics/documentation/AD%20Converters/ADC1000%20OEM%20Data%20Sheet.pdf) and the communication protocol is a bit different compared to the other spectrometers that are supported by `python-seabreeze`.

It would be possible to implement support, but it's a quite bit of work since it touches every part of pyseabreeze to be able to reuse the existing code.

But if you're interested to implement support yourself, I'd be happy to point you in the right direction and help if any immediate questions rise up.

Cheers,
Andreas

drmcnelson, 2021-02-14:
That looks helpful, I think the main hump is if it appears as a serial interface under /dev      After that it should be straightforward, though I am not clear about how the triggering works since it is purportedly not a gated chip.

drmcnelson, 2021-02-14:
Does pyseabreeze support the PCI ADC interfaces?  I have one of those also.

Aside, it might be of interest that the USB adapter seems to be pretty popular, a lot of the second hand units (i.e., affordable, c.f. ebay) are pre-usb,

ap--, 2021-02-14:
I assume the ADC-1000 USB predates a lot of the supported spectrometers, if you have a look at a datasheet for i.e. a USB2000 spectrometer, the command set is almost identical, just that the command codes differ.

If you look at https://github.com/ap--/python-seabreeze/blob/ff806a834019e2d5a548d2af629d01c90646fea8/src/seabreeze/pyseabreeze/protocol.py#L44 there's an implementation of the "OOIProtocol" which uses pyUSB to send the binary commands to the spectrometer. It looks like you would be able to pick the right transfers from the different implementations in https://github.com/ap--/python-seabreeze/tree/master/src/seabreeze/pyseabreeze/features and call them using the right opcodes for the ADC

Currently only USB devices are supported. There's some code in the C++ backend that implements RS232 communication for some of the spectrometers that support it, but non of that is exposed to the python wrapper. But the none of the PCI cards are supported by the backends. 

I'm happy to merge support for the ADC-1000 it into the codebase, if you're interested in working on a pull request for it.
I can also provide feedback for how to navigate the ~huge mess~ modules in `python-seabreeze` 😄

ap--, 2023-05-07:
this was added in #170.

closed 2023-05-07 by ap--