ap--/python-seabreeze #113: maya2000 driver issue

isikfurkan, 2021-02-17:
### spectrometer and system information ###

- model: _(maya2000)_
- operating system: _(Windows 10 64bit)_
- python version: _(python 3.9)_
- python-seabreeze version: _(i.e. 0.5.3 or 'current master')_
- installed-via: _(python setup.py install)_

### current problem ###

I could not install using seabreeze_os_seebreeze, when I tried, received :
'seabreeze_os_setup' is not recognized as an internal or external command,
operable program or batch file.

Then I found the Seebreeze_os_setup in the directory and installed it.
But it couldn't install the driver of maya2000.

[image]
[image]

I have no experience with python, so I might be doing things so stupidly. Ty for ur patience in advance :)

ap--, 2021-02-17:
Hi @isikfurkan 

you can try downloading the windows driver zip file from here https://github.com/ap--/python-seabreeze/tree/master/os_support
and install the driver yourself manually.

Let me know if that helps :blush: 

Cheers,
Andreas

isikfurkan, 2021-02-18:
Hello Andreas,

I could solve the problem by changing the settings on windows 10 which requires inf file to install drivers. I followed the second precedure mentioned in the link: [](https://www.urtech.ca/2016/01/solved-the-third-party-inf-does-not-contain-digital-signature-information/)

Ty for ur answer.

Cheers,
Furkan

closed 2021-02-18 by isikfurkan