ap--/python-seabreeze #114: Error upon installation with no consequences for me

LMSC-NTappy, 2021-02-26:
### spectrometer and system information ###

- model:  USB2000+
- operating system: Windows XP 64bit
- python version: 3.9.1
- python-seabreeze version: 1.3.0
- installed-via: conda

### current problem ###

Nothing wrong for me but I got an error upon '''seabreeze_os_setup''' execution, reporting invalid signatures for some of the drivers. 

Reporting it because 

### steps to reproduce ###

1. conda install -c conda-forge seabreeze
2. Restart the anaconda prompt several times, not understanding seabreeze_os_setup fails because it has to be run in admin mode (grating admin at execution time smh doesn't work). Then run anaconda prompt in admin mode
3. seabreeze_os_setup finally works but outputs a warning telling me I need to report error on github, hence here I am
4. Using seabreeze as documented! EPIC package man, love it!

### minimal code example and error (very helpful if available) ###

```python
Downloading windows drivers from github
Extracted to temporary directory C:\Users\Attomic\AppData\Local\Temp\seabreeze-os-kro8fz3m
Microsoft PnP Utility

Processing inf :            OOI_ADC1000USB.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_Apex.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_Apex_XP.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_FlameNIR.inf
Adding the driver package failed : The hash for the file is not present in the specified catalog file. The file is likely corrupt or the victim of tampering.

Processing inf :            OOI_FlameU.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_FlameX.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_FlameX_XP.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_Generic_Smartphone_Module.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_HR2000.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_HR2000Plus.inf
Successfully installed the driver on a device on the system.
Driver package added successfully.
Published name :            oem12.inf

Processing inf :            OOI_HR2000Plus_XP.inf
Successfully installed the driver on a device on the system.
Driver package added successfully.
Published name :            oem62.inf

Processing inf :            OOI_HR4000.inf
Successfully installed the driver on a device on the system.
Driver package added successfully.
Published name :            oem47.inf

[... 168 lines cut ...]
Successfully installed the driver on a device on the system.
Driver package added successfully.
Published name :            oem55.inf

Processing inf :            OOI_USB650_XP.inf
Successfully installed the driver on a device on the system.
Driver package added successfully.
Published name :            oem56.inf

Processing inf :            OOI_Ventana.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_Ventana_XP.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Total attempted:              53
Number successfully imported: 33

Note: Some of the drivers currently don't have valid signatures.
Look at the output above. If the spectrometer you want to use only
provides an unsigned driver, you might have to install it manually.
If you encounter this issue, please report it on github.
pnputil returned with 3758096943
Press [enter] to close.
```

ap--, 2021-02-26:
Hi @LMSC-NTappy,

Thank you for reporting ❤️ 
It's a bit funny, because this message has been in `python-seabreeze` for more than 2 years and you are the first to report 🥇 🎉 

The good thing is that everything worked for you out of the box.
Sadly, there's nothing we can do to get newly signed drivers so we could remove this issue. Unless OceanInsight would sign these drivers again.
But, maybe it's worth changing the message in the tool for the next version.

I'll rename the issue and keep it open as a reminder ☺️ 

Have a great day!
Cheers,
Andreas 😃

ap--, 2021-02-26:
Oh, but reading your message again:
> Restart the anaconda prompt several times, not understanding seabreeze_os_setup fails because it has to be run in admin mode (grating admin at execution time smh doesn't work). Then run anaconda prompt in admin mode

this is actually a bug... I have never tested this on 😮  Windows XP 😮 
I'll try testing it too in a vm and will add a note to the docs.

Cheers,
Andreas

LMSC-NTappy, 2021-02-27:
Hello Andreas,

Oops, reading your message i realized I wrote Windows XP when I meant windows 10. Probably tired on a friday afternoon. ^^ 

Sorry about that I hope you don't get busy because of this 

Have a nice week-end and thanks again for your work, I am a huge fan

Cheers

Nico

ap--, 2021-02-27:
Hi Nico,

ah okay. Hmm... still a bug then. I think I'll add some output to the script so that it's clear it needs to run in an admin shell.
Thanks again for reporting :smiley: 

Cheers,
Andreas