ap--/python-seabreeze #115: usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

Tb8854, 2021-05-09:
### spectrometer and system information ###

- model: HDX
- operating system: Windows 10 64 Bit
- python version: Python 3.9.1
- python-seabreeze version: current master
- installed-via: pip

### steps to reproduce ###

**C:\WINDOWS\system32>python**
Python 3.9.1 (default, Dec 11 2020, 09:29:25) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
**>>> import seabreeze**
**>>> seabreeze.use('pyseabreeze')**
**>>> from seabreeze.spectrometers import Spectrometer, list_devices**
**>>> list_devices()**
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\seabreeze\spectrometers.py", line 36, in list_devices
    return list_devices._api.list_devices()
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\seabreeze\pyseabreeze\api.py", line 104, in list_devices
    dev.open()
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\seabreeze\pyseabreeze\devices.py", line 346, in open
    self._transport.open_device(self._raw_device)
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\seabreeze\pyseabreeze\transport.py", line 201, in open_device
    if pyusb_device.is_kernel_driver_active(0):
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\usb\core.py", line 1097, in is_kernel_driver_active
    self._ctx.managed_open()
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\usb\core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\usb\backend\libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\usb\backend\libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "C:\Users\asty_admin\miniconda3\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

Any idea why this might be happening? Trying to connect to HDX. This is a fresh computer, and just setting out installing everything again.

Tb8854, 2021-05-09:
Almost there - now I get the strange behaviour of  

**>>> devices**
[<SeaBreezeDevice HDX:?>]

Which I'm not quite sure how to deal with...

Tb8854, 2021-05-10:
Resolved by reinstalling Anaconda and using spyder and messing with the paths.

closed 2021-05-10 by Tb8854

ap--, 2021-05-12:
Glad you could fix it.
Do you think the docs could be improved to prevent the environment setup issue you ran into?
If so, I'd be happy to get feedback on how to improve them :)

Cheers 😃 
Andreas