ap--/python-seabreeze #119: Getting temperature of FLAMES Spectrometer

joellama, 2021-06-22:
### spectrometer and system information ###

- model: USB2000PLUS
- operating system: Windows 10
- python version: 3.8.5
- python-seabreeze version: 1.3.0
- installed-via: pip install seabreeze

### current problem ###

I'm trying to get the temperature of my spectrograph. When I try 
`from seabreeze.spectrometers import Spectrometer
 spec = Spectrometer.from_serial_number()
 spec.features`

Temperature is listed as []. Running `spec.tec_get_temperature_C()` fails with a NoneType error (as expected). 

I tried to follow Issue 64 that talks about pyseabreeze; however, I cannot get that to work at all because I just a no pyusb backend error. 

So I'm at a bit of a loss, any help would be very much appreciated! 

### steps to reproduce ###

_Step by step instructions to reproduce the error. The more detailed the better, but please use some common sense:_

1. run code example
2. ...

### minimal code example and error (very helpful if available) ###

```python
copy of minimal code example and error messages
```

ap--, 2021-06-24:
Hi @joellama 

TEC stands for "thermo electric cooler" and the Flame-S doesn't support that feature.

As you already noticed, the code for reading temperatures on a USB2000Plus / Flame-S from #64 requires pyusb.

Are you sure you have pyusb installed? What happens if you run the following in your python interpreter:

```python
import usb.core
usb.core.find()
```

Cheers,
Andreas :smiley:

joellama, 2021-06-25:
Hi @ap-- ,

Thanks so much for writing back. After a bit of experimenting, I managed to get around the pyusb error by installing seabreeze with conda and not pip. 

I think my issue now is how to get the temperature using the `spec.f.raw_usb_bus_access.raw_usb_write` functions - issue #64 I think uses different protocols that don't seem to exist in the code anymore (though again I could be wrong). 

Any help would again be very much appreciated!

ap--, 2021-06-27:
Hi @joellama 

can you provide a complete code example?
Have you tried running the code in issue #64 as is? If so what's the error message?

Cheers,
Andreas

joellama, 2021-07-06:
Hi Andreas,

Thanks for writing back and apologies for my slow delay. I tried running the code in #64; however there I cannot get past `spec._dev.interface` because there is no interface option listed. The available options I see are close(), f, featrues, get_serial_number(), is_open, model, open(), and serial_number. 

Any thoughts?

ap--, 2021-07-06:
Hi @joellama 

I had a look, and you're right, it seems that code won't work anymore.
Try the snippet below. It's also untested (I dont have any spectrometer hardware with me right now to verify)

```python
import seabreeze
seabreeze.use('pyseabreeze')
import seabreeze.spectrometers as sb
import struct

# open first spectrometer
spec = sb.Spectrometer.from_serial_number()

# query PCB temperature
# https://oceanoptics.com/wp-content/uploads/FlameIO.pdf, page 79f
spec.f.raw_usb_bus_access.raw_usb_write(struct.pack('<B', 0x6C), endpoint='primary_out')
raw_bytes_response = spec.f.raw_usb_bus_access.raw_usb_read(endpoint='primary_in')
temp = 0.003906 * struct.unpack("<h", raw_bytes_response[1:3])[0]  # should be 0x08 LSB MSB 

print("PCB temperature", temp)
```

Let me know how things go :smiley: 

Cheers,
Andreas

joellama, 2021-07-06:
Thank you so much @ap-- , that worked a treat! Thank you for your hard work on this project, it will be very useful for me!

closed 2021-07-06 by joellama

VladislavSu, 2021-08-04:
Hello @ap-- !

I'm using FLAME-NIR Spectrometer.
Is it possible to get PCB temperature via RS232 ?
If yes how can I get it?

Thanks.

ap--, 2021-08-07:
Hello @VladislavSu,
I opened a new issue #128.  Let's continue discussion there 😃