ap--/python-seabreeze #122: Module not using sensitivity calibration

juliowissing, 2021-07-13:
### spectrometer and system information ###

- model: _USB4000:FLMT03472_
- operating system: _Windows 10 64bit_
- python version: _3.9.5_
- python-seabreeze version: _1.3.0_
- installed-via: _conda_

### current problem ###
When taking a spectrum with the above mentioned spectrometer, the output seems to be the raw adc values without the sensitivity calibration. Is there a possibility to get the corrected values or the calibration values to do a manual correction?

### steps to reproduce ###
1. Take spectrum with halogen lamp from Thor Labs
2. Compare spectrum read by python-seabreeze and ocean optics software
3. Spectrum read by python-seabreeze falls of at higher wavelengths
4. Spectrum read by ocean optics software shows right spectrum of halogon lamp

### minimal code example and error (very helpful if available) ###

    integration_time_micros = 100000  # µs

    # Connecting to spectrometer and setting integration time
    spec = Spectrometer.from_first_available()
    spec.integration_time_micros(integration_time_micros)

    # Acquiring spectrum
    spectrum = spec.spectrum()

    # Closing connection to Spectrometer
    spec.close()

    # Plotting Spectrum
    plt.plot(spectrum[0], spectrum[1])
    plt.show()

ap--, 2021-07-13:
Hi @juliowissing 

yes, python-seabreeze returns the raw readings, and has support for dark_count and nonlinearity correction.

What are you referring to specifically when you mention `sensitivity calibration`?
Did you store calibration coefficients on you spectrometer using the ocean optics software?

Can you show the plot comparing the two measurements?

Cheers,
Andreas 😃

juliowissing, 2021-07-14:
Hi Andreas,

thanks for the reply. The Flame Spectrometer is currently in use, but we did some testing with our maya spectrometer and it shows the same behavior. In both cases the dark_count and nonlinearity corrections are turned off.

Spectrum obatained with Ocean Optics Software 
[image]

Spectrum obtained with python seabreeze
[image]

As you can see in the case of the seabreeze version, the intensities decrease after ~800nm, which should not be the case for the spectrum of a halogen lamp. The optics expert at my group stated that this behaviour might be caused by not applying a sensitvity correction in regards to the Si Photodiods. This is usually done after the ADC readout, but before the end user gets the spectrum data.
Thank you for your support!

Best regards,
Julio

ap--, 2021-07-21:
Hi Julio,
sorry for the delay, it's been a busy week.
I think you're actually taking an irradiance measurement in the oceanoptics software.
There should be a way to take a raw measurement as provided by python-seabreeze too.

Can you checkout this manual https://mmrc.caltech.edu/Biologic%20Stopped%20Flow/Manuals/Ocean%20Optics/SpectraSuite.pdf
and describe in detail how you are taking these measurements?

Cheers,
Andreas