ap--/python-seabreeze #126: Flame-S Edge Trigger Mode Didn't get into 1.3.0?

waldowda, 2021-07-25:
### spectrometer and system information ###

- model: Flame-S
- operating system: Win10 64bit (or MacOS 11.4)
- python version: 3.8.10
- python-seabreeze version: 1.3.0
- installed-via: conda

### current problem ###

We addressed (fixed?) the edge trigger issue back with closed issue #97 but that fixed didn't propagate into 1.3.0 as far as I can tell. Or it was deemed incorrect? I have gone back into devices.py and redid the changes from #97 and it appears to now be triggered by my potentiostat trigger pulse.

### steps to reproduce ###

Follow steps to get Flame-S talking to py-seabreeze and then change trigger mode via:

```
spec.trigger_mode(4)
```

and it fails / errors but modes 0, 1, 2, and 3 all are able to be set. After the change below and as referenced in #97 I can set to '4' and it is triggered apparently properly from my potentiostat as far as I can tell. 

Is this the proper way to fix it? There was an EDGE at 0x03 but from my info it is supposed to be 0x04. Am I wrong?

### minimal code example and error (very helpful if available) ###
 
```devices.py

class TriggerMode(enum.IntEnum):
    """internal trigger modes enum"""

    NORMAL = 0x00
    SOFTWARE = 0x01
    LEVEL = 0x01
    SYNCHRONIZATION = 0x02
    HARDWARE = 0x03
#    EDGE = 0x03
    EDGE = 0x04 
    SINGLE_SHOT = 0x04
...
...
```

I don't know if I should commented out SINGLE_SHOT. I didn't know what EDGE was at 0x03 as well as HARDWARE was as well? Maybe it is used in the class definition. 

And...

``` devices.py

class FLAMES(USB2000PLUS):

    model_name = "FLAMES"

    # spectrometer config
    dark_pixel_indices = DarkPixelIndices.from_ranges((6, 21))  # as in seabreeze-3.0.9
    integration_time_min = 1000
    integration_time_max = 655350000
    integration_time_base = 1
    spectrum_num_pixel = 2048
    spectrum_raw_length = (2048 * 2) + 1
    spectrum_max_value = 65535
    trigger_modes = TriggerMode.supported(
#        "NORMAL", "SOFTWARE", "SYNCHRONIZATION", "HARDWARE", "SINGLE_SHOT"
        "NORMAL", "SOFTWARE", "SYNCHRONIZATION", "HARDWARE", "EDGE"
    )
...
...
```

ap--, 2021-07-28:
Hi @waldowda 

hmmm... by changing the TriggerMode.EDGE enum to 0x04 and switching from "SINGLE_SHOT" to "EDGE" you basically didn't change anything.

```
trigger_modes = TriggerMode.supported(
    "NORMAL", "SOFTWARE", "SYNCHRONIZATION", "HARDWARE", "SINGLE_SHOT"
)
# is effectively equivalent to
{0, 1, 2, 3, 4}
```

And after your change it's the same. Are there other changes in your code base?

Cheers,
Andreas 😃

waldowda, 2021-07-28:
Hum. Not sure. I will go back and see if j ok changed anything else. All I know is that tigger mode 4 erred until I made that change. I changed it both in USB2000plus and FlameS. I can reinstall 1.3.0 and try to reproduce the error? Thanks.