ap--/python-seabreeze #127: data is just noise on usb4000

chesse20, 2021-07-28:
### spectrometer and system information ###

usb4000 windows 10 pro

### current problem ###

my data looks like noise and not like a spectrometer curve and it doesn't look like anything close to what it looks like on loggerlite

### steps to reproduce ###

run the code provided on usb 4000 on windows 10 pro and measure something that isnt water

### minimal code example and error (very helpful if available) ###

copy of minimal code example and error messages
[image]
[image]

Kampfkarren, 2021-07-28:
fix this NAO

CoffeeDragon16, 2021-07-28:
I'm haming this issue too.

ap--, 2021-07-31:
Hi @chesse20,

try doubling the intergration time until you get a signal. You have it currently set to `20000 micro seconds`.
Also you mentioned `measuring something that isn't water`. Does that mean under other conditions it does work?

Cheers,
Andreas

chesse20, 2021-08-02:
changing the integration time(from 20k to 190k) and changing the liquid from tea to water to tea doesn't seem to change the output from anything but noise