ap--/python-seabreeze #129: FLAMES sporadic data transfer error

joellama, 2021-08-11:
### spectrometer and system information ###
    model: USB2000PLUS
    operating system: Ubuntu 20.10
    python version: 3.8.5
    python-seabreeze version: 1.3.0
    installed-via: pip install seabreeze

### current problem ###

I have a script that connects to 3 spectrometers and samples them every second to record the intensities. The problem is that everything works for a given amount of time (sometimes tens of minutes, sometimes hours) and then the code just hangs. On Ctrl+C I get a data transfer error. 

 
### minimal code example and error (very helpful if available) ###

```python
from seabreeze.spectrometers import Spectrometer, list_devices
import time

def get_temperature(spec):
    spec.f.raw_usb_bus_access.raw_usb_write(
        struct.pack('<B', 0x6C), endpoint='primary_out')
    raw_bytes_response = spec.f.raw_usb_bus_access.raw_usb_read(
        endpoint='primary_in')
    temp = 0.003906 * float(struct.unpack("<h", raw_bytes_response[1:3])[0])
    return temp

if __name__ == '__main__':
    blue = Spectrometer.from_serial_number('FLMS18610')
    green = Spectrometer.from_serial_number('FLMS18611')
    red = Spectrometer.from_serial_number('FLMT05929')
    blue_exp_time = 380000
    green_exp_time = 80000
    red_exp_time = 500000
    blue.integration_time_micros(int(blue_exp_time))
    green.integration_time_micros(int(green_exp_time))
    red.integration_time_micros(int(red_exp_time))
    while True:
        blue_wavelengths = blue.wavelengths()
        blue_intensity = blue.intensities(correct_dark_counts=True,
                                          correct_nonlinearity=True)
        blue_temperature = get_temperature(blue)
        green_wavelengths = green.wavelengths()
        green_intensity = green.intensities(correct_dark_counts=True,
                                            correct_nonlinearity=True)
        green_temperature = get_temperature(green)
        red_wavelengths = red.wavelengths()
        red_intensity = red.intensities(correct_dark_counts=True,
                                        correct_nonlinearity=True)
        red_temperature = get_temperature(red)
        # Save the file redacted for brevity
        time.sleep(1)
```

ap--, 2021-08-11:
Hi @joellama,

what does
```console
cat /sys/module/usbcore/parameters/autosuspend
```
return on your ubuntu 20.10 machine?

Cheers,
Andreas 😃

joellama, 2021-08-11:
Uhoh, I really hope it's not something stupid like that! It says 2?

ap--, 2021-08-11:
You mentioned it's sporadic, so my guess would be some weird USB problem...
On linux PCs that are used for scientific measurements I'd generally recommend disabling everything related to power management.

I think usb autosuspend is disabled by adding: `usbcore.autosuspend=-1` as a grub boot parameter.
You should be able to verify the setting with the `cat` command from above.

I hope that helps. If not we'll have to dig a little deeper and try to create a reproducible error case.

joellama, 2021-08-12:
Thanks for this @ap--. I've swapped out the USB cables just to be safe and have set `usbcore.autosuspend=-1`, I'll let it run for today and tomorrow and see how it goes!

joellama, 2021-08-18:
Hi @ap--, sadly this didn't help and I'm still getting random dropouts and the code hanging. I'm now trying the windows softare that comes with the spectrographs to see if that can handle running for 12 hours straight.

ap--, 2021-08-22:
Hi Joel,

Another thing worth trying would be to run the 3 spectrometers in three different processes. (You could test by running a script continuously with only one spectrometer.)

Did your tests with the default software run for more than 12h?