ap--/python-seabreeze #133: Unable to connect to the HDX Spectrometer

stepandr47, 2021-09-20:
### spectrometer and system information ###

- model: HDX Spectrometer
- operating system: Ubuntu
- python version: python3.7
- python-seabreeze version: current
- installed-via: pip3 install seabreeze

### current problem ###

I am unable to connect to the HDX spectrometer. When I run list_devices() it displays:
[image]

This is the correct serial number for my device. I then have tried to connect to the device with from_serial_number(HDX01168) or from_first_available(). Both of these will give the below output on the first attempt:
[image]

Looking through the tickets I noticed that HDX support has been added, so I am unsure what I am doing incorrectly here.

### steps to reproduce ###

Run below code with the HDX connected:

1. import seabreeze
2. seabreeze.use('pyseabreeze')
3. from seabreeze.spectrometer import list_devices, Spectrometer
4. list_devices()
5. spectrometer = Spectrometer.from_first_available()

It seems like other users have gotten functionality working for an HDX spectrometer so I was wondering what I am doing incorrectly in the above implmentation.

ap--, 2021-09-20:
Hi @stepandr47 

Did you install the udev rules? If not please run `seabreeze_os_setup`.
Also try running the code without calling `list_devices()` first.

If that didn't solve it, please verify that the spectrometer is working by running it on a different machine with the original spectrasuite or oceanview software.

Cheers,
Andreas :smiley:

stepandr47, 2021-09-21:
Hi Andreas,

I have tried running it without list_devices() and I get the same result.

Below is the rule created by seabreeze_os_setup for my sensor:
[image]

I have verified functionality using the OceanView software with no issue. Please let me know if there is any other information that would be useful for you.

Thanks for the quick reply.

ap--, 2021-09-21:
Which version of `pyusb` are your running? 

```python
import usb; print(usb.__version__)
```

stepandr47, 2021-09-21:
Based on the image below it is version number 1.2.1
[image]

ap--, 2021-09-21:
So from your other issue, it seems that you have a flame-t running reliably?
Is your ubuntu machine a raspberry pi? and if so does the flame-t work reliably with the pi?
If the Flame-T works well on the same machine:

You could try testing a few different versions of pyusb: 1.1.0, 1.2.0, 1.2.1

If that doesn't work we'll have to figure out what the important difference is between the setup in https://github.com/ap--/python-seabreeze/pull/98 and yours.
So if you have access to a windows machine I would try running the hdx together with pyseabreeze there.

I can't really provide any hands-on help with debugging because I don't have an HDX...

stepandr47, 2021-09-21:
Yeah that is correct. I am able to do everything fine with the Flame-T. I have a test setup on a Ubuntu VM and a raspberry pi. The Flame-T works with both setups, but I can't get the HDX to work on either. I don't have the HDX with me today, so I will get back to you tomorrow on testing different versions of pyusb.

ap--, 2021-09-21:
Okay great. That's valuable information 👍 
So we can at least rule out some strange installation issues.

The pyusb versions are a longshot, but it's a super quick test.

Another longshot would be:
(packet captures for the OceanFX seem to indicate that the oceanview software uses protocol version 0x0000 to trigger the default OBP protocol of the spectrometer, and since the OceanHDX is also a OBP protocol spectrometer, maybe this will have an effect.)
```python
# run this before any other code
from seabreeze.pyseabreeze.protocol import OBPProtocol
OBPProtocol.OBP.HEADER_PROTOCOL_VERSION = 0x0000
```

stepandr47, 2021-09-22:
I tried the different versions of pyusb on Ubuntu. I tried each of them with and without the OBPProtocol addition stated in your last comment. It was unable to connect in all case.

I just now tried connecting to the device on Windows and it worked without any issues. I was also able to get data from it.

Any ideas what could be causing the issue on Ubuntu?

ap--, 2021-09-22:
Okay. This is great. At least we know now that it's some difference between the OSes that's unrelated to the pyusb version.

Hmmm :thinking: 
- So it could be that it's a difference between libusb handling and winusb.
- It could also be that the pyseabreeze code for opening the spectrometer doesn't work with the HDX on linux.

A simple thing I would try is to setup a virtualenv with seabreeze **as root** on the linux machine and run your test code as root. That rules out that it's some sort of permission issue.

Also try running your script via:
```
LIBUSB_DEBUG=4 PYUSB_DEBUG=debug python your_code.py
```
and maybe compare the debug output between linux and windows.

I'm pretty sure we'll be able to figure out what's causing this :+1:

stepandr47, 2021-09-22:
Running the script as root on Ubuntu got the same timeout error. I haven't looked through it yet but below are the outputs from linux. LIBUSB_DEBUG and PYUSB_DEBUG are not recognized on windows. During a quick search I was unable to find how to resolve this issue. One other thing worth mentioning is on windows I was only able to get it working when in a conda environment. I didn't realize my terminal was in one in the morning when I tested, but when I just tested now it won't work without being in the conda environment. Do to that I tried running the ubuntu setup in a conda environment, but it still gets the timed out error.

Linux Output:
2021-09-22 16:14:47,735 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 2d6a360 at 0x7fed8675ebe0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000007] [0000af45] libusb: debug [libusb_init] created default context
[ 0.000071] [0000af45] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000107] [0000af45] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000172] [0000af45] libusb: debug [get_kernel_version] reported kernel version is 5.11.0-34-generic
[ 0.000223] [0000af45] libusb: debug [op_init] bulk continuation flag supported
[ 0.000270] [0000af45] libusb: debug [op_init] zero length packet flag supported
[ 0.000316] [0000af45] libusb: debug [op_init] max iso packet length is (likely) 49152 bytes
[ 0.000342] [0000af45] libusb: debug [op_init] sysfs can relate devices
[ 0.000373] [0000af45] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000793] [0000af49] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.003652] [0000af45] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.003708] [0000af45] libusb: debug [linux_get_device_address] scan usb2
[ 0.003774] [0000af45] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.003789] [0000af45] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.003802] [0000af45] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.003943] [0000af45] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.003961] [0000af45] libusb: debug [linux_get_device_address] scan 2-1
[ 0.004011] [0000af45] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.004026] [0000af45] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.004038] [0000af45] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.004085] [0000af45] libusb: debug [linux_get_parent_info] Dev 0x2bf4b40 (2-1) has parent 0x2bf1b60 (usb2) port 1
[ 0.004223] [0000af45] libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
[ 0.004242] [0000af45] libusb: debug [linux_get_device_address] scan 2-2
[ 0.004293] [0000af45] libusb: debug [linux_get_device_address] bus=2 dev=3
[ 0.004308] [0000af45] libusb: debug [linux_enumerate_device] busnum 2 devaddr 3 session_id 515
[ 0.004320] [0000af45] libusb: debug [linux_enumerate_device] allocating new device for 2/3 (session 515)
[ 0.004368] [0000af45] libusb: debug [linux_get_parent_info] Dev 0x2818ea0 (2-2) has parent 0x2bf1b60 (usb2) port 2
[ 0.004484] [0000af45] libusb: debug [linux_get_device_address] getting address for device: 2-2.1 detached: 0
[ 0.004502] [0000af45] libusb: debug [linux_get_device_address] scan 2-2.1
[ 0.004559] [0000af45] libusb: debug [linux_get_device_address] bus=2 dev=11
[ 0.004575] [0000af45] libusb: debug [linux_enumerate_device] busnum 2 devaddr 11 session_id 523
[ 0.004587] [0000af45] libusb: debug [linux_enumerate_device] allocating new device for 2/11 (session 523)
[ 0.004634] [0000af45] libusb: debug [linux_get_parent_info] Dev 0x28f5fc0 (2-2.1) has parent 0x2818ea0 (2-2) port 1
[ 0.004738] [0000af45] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004755] [0000af45] libusb: debug [linux_get_device_address] scan usb1
[ 0.004807] [0000af45] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004822] [0000af45] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004834] [0000af45] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004968] [0000af45] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.004984] [0000af45] libusb: debug [linux_get_device_address] scan 1-1
[ 0.005036] [0000af45] libusb: debug [linux_get_device_address] bus=1 dev=19
[ 0.005051] [0000af45] libusb: debug [linux_enumerate_device] busnum 1 devaddr 19 session_id 275
[ 0.005063] [0000af45] libusb: debug [linux_enumerate_device] allocating new device for 1/19 (session 275)
[ 0.005110] [0000af45] libusb: debug [linux_get_parent_info] Dev 0x29e2b00 (1-1) has parent 0x298c4f0 (usb1) port 1
[ 0.005237] [0000af45] libusb: debug [usbi_add_pollfd] add fd 6 events 1
[ 0.005265] [0000af45] libusb: debug [usbi_io_init] using timerfd for timeouts
[ 0.005278] [0000af45] libusb: debug [usbi_add_pollfd] add fd 8 events 1
2021-09-22 16:14:47,741 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2021-09-22 16:14:47,741 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.005871] [0000af45] libusb: debug [libusb_get_device_list] 
[... cut ...]
[ 2.663519] [0000af45] libusb: debug [libusb_unref_device] destroy device 2.11
[ 2.663539] [0000af45] libusb: debug [libusb_unref_device] destroy device 2.3
[ 2.663545] [0000af45] libusb: debug [libusb_unref_device] destroy device 2.2
[ 2.663565] [0000af45] libusb: debug [libusb_unref_device] destroy device 2.1
[ 2.663571] [0000af45] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 2.663618] [0000af45] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 2.663671] [0000af49] libusb: debug [linux_udev_event_thread_main] udev event thread exiting

stepandr47, 2021-09-22:
Linux Output within conda environment:
2021-09-22 16:55:55,639 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL '/home/ubuntu/anaconda3/envs/seabreeze-test/lib/libusb-1.0.so.0', handle 558ea0fe90d0 at 0x7f27a4e62a00>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000138] [0000b426] libusb: debug [libusb_init] created default context
[ 0.000211] [0000b426] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000277] [0000b426] libusb: debug [get_kernel_version] reported kernel version is 5.11.0-34-generic
[ 0.000367] [0000b426] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000432] [0000b426] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000499] [0000b426] libusb: debug [op_init] sysfs is available
[ 0.000970] [0000b429] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.003774] [0000b426] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.003894] [0000b426] libusb: debug [linux_get_device_address] scan usb2
[ 0.004057] [0000b426] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.004128] [0000b426] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.004196] [0000b426] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.004386] [0000b426] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.004462] [0000b426] libusb: debug [linux_get_device_address] scan 2-1
[ 0.004562] [0000b426] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.004632] [0000b426] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.004699] [0000b426] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.004805] [0000b426] libusb: debug [linux_get_parent_info] dev 0x558ea0eeb420 (2-1) has parent 0x558ea0e576d0 (usb2) port 1
[ 0.004993] [0000b426] libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
[ 0.005023] [0000b426] libusb: debug [linux_get_device_address] scan 2-2
[ 0.005095] [0000b426] libusb: debug [linux_get_device_address] bus=2 dev=3
[ 0.005113] [0000b426] libusb: debug [linux_enumerate_device] busnum 2 devaddr 3 session_id 515
[ 0.005124] [0000b426] libusb: debug [linux_enumerate_device] allocating new device for 2/3 (session 515)
[ 0.005168] [0000b426] libusb: debug [linux_get_parent_info] dev 0x558ea0dd58f0 (2-2) has parent 0x558ea0e576d0 (usb2) port 2
[ 0.005286] [0000b426] libusb: debug [linux_get_device_address] getting address for device: 2-2.1 detached: 0
[ 0.005302] [0000b426] libusb: debug [linux_get_device_address] scan 2-2.1
[ 0.005344] [0000b426] libusb: debug [linux_get_device_address] bus=2 dev=12
[ 0.005357] [0000b426] libusb: debug [linux_enumerate_device] busnum 2 devaddr 12 session_id 524
[ 0.005369] [0000b426] libusb: debug [linux_enumerate_device] allocating new device for 2/12 (session 524)
[ 0.005413] [0000b426] libusb: debug [linux_get_parent_info] dev 0x558ea0e2ec80 (2-2.1) has parent 0x558ea0dd58f0 (2-2) port 1
[ 0.005509] [0000b426] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.005524] [0000b426] libusb: debug [linux_get_device_address] scan usb1
[ 0.005565] [0000b426] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.005578] [0000b426] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.005589] [0000b426] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.005713] [0000b426] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.005730] [0000b426] libusb: debug [linux_get_device_address] scan 1-1
[ 0.005772] [0000b426] libusb: debug [linux_get_device_address] bus=1 dev=24
[ 0.005785] [0000b426] libusb: debug [linux_enumerate_device] busnum 1 devaddr 24 session_id 280
[ 0.005796] [0000b426] libusb: debug [linux_enumerate_device] allocating new device for 1/24 (session 280)
[ 0.005843] [0000b426] libusb: debug [linux_get_parent_info] dev 0x558ea0abb850 (1-1) has parent 0x558ea0f90a90 (usb1) port 1
[ 0.005948] [0000b426] libusb: debug [usbi_add_event_source] add fd 6 events 1
[ 0.005971] [0000b426] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.005982] [0000b426] libusb: debug [usbi_add_event_source] add fd 7 events 1
2021-09-22 16:55:55,646 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2021-09-22 16:55:55,646 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.006561] [0000b426] libusb: debug [libusb_get_device_list]  
2021-09-22 16:55:55,646 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7f27995a9550>)
[ 0.006788] [0000b426] libusb: debug [libusb_get_device_descriptor]  
2021-09-22 16:55:55,646 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x7f27995a9550>)
[ 0.007116] [0000b426] libusb: debug [libusb_open] open 1.24
[ 0.007159] [0000b426] libusb: debug [usbi_add_event_source] add fd 8 events 4
2021-09-22 16:55:55,647 DEBUG:usb.backend.libusb1:_LibUSB.reset_device(<usb.backend.libusb1._DeviceHandle object at 0x7f27995a9730>)
[ 0.007298] [0000b426] libusb: debug [libusb_reset_device]  
2021-09-22 16:55:56,121 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x7f27995a9730>)
[... cut ...]
[ 2.507436] [0000b426] libusb: debug [libusb_unref_device] destroy device 2.12
[ 2.507448] [0000b426] libusb: debug [libusb_unref_device] destroy device 2.3
[ 2.507460] [0000b426] libusb: debug [libusb_unref_device] destroy device 2.2
[ 2.507472] [0000b426] libusb: debug [libusb_unref_device] destroy device 2.1
[ 2.507484] [0000b426] libusb: debug [usbi_remove_event_source] remove fd 7
[ 2.507526] [0000b426] libusb: debug [usbi_remove_event_source] remove fd 6
[ 2.507658] [0000b429] libusb: debug [linux_udev_event_thread_main] udev event thread exiting

ap--, 2021-09-23:
To get the same output for windows, you need to set them as environment variables:
```
set LIBUSB_DEBUG=4
set PYUSB_DEBUG=debug
python your_code.py
```
(I think that's the right way for cmd.exe, powershell might be different)

stepandr47, 2021-09-23:
Was able to get these debugs working by adding these lines to the start of the script in windows:
import os
os.environ['PYUSB_DEBUG'] = 'debug'
os.environ['LIBUSB_DEBUG'] = '4'

Windows Output:
'LIBUSB_DEBUG' is not recognized as an internal or external command,
operable program or batch file.

(seabreeze-test) C:\Users\astephen\Documents>python Seabreeze_test.py
Traceback (most recent call last):
  File "C:\Users\astephen\Documents\Seabreeze_test.py", line 6, in <module>
    os.environ['LIBUSB_DEBUG'] = 4
  File "C:\Users\astephen\Anaconda3\envs\seabreeze-test\lib\os.py", line 684, in __setitem__
    value = self.encodevalue(value)
  File "C:\Users\astephen\Anaconda3\envs\seabreeze-test\lib\os.py", line 742, in check_str
    raise TypeError("str expected, not %s" % type(value).__name__)
TypeError: str expected, not int

(seabreeze-test) C:\Users\astephen\Documents>python Seabreeze_test.py
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000376] [00001914] libusb: debug [libusb_init] created default context
[ 0.000419] [00001914] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000512] [00001914] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.000587] [00001914] libusb: debug [htab_create] using 1021 entries hash table
[ 0.001231] [00001914] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.001381] [00001914] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.002610] [00001914] libusb: info [windows_init] UsbDk backend is not available
[ 0.002817] [00001914] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000118 events 0
[ 0.002882] [00001914] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.002936] [00001914] libusb: debug [usbi_add_event_source] add HANDLE 000000000000031C events 0
[ 0.003101] [00001914] libusb: debug [libusb_get_device_list]
[ 0.003447] [000005e0] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.015397] [00001914] libusb: debug [winusb_get_device_list] allocating new device for session [51]
[ 0.015614] [00001914] libusb: debug [winusb_get_device_list] allocating new device for session [52]
[ 0.015798] [00001914] libusb: debug [winusb_get_device_list] allocating new device for session [53]
[ 0.015976] [00001914] libusb: debug [winusb_get_device_list] allocating new device for session [54]
[ 0.016420] [00001914] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.016595] [00001914] libusb: debug [winusb_get_device_list] allocating new device for session [7]
[ 0.017077] [00001914] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.017285] [00001914] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.017365] [00001914] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.017430] [00001914] libusb: debug [winusb_get_device_list] allocating new device for session [1E]
[ 0.017939] [00001914] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.018248] [00001914] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.018645] [00001914] libusb: debug [winusb_get_device_list] allocating new device for session [39]
[ 0.019525] [00001914] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1C26&SUBSYS_21CF17AA&REV_04\3&B1BFB68&0&E8' bus number 1
[ 0.019740] [00001914] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1C2D&SUBSYS_21CF17AA&REV_04\3&B1BFB68&0&D0' bus number 2
[ 0.021423] [00001914] libusb: debug [winusb_get_device_list] found existing device for session [52]
[ 0.118831] [00001914] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.119842] [00001914] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.123963] [00001914] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_8087&PID_0024\5&2DF2E661&0&1'
[ 0.129739] [00001914] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&35D3A64E&0' reports 3 ports
[ 0.130799] [00001914] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&35D3A64E&0'
[ 0.133403] [00001914] libusb: debug [winusb_get_device_list] found existing device for session [7]
[ 0.135772] [00001914] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.144541] [00001914] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 216 bytes)
[ 0.146628] [00001914] libusb: debug [init_device] (bus: 2, addr: 2, depth: 2, port: 4): 'USB\VID_0A5C&PID_217F\60D819AF2C20'
[ 0.149292] [00001914] libusb: debug [winusb_get_device_list] found existing device for session [1E]
[ 0.151177] [00001914] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.163520] [00001914] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 93 bytes)
[ 0.163931] [00001914] libusb: debug [init_device] (bus: 1, addr: 2, depth: 2, port: 5): 'USB\VID_17EF&PID_1003\6&37ACA2D4&0&5'
[ 0.167231] [00001914] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&3423CC03&0' reports 3 ports
[ 0.172506] [00001914] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&3423CC03&0'
[ 0.175209] [00001914] libusb: debug [winusb_get_device_list] found existing device for session [51]
[... cut ...]
[ 0.824540] [00001914] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.824992] [00001914] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.825464] [00001914] libusb: debug [libusb_exit]
[ 0.825872] [00001914] libusb: debug [libusb_exit] destroying default context
[ 0.826475] [00001914] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000031C
[ 0.830112] [00001914] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000118
[ 0.830770] [000005e0] libusb: debug [windows_iocp_thread] I/O completion thread exiting

ap--, 2021-09-23:
Hi @stepandr47 

hmm, I was hoping we'd see that the libusb version would differ between the two. As it is, it is probably a bit hard to compare. (the linux output you posted seems to be from an interactive session and the windows output lacks the pyusb debug info)

Could you save the following script on both your windows and linux machine and run it again and copy the full output of the script to this issue:

```python
# test_open_hdx.py
import os
os.environ['LIBUSB_DEBUG'] = "4"
os.environ['PYUSB_DEBUG'] = "debug"

import seabreeze
seabreeze.use('pyseabreeze')
from seabreeze.spectrometers import Spectrometer

print("=== SPECTROMETER OPEN ===")
spec = Spectrometer.from_first_available()
print("=== SPECTROMETER STOP ===")
print(spec)

```

the following command should work in your linux shell AND in the windows cmd prompt:
```
python test_open_hdx.py > output.txt 2>&1
```
It should write all output to the `output.txt` file.

stepandr47, 2021-09-23:
Linux Output:
2021-09-23 14:43:24,769 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL '/home/ubuntu/anaconda3/envs/seabreeze-test/lib/libusb-1.0.so.0', handle 56411679e4c0 at 0x7fed8eccbdc0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000023] [0000e75a] libusb: debug [libusb_init] created default context
[ 0.000041] [0000e75a] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000054] [0000e75a] libusb: debug [get_kernel_version] reported kernel version is 5.11.0-34-generic
[ 0.000087] [0000e75a] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000097] [0000e75a] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000109] [0000e75a] libusb: debug [op_init] sysfs is available
[ 0.000523] [0000e75e] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.003073] [0000e75a] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.003109] [0000e75a] libusb: debug [linux_get_device_address] scan usb2
[ 0.003155] [0000e75a] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.003164] [0000e75a] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.003172] [0000e75a] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.003317] [0000e75a] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.003328] [0000e75a] libusb: debug [linux_get_device_address] scan 2-1
[ 0.003364] [0000e75a] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.003373] [0000e75a] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.003380] [0000e75a] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.003417] [0000e75a] libusb: debug [linux_get_parent_info] dev 0x56411658ae10 (2-1) has parent 0x5641165ca140 (usb2) port 1
[ 0.003508] [0000e75a] libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
[ 0.003519] [0000e75a] libusb: debug [linux_get_device_address] scan 2-2
[ 0.003554] [0000e75a] libusb: debug [linux_get_device_address] bus=2 dev=3
[ 0.003563] [0000e75a] libusb: debug [linux_enumerate_device] busnum 2 devaddr 3 session_id 515
[ 0.003570] [0000e75a] libusb: debug [linux_enumerate_device] allocating new device for 2/3 (session 515)
[ 0.003611] [0000e75a] libusb: debug [linux_get_parent_info] dev 0x56411631bb20 (2-2) has parent 0x5641165ca140 (usb2) port 2
[ 0.003710] [0000e75a] libusb: debug [linux_get_device_address] getting address for device: 2-2.1 detached: 0
[ 0.003720] [0000e75a] libusb: debug [linux_get_device_address] scan 2-2.1
[ 0.003757] [0000e75a] libusb: debug [linux_get_device_address] bus=2 dev=13
[ 0.003766] [0000e75a] libusb: debug [linux_enumerate_device] busnum 2 devaddr 13 session_id 525
[ 0.003773] [0000e75a] libusb: debug [linux_enumerate_device] allocating new device for 2/13 (session 525)
[ 0.003813] [0000e75a] libusb: debug [linux_get_parent_info] dev 0x564116543fe0 (2-2.1) has parent 0x56411631bb20 (2-2) port 1
[ 0.003902] [0000e75a] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.003913] [0000e75a] libusb: debug [linux_get_device_address] scan usb1
[ 0.003949] [0000e75a] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.003958] [0000e75a] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.003965] [0000e75a] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004083] [0000e75a] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.004093] [0000e75a] libusb: debug [linux_get_device_address] scan 1-1
[ 0.004129] [0000e75a] libusb: debug [linux_get_device_address] bus=1 dev=28
[ 0.004138] [0000e75a] libusb: debug [linux_enumerate_device] busnum 1 devaddr 28 session_id 284
[ 0.004145] [0000e75a] libusb: debug [linux_enumerate_device] allocating new device for 1/28 (session 284)
[ 0.004201] [0000e75a] libusb: debug [linux_get_parent_info] dev 0x56411660c9d0 (1-1) has parent 0x5641165440f0 (usb1) port 1
[ 0.004277] [0000e75a] libusb: debug [usbi_add_event_source] add fd 6 events 1
[ 0.004291] [0000e75a] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.004298] [0000e75a] libusb: debug [usbi_add_event_source] add fd 7 events 1
2021-09-23 14:43:24,773 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2021-09-23 14:43:24,774 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.004848] [0000e75a] libusb: debug [libusb_get_device_list]  
2021-09-23 14:43:24,774 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fed8304acd0>)
[ 0.005061] [0000e75a] libusb: debug [libusb_get_device_descriptor]  
2021-09-23 14:43:24,774 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x7fed8304acd0>)
[ 0.005365] [0000e75a] libusb: debug [libusb_open] open 1.28
[ 0.005404] [0000e75a] libusb: debug [usbi_add_event_source] add fd 8 events 4
2021-09-23 14:43:24,775 DEBUG:usb.backend.libusb1:_LibUSB.reset_device(<usb.backend.libusb1._DeviceHandle object at 0x7fed8304aeb0>)
[ 0.005526] [0000e75a] libusb: debug [libusb_reset_device]  
2021-09-23 14:43:25,246 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x7fed8304aeb0>)
[... cut ...]
[ 2.026377] [0000e75a] libusb: debug [libusb_unref_device] destroy device 2.13
[ 2.026403] [0000e75a] libusb: debug [libusb_unref_device] destroy device 2.3
[ 2.026430] [0000e75a] libusb: debug [libusb_unref_device] destroy device 2.2
[ 2.026455] [0000e75a] libusb: debug [libusb_unref_device] destroy device 2.1
[ 2.026481] [0000e75a] libusb: debug [usbi_remove_event_source] remove fd 7
[ 2.026527] [0000e75a] libusb: debug [usbi_remove_event_source] remove fd 6
[ 2.026828] [0000e75e] libusb: debug [linux_udev_event_thread_main] udev event thread exiting

stepandr47, 2021-09-23:
Windows Output:
2021-09-23 14:57:29,833 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\Users\astephen\Anaconda3\envs\seabreeze-test\Library\bin\libusb-1.0.dll', handle 7ffb354e0000 at 0x18e1fed4eb0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000063] [00002d14] libusb: debug [libusb_init] created default context
[ 0.000086] [00002d14] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000154] [00002d14] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.000177] [00002d14] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000641] [00002d14] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.000914] [00002d14] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.002100] [00002d14] libusb: info [windows_init] UsbDk backend is not available
[ 0.002233] [00002d14] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000358 events 0
[ 0.002257] [00002d14] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.002266] [00002d14] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000380 events 0
2021-09-23 14:57:29,835 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2021-09-23 14:57:29,835 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.002865] [00002d14] libusb: debug [libusb_get_device_list]  
[ 0.002988] [00003128] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.014861] [00002d14] libusb: debug [winusb_get_device_list] allocating new device for session [51]
[ 0.014978] [00002d14] libusb: debug [winusb_get_device_list] allocating new device for session [52]
[ 0.015084] [00002d14] libusb: debug [winusb_get_device_list] allocating new device for session [53]
[ 0.015172] [00002d14] libusb: debug [winusb_get_device_list] allocating new device for session [54]
[ 0.015494] [00002d14] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.015559] [00002d14] libusb: debug [winusb_get_device_list] allocating new device for session [7]
[ 0.015955] [00002d14] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.016040] [00002d14] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.016052] [00002d14] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.016060] [00002d14] libusb: debug [winusb_get_device_list] allocating new device for session [1E]
[ 0.016458] [00002d14] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.016518] [00002d14] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.016527] [00002d14] libusb: debug [winusb_get_device_list] allocating new device for session [39]
[ 0.016824] [00002d14] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1C26&SUBSYS_21CF17AA&REV_04\3&B1BFB68&0&E8' bus number 1
[ 0.016882] [00002d14] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_1C2D&SUBSYS_21CF17AA&REV_04\3&B1BFB68&0&D0' bus number 2
[ 0.017914] [00002d14] libusb: debug [winusb_get_device_list] found existing device for session [52]
[ 0.114514] [00002d14] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.115399] [00002d14] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.115550] [00002d14] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_8087&PID_0024\5&2DF2E661&0&1'
[ 0.116696] [00002d14] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&35D3A64E&0' reports 3 ports
[ 0.116837] [00002d14] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&35D3A64E&0'
[ 0.117359] [00002d14] libusb: debug [winusb_get_device_list] found existing device for session [7]
[ 0.117617] [00002d14] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.117706] [00002d14] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 216 bytes)
[ 0.117763] [00002d14] libusb: debug [init_device] (bus: 2, addr: 2, depth: 2, port: 4): 'USB\VID_0A5C&PID_217F\60D819AF2C20'
[ 0.118322] [00002d14] libusb: debug [winusb_get_device_list] found existing device for session [1E]
[ 0.118739] [00002d14] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.122161] [00002d14] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 93 bytes)
[ 0.122326] [00002d14] libusb: debug [init_device] (bus: 1, addr: 2, depth: 2, port: 5): 'USB\VID_17EF&PID_1003\6&37ACA2D4&0&5'
[ 0.123291] [00002d14] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&3423CC03&0' reports 3 ports
[ 0.123408] [00002d14] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&3423CC03&0'
[ 0.123952] [00002d14] libusb: debug [winusb_get_device_list] found existing device for session [51]
[ 0.124184] [00002d14] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.125032] [00002d14] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.125165] [00002d14] libusb: debug [init_device] (bus: 2, addr: 1, depth: 1, port: 1): 'USB\VID_8087&PID_0024\5&C22FC1F&0&1'
[ 0.125712] [00002d14] libusb: debug [winusb_get_device_list] extra GUID: {DBBAD306-1786-4F2E-A8AB-340D45F0653F}
[ 0.125780] [00002d14] libusb: debug [winusb_get_device_list] found existing device for session [39]
[... cut ...]
[ 0.250692] [00002d14] libusb: debug [libusb_unref_device] destroy device 2.0
2021-09-23 14:57:30,083 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
[ 0.250843] [00002d14] libusb: debug [libusb_exit]  
[ 0.250854] [00002d14] libusb: debug [libusb_exit] destroying default context
[ 0.250861] [00002d14] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000380
[ 0.250879] [00002d14] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000358
[ 0.251001] [00003128] libusb: debug [windows_iocp_thread] I/O completion thread exiting

ap--, 2021-09-23:
Hi @stepandr47
Thanks for the clean debug logs ❤️ 
I asked for advice on how to best proceed with the debugging at pyusb/pyusb#390
I'm sure they'll be able to help.

ap--, 2021-09-24:
So, new things to test:
- [ ] create debug log on a different linux DISTRIBUTION compared to the one you used.
       (are you running the same distro on both the VM and the pi?)
- [ ] use wireshark to create a packet capture to record the usb communication.
       (recently in another issue https://github.com/ap--/python-seabreeze/issues/59#issuecomment-923275781 a user recorded the usb communication for the OceanFX, so we could ask if you have any questions)

stepandr47, 2021-09-24:
The first Linux output came from a Ubuntu VM. Below is the output from my current Raspberry Pi setup that I had captured yesterday. I won't be able to test with wireshark until Monday as I don't have the sensor with me today. What specifically should I be looking for when going through packets on wireshark? Are you wanting me to just capture using pcap and just post the pcap file here?

Raspberry Pi Output:
2021-09-23 20:15:26,550 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 208c318 at 0xb4cda610>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000026] [000001ff] libusb: debug [libusb_init] created default context
[ 0.000068] [000001ff] libusb: debug [libusb_init] libusb v1.0.22.11312
[ 0.000138] [000001ff] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000164] [000001ff] libusb: debug [get_kernel_version] reported kernel version is 5.4.83-v7l+
[ 0.000177] [000001ff] libusb: debug [op_init] bulk continuation flag supported
[ 0.000189] [000001ff] libusb: debug [op_init] zero length packet flag supported
[ 0.000201] [000001ff] libusb: debug [op_init] max iso packet length is (likely) 49152 bytes
[ 0.000229] [000001ff] libusb: debug [op_init] sysfs can relate devices
[ 0.000241] [000001ff] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000575] [00000203] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.005170] [000001ff] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.005192] [000001ff] libusb: debug [linux_get_device_address] scan usb1
[ 0.005289] [000001ff] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.005304] [000001ff] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.005317] [000001ff] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.005621] [000001ff] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.005637] [000001ff] libusb: debug [linux_get_device_address] scan 1-1
[ 0.005726] [000001ff] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.005740] [000001ff] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.005753] [000001ff] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.005840] [000001ff] libusb: debug [linux_get_parent_info] Dev 0x20af5b0 (1-1) has parent 0x20af2b8 (usb1) port 1
[ 0.006083] [000001ff] libusb: debug [linux_get_device_address] getting address for device: 1-1.2 detached: 0
[ 0.006099] [000001ff] libusb: debug [linux_get_device_address] scan 1-1.2
[ 0.006189] [000001ff] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.006203] [000001ff] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.006215] [000001ff] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.006303] [000001ff] libusb: debug [linux_get_parent_info] Dev 0x209d3e8 (1-1.2) has parent 0x20af5b0 (1-1) port 2
[ 0.006553] [000001ff] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.006568] [000001ff] libusb: debug [linux_get_device_address] scan usb2
[ 0.006656] [000001ff] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.006670] [000001ff] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.006682] [000001ff] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.006842] [000001ff] libusb: debug [usbi_add_pollfd] add fd 6 events 1
[ 0.006869] [000001ff] libusb: debug [usbi_io_init] using timerfd for timeouts
[ 0.006882] [000001ff] libusb: debug [usbi_add_pollfd] add fd 8 events 1
2021-09-23 20:15:26,558 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2021-09-23 20:15:26,558 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.007703] [000001ff] libusb: debug [libusb_get_device_list] 
2021-09-23 20:15:26,559 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0xb5d82a10>)
[ 0.008181] [000001ff] libusb: debug [libusb_get_device_descriptor] 
2021-09-23 20:15:26,559 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0xb4ba6470>)
[ 0.008668] [000001ff] libusb: debug [libusb_get_device_descriptor] 
2021-09-23 20:15:26,560 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0xb4ba6470>)
[ 0.009143] [000001ff] libusb: debug [libusb_open] open 1.3
[ 0.009202] [000001ff] libusb: debug [usbi_add_pollfd] add fd 9 events 4
2021-09-23 20:15:26,560 DEBUG:usb.backend.libusb1:_LibUSB.reset_device(<usb.backend.libusb1._DeviceHandle object at 0xb4ba6410>)
[ 0.009397] [000001ff] libusb: debug [libusb_reset_device] 
2021-09-23 20:15:26,790 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0xb4ba6410>)
[ 0.239256] [000001ff] libusb: debug [libusb_close] 
[ 0.239278] [000001ff] libusb: debug [usbi_remove_pollfd] remove fd 9
2021-09-23 20:15:26,790 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0xb4ba6410>)
[ 0.239584] [000001ff] libusb: debug [libusb_get_device_descriptor] 
2021-09-23 20:15:26,790 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0xb4ba6570>)
[... cut ...]
[ 1.741085] [000001ff] libusb: debug [libusb_unref_device] destroy device 2.1
[ 1.741117] [000001ff] libusb: debug [libusb_unref_device] destroy device 1.3
[ 1.741145] [000001ff] libusb: debug [libusb_unref_device] destroy device 1.2
[ 1.741174] [000001ff] libusb: debug [libusb_unref_device] destroy device 1.1
[ 1.741202] [000001ff] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 1.741290] [000001ff] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 1.741420] [00000203] libusb: debug [linux_udev_event_thread_main] udev event thread exiting

ap--, 2021-09-26:
> What specifically should I be looking for when going through packets on wireshark?
> Are you wanting me to just capture using pcap and just post the pcap file here?

If you could just post the pcap file that would be great. Tbh, I'm not entirely sure what to look for either. Which is why I think having the pcap file will make it a lot easier to ask the libusb experts for further advice.

If you compare for example your virtual machine logs with the Pi logs, you can see, that they error in different parts of the code. The VM log errors on the first read, whereas the Pi log goes through a few more commands before it errors.

Interestingly both error when requesting the length of the serial number buffer, just after the device gets opened. I will open a new branch and increase the write/read timeout for that command to something large. Maybe that solves the issue.

stepandr47, 2021-09-27:
Command 1 :
pip3 install git+https://github.com/ap--/python-seabreeze.git@hdx-increased-timeout-test

Command 2:
git clone https://github.com/ap--/python-seabreeze.git
cd python-seabreeze
git fetch hdx-increased-timeout-test
git checkout feature hdx-increased-timeout-test
cd python-seabreeze
pip3 install .

I am unsure what I did wrong for installing a specific branch on my Ubuntu VM. The above commands to install a specific branch work on windows, but when I try it on my Ubuntu VM I get:
Processing /home/ubuntu/Documents/python-seabreeze
  DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
   pip 21.3 will remove support for this functionality. You can find discussion regarding this at https://github.com/pypa/pip/issues/7555.
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /home/ubuntu/anaconda3/envs/seabreeze-test/bin/python3.9 /home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpkrt1q8iw
       cwd: /tmp/pip-req-build-u9268cpc
  Complete output (22 lines):
  Traceback (most recent call last):
    File "/home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 349, in <module>
      main()
    File "/home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 331, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 117, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 154, in get_requires_for_build_wheel
      return self._get_build_requires(
    File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 135, in _get_build_requires
      self.run_setup()
    File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 258, in run_setup
      super(_BuildMetaLegacyBackend,
    File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 150, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 71, in <module>
      compile_opts = pkgconfig.parse("libusb")
    File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/pkgconfig/pkgconfig.py", line 248, in parse
      _raise_if_not_exists(package)
    File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/pkgconfig/pkgconfig.py", line 103, in _raise_if_not_exists
      raise PackageNotFoundError(package)
  pkgconfig.pkgconfig.PackageNotFoundError: libusb not found
  ----------------------------------------
WARNING: Discarding file:///home/ubuntu/Documents/python-seabreeze. Command errored out with exit status 1: /home/ubuntu/anaconda3/envs/seabreeze-test/bin/python3.9 /home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpkrt1q8iw Check the logs for full command output.
ERROR: Command errored out with exit status 1: /home/ubuntu/anaconda3/envs/seabreeze-test/bin/python3.9 /home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpkrt1q8iw Check the logs for full command output.

stepandr47, 2021-09-27:
Command 1 :
pip3 install git+https://github.com/ap--/python-seabreeze.git@hdx-increased-timeout-test

Command 2:
git clone https://github.com/ap--/python-seabreeze.git
cd python-seabreeze
git fetch hdx-increased-timeout-test
git checkout feature hdx-increased-timeout-test
cd python-seabreeze
pip3 install .

I am unsure what I did wrong for installing a specific branch on my Ubuntu VM. The above commands to install a specific branch work on windows, but when I try it on my Ubuntu VM I get the below output. I see that it says libusb not found at the end, but libusb is already installed so I am unsure what is causing that.

Ouput:
Processing /home/ubuntu/Documents/python-seabreeze
DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
pip 21.3 will remove support for this functionality. You can find discussion regarding this at pypa/pip#7555.
Installing build dependencies ... done
Getting requirements to build wheel ... error
ERROR: Command errored out with exit status 1:
command: /home/ubuntu/anaconda3/envs/seabreeze-test/bin/python3.9 /home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpkrt1q8iw
cwd: /tmp/pip-req-build-u9268cpc
Complete output (22 lines):
Traceback (most recent call last):
File "/home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 349, in
main()
File "/home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 331, in main
json_out['return_val'] = hook(**hook_input['kwargs'])
File "/home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 117, in get_requires_for_build_wheel
return hook(config_settings)
File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 154, in get_requires_for_build_wheel
return self._get_build_requires(
File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 135, in _get_build_requires
self.run_setup()
File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 258, in run_setup
super(_BuildMetaLegacyBackend,
File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 150, in run_setup
exec(compile(code, file, 'exec'), locals())
File "setup.py", line 71, in
compile_opts = pkgconfig.parse("libusb")
File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/pkgconfig/pkgconfig.py", line 248, in parse
_raise_if_not_exists(package)
File "/tmp/pip-build-env-o1v4khou/overlay/lib/python3.9/site-packages/pkgconfig/pkgconfig.py", line 103, in _raise_if_not_exists
raise PackageNotFoundError(package)
pkgconfig.pkgconfig.PackageNotFoundError: libusb not found
WARNING: Discarding file:///home/ubuntu/Documents/python-seabreeze. Command errored out with exit status 1: /home/ubuntu/anaconda3/envs/seabreeze-test/bin/python3.9 /home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpkrt1q8iw Check the logs for full command output.
ERROR: Command errored out with exit status 1: /home/ubuntu/anaconda3/envs/seabreeze-test/bin/python3.9 /home/ubuntu/anaconda3/envs/seabreeze-test/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py get_requires_for_build_wheel /tmp/tmpkrt1q8iw Check the logs for full command output.

ap--, 2021-09-27:
You need to follow: https://python-seabreeze.readthedocs.io/en/latest/install.html#step-1-linux
probably you're missing: `sudo apt-get install git-all build-essential libusb-dev`

stepandr47, 2021-09-27:
That was it thanks. I installed that branch on windows, ubuntu and RPi. Still only working on windows. I have also captured the pcap for windows and ubuntu. I am unsure of a good way to do this on a RPi via ssh. Both of the files can be found in the below zip file.
[pcap_files.zip](https://github.com/ap--/python-seabreeze/files/7237325/pcap_files.zip)

ap--, 2021-09-27:
Just a quick question: do you connect the spectrometer through a USB hub on the PI?
Just looking at the packet captures it seems the VM communicates through a USB hub.
The Windows captures don't mention a hub.

stepandr47, 2021-09-27:
I did the packet capture through Windows Wireshark while the device was connected to a Ubuntu VM. So windows seems to see it from the VM as a USB hub.

ap--, 2021-09-27:
So just to be clear, the three test systems are:

1. Windows Host --- Spectrometer connected directly to Host PC
2. Windows Host running Ubuntu VM --- Spectrometer directly connected to Host PC & attached to virtual machine
3. Raspberry Pi Host --- Spectrometer connected directly to Raspberry Pi

And neither in 1+2 nor in 3 is a physical USB hub involved.

stepandr47, 2021-09-27:
Yes that is all correct

ap--, 2021-09-28:
Okay, so the packet capture was not conclusive. https://github.com/pyusb/pyusb/issues/390#issuecomment-928607736
I guess we continue the Quest and our next stop is the linux-usb mailing list. :crossed_swords: :european_castle: :dragon:  

In the meantime we should investigate if the control requests are problematic.
This all relates to this class: https://github.com/ap--/python-seabreeze/blob/5ed92090d773932b04c5cc4da3cee3709516767d/src/seabreeze/pyseabreeze/transport.py#L161
Basically pyseabreeze grew organically out of the predecessor package `python-oceanoptics` and I refactored everything back then to adhere to the same api that libseabreeze exposed. It's possible that the order of calls made in there is incorrect, but has not affected a significant amount of the userbase so far.

So, if you could create two more packet captures from the linux VM for the following two scripts, that would allow us to check if there are differences between the two backends. **NOTE: this would need to be done with the `Flame-T`**

```python
# cseabreeze test
import seabreeze
seabreeze.use('cseabreeze')
from seabreeze.spectrometers import Spectrometer
print(Spectrometer.from_first_available())
```

```python
# pyseabreeze test
import seabreeze
seabreeze.use('pyseabreeze')
from seabreeze.spectrometers import Spectrometer
print(Spectrometer.from_first_available())
```

stepandr47, 2021-09-29:
Below are the two files pcap files stored in a zip file. One for pyseabreeze and one for cseabreeze when using the above scripts to detect the Flame-T.

[seabreeze_test.zip](https://github.com/ap--/python-seabreeze/files/7252211/seabreeze_test.zip)

Lavikjo, 2021-10-01:
I have witnessed a similar behavior before when running Linux (I think it was Ubuntu) in a VM and connecting to the HDX through USB. In my experience the connection was working when I installed the correct version of libusb there. For life of me I cannot remember which version was it as I'm currently running HDX on a daily basis through the ethernet with a highly experimental fork of the library.

ap--, 2021-10-01:
Hi @Lavikjo 
Thanks, that's value information! Do you at least remember if you built libusb from source or a pre-packaged version?

Cheers,
Andreas 😃

ap--, 2021-10-02:
Hi @stepandr47 

Given Lavikjo's comment so far, I think it might be worth trying different pyusb backends.
I pushed changes to the master branch that allow requesting a specific backend for `pyusb` via `pyseabreeze`.
Please install the latest version of python-seabreeze from the master branch and try if one of those might work for you:

#### libusb1

```python
# pyseabreeze using libusb1 (this is what currently doesn't work for you)
import seabreeze
print(seabreeze.__version__)
seabreeze.use('pyseabreeze', pyusb_backend='libusb1')
from seabreeze.spectrometers import Spectrometer
print(Spectrometer.from_first_available())
```

#### libusb0

```python
# pyseabreeze using libusb0 (you might have to `apt install libusb-0.1-4`)
import seabreeze
print(seabreeze.__version__)
seabreeze.use('pyseabreeze', pyusb_backend='libusb0')
from seabreeze.spectrometers import Spectrometer
print(Spectrometer.from_first_available())
```

#### openusb

```python
# pyseabreeze using openusb (you might have to `apt install libopenusb0`)
import seabreeze
print(seabreeze.__version__)
seabreeze.use('pyseabreeze', pyusb_backend='openusb')
from seabreeze.spectrometers import Spectrometer
print(Spectrometer.from_first_available())
```

stepandr47, 2021-10-04:
The outputs for each script can be seen below.

libusb1:
1.3.0.post23+gf277e90
Traceback (most recent call last):
  File "libusb1.py", line 5, in <module>
    print(Spectrometer.from_first_available())
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 87, in from_first_available
    for dev in list_devices():
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 36, in list_devices
    return list_devices._api.list_devices()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/api.py", line 104, in list_devices
    dev.open()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py", line 345, in open
    self._serial_number = self.get_serial_number()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py", line 385, in get_serial_number
    serial_len = ord(protocol.query(0x00000101))
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 369, in query
    return self.receive(timeout_ms=timeout_ms)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 310, in receive
    response = self.transport.read(size=64, timeout_ms=timeout_ms)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/transport.py", line 250, in read
    return self._device.pyusb_device.read(endpoint, size, timeout=timeout_ms)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

libusb0:
1.3.0.post23+gf277e90
Traceback (most recent call last):
  File "libusb0.py", line 5, in <module>
    print(Spectrometer.from_first_available())
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 87, in from_first_available
    for dev in list_devices():
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 36, in list_devices
    return list_devices._api.list_devices()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/api.py", line 104, in list_devices
    dev.open()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py", line 345, in open
    self._serial_number = self.get_serial_number()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py", line 385, in get_serial_number
    serial_len = ord(protocol.query(0x00000101))
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 369, in query
    return self.receive(timeout_ms=timeout_ms)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 310, in receive
    response = self.transport.read(size=64, timeout_ms=timeout_ms)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/transport.py", line 250, in read
    return self._device.pyusb_device.read(endpoint, size, timeout=timeout_ms)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/backend/libusb0.py", line 553, in bulk_read
    return self.__read(_lib.usb_bulk_read,
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/backend/libusb0.py", line 692, in __read
    ret = int(_check(fn(
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/backend/libusb0.py", line 446, in _check
    raise USBTimeoutError(errmsg, ret, -ret)
usb.core.USBTimeoutError: [Errno 110] b'could not get bound driver: No data available'

openusb:
1.3.0.post23+gf277e90
Traceback (most recent call last):
  File "openusb.py", line 5, in <module>
    print(Spectrometer.from_first_available())
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 87, in from_first_available
    for dev in list_devices():
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 36, in list_devices
    return list_devices._api.list_devices()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/api.py", line 104, in list_devices
    dev.open()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py", line 345, in open
    self._serial_number = self.get_serial_number()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py", line 385, in get_serial_number
    serial_len = ord(protocol.query(0x00000101))
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 369, in query
    return self.receive(timeout_ms=timeout_ms)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 310, in receive
    response = self.transport.read(size=64, timeout_ms=timeout_ms)
[... cut ...]
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/backend/openusb.py", line 664, in bulk_read
    _check(_lib.openusb_bulk_xfer(dev_handle, intf, ep, byref(request)))
  File "/home/ubuntu/.local/lib/python3.8/site-packages/usb/backend/openusb.py", line 510, in _check
    raise USBTimeoutError(_lib.openusb_strerror(ret), ret, _openusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] b'I/O timeout'

felipebrandes, 2023-01-20:
Hi @stepandr47, I'm having the same issue, have we had any changes since your first report?

zachjacobs, 2023-08-31:
I'm having the same issue with the FX as well.  I have one on hand and can provide any wireshark or usb debug logs as needed.

zachjacobs, 2023-09-28:
I'm seeing the same issue with Ubuntu 22.04 and also Raspberry Pi running Raspian.  Interestingly, I can get it to work periodically on Ubuntu Desktop. Usb debug logs here:

[working.txt](https://github.com/ap--/python-seabreeze/files/12744880/working.txt)
[timeout.txt](https://github.com/ap--/python-seabreeze/files/12744881/timeout.txt)

ghislainp, 2023-10-11:
Same problem here with a HDX connected by usb on  a dell laptop. I've tried libusb0 and libusb1.

Interestingly when I use seabreeze.use('pyseabreeze', pyusb_backend='openusb') and run list_devices():
-  the first time I get "usb.core.USBTimeoutError: [Errno 110] b'could not get bound driver: No data available'"
- the second time I get [<SeaBreezeDevice HDX:?>]

If I'm using csearbreeze, I always get an empty list, without error.

It seems a lot of debugging have been tried, but I can help if necessary...

dstr951, 2024-01-29:
I'm also in a similar condition, spectrometer works with cbreeze, but when I try to use pyseabreeze, I get an error saying that the usb port is busy.... my spectrometer is QEPRO

ap--, 2024-02-06:
> I'm also in a similar condition, spectrometer works with cbreeze, but when I try to use pyseabreeze, I get an error saying that the usb port is busy.... my spectrometer is QEPRO

@dstr951 Please open a new issue for the QEPRO.

ap--, 2024-06-27:
This is now possible if connecting via Ethernet.