ap--/python-seabreeze #139: src/libseabreeze/src/api/seabreezeapi/DHCPServerFeatureAdapter.cpp:64:16: error: template argument 1 is invalid

greyltc, 2021-11-26:
Hi, I'm having trouble building the project under Arch Linux:
```
# git clone python-seabreeze
# cd python-seabreeze
python setup.py build >>file.txt 2>&1
```
[file.txt](https://github.com/ap--/python-seabreeze/files/7610145/file.txt) is attached

The first compile error is:
```
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fno-semantic-interposition -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -march=x86-64 -mtune=generic -O3 -pipe -fno-plt -fPIC -I/usr/include/libusb-1.0 -Isrc/libseabreeze/include -I/usr/include/python3.9 -c src/libseabreeze/src/api/seabreezeapi/DHCPServerFeatureAdapter.cpp -o build/temp.linux-x86_64-3.9/src/libseabreeze/src/api/seabreezeapi/DHCPServerFeatureAdapter.o
src/libseabreeze/src/api/seabreezeapi/DHCPServerFeatureAdapter.cpp: In member function ‘void seabreeze::api::DHCPServerFeatureAdapter::getServerAddress(int*, unsigned char, unsigned char (*)[4], unsigned char*)’:
src/libseabreeze/src/api/seabreezeapi/DHCPServerFeatureAdapter.cpp:64:16: error: template argument 1 is invalid
   64 |     vector<byte> serverAddressVector;
      |                ^
```
Any ideas?

ap--, 2021-12-12:
Hi @greyltc 

What's your gcc version?
I will try to reproduce this error on my machine.

Cheers,
Andreas 😃

greyltc, 2021-12-12:
```
$ gcc --version                                                                                                                                                                                                                                                             
gcc (GCC) 11.1.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ cython --version
Cython version 0.29.25
```

ap--, 2021-12-13:
Hi @greyltc 

With `gcc-11` I get the same error as you reported.
But seabreeze builds fine on ArchLinux using `gcc-10`.

```console
$ gcc --version
gcc (Arch Linux 10.3.0-2) 10.3.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

If you could work out what compiler flag changes or code changes would be required to support older and newer versions of `gcc` (see: https://gcc.gnu.org/gcc-11/changes.html) that would be great.
I would suggest opening a PR and I could give some guidance in there on how to either implement the compiler flag changes dependent on gcc version or how to make the necessary code changes.

Let me know what you think,
Cheers,
Andreas :smiley:

PR #143 (2021-12-23): replace usage of "byte" with "unsigned char" as needed
Fixes #139. Only tested with gcc version 11
closed 2021-12-23 by ap--