ap--/python-seabreeze #140: Subclassing Spectrometer throws error

jtmbeta, 2021-12-07:
### spectrometer and system information ###

- model: USB2000
- operating system: macOS Big Sur (M1)
- python version: Python 3.7.7
- python-seabreeze version: 1.3.0
- installed-via: conda

### current problem ###

When subclassing the Spectrometer class, an error is thrown when using the from_first_available() method. Previosuly (with STS) this didn't happen and I'm not sure if its down to code or hardware changes. I could be missing something really simple...

### steps to reproduce ###

_Step by step instructions to reproduce the error. The more detailed the better, but please use some common sense:_

1. Connect spectrometer
2. Run code

### minimal code example and error (very helpful if available) ###

```python
from seabreeze.spectrometers import Spectrometer

class OceanOptics(Spectrometer):

    def __init__(self):
        super(Spectrometer, self).__init__()
        
oo = OceanOptics.from_first_available()

Traceback (most recent call last):

  File "/var/folders/c9/7yddvl1n2ss863cgfngj0wpm0000gp/T/ipykernel_48551/1377001751.py", line 1, in <module>
    oo = OceanOptics.from_first_available()

  File "/Users/jtm545/opt/anaconda3/envs/pyplr/lib/python3.7/site-packages/seabreeze/spectrometers.py", line 89, in from_first_available
    return cls(dev)

TypeError: __init__() takes 1 positional argument but 2 were given
```

ap--, 2021-12-08:
Hi @jtmbeta,

Your `__init__` is missing a required argument. The following should work: 

```python
from seabreeze.spectrometers import Spectrometer

class OceanOptics(Spectrometer):
    def __init__(self, device):
        super().__init__(device)
```

Let me know if this solves your problem 😊 

Cheers,
Andreas 😃

jtmbeta, 2021-12-08:
Hi Andreas,

That’s it! Thank you!

Best,
Joel

> On 8 Dec 2021, at 12:02, Andreas Poehlmann ***@***.***> wrote:
> 
> class OceanOptics(Spectrometer
> ):
>     
> def __init__(self, device
> ):
>         
> super().__init__(device)

ap--, 2021-12-08:
Happy I could help 🎉 

If there's any other issues feel free to open a new one.

closed 2021-12-08 by ap--