ap--/python-seabreeze #146: Timeout needed for proper trigger support

drmcnelson, 2022-01-19:
### spectrometer and system information ###

USB2000, etc - all models with external trigger capability

### current problem ###

To adequately support triggered operation,  the API to retrieve spectra, absolutely MUST have a timeout parameter.

Without this,  the system will either hang or fail on an internal not-user-controlled timeout.

So, this is a fantastically high priority.

ap--, 2022-01-20:
Hi @drmcnelson,

This will be a bit more involved. It's possible to change the `pyseabreeze` backend to allow pass-through of a timeout value. For the `cseabreeze` backend these USB timeouts are hardcoded, so it'll involve more work to implement.

I will implement the functionality for `pyseabreeze` first.
But I'm currently pretty overloaded with work, so this is a very low priority right now.

I'll update this issue when I start work on this.
I can also give some guidance on how to implement this feature, if someone volunteers to work on a PR.

Cheers,
Andreas 😃

drmcnelson, 2022-01-20:
Yes, I saw where it would go in the pyseabreeze code.     An abort could be a useful alternative for some use cases.

ap--, 2022-08-07:
Note to self: It seems there's no easy hacky way to do this with the sync interface: https://libusb.sourceforge.io/api-1.0/libusb_io.html#sync

drmcnelson, 2022-08-07:
Yes, it looks messy.   The comment that they need to be called to handle events seems a little concerning.

I have been using the device in triggered mode for clocked data acquisitions.   So I'm getting along with out it for the moment.

Thank you for looking into it.

drmcnelson, 2022-08-07:
user space thread to implement a watchdog timer?   (provided the overhead is small)

But again, I don't want to cause you extra work.    It has been working very well as is, for my clocked data acquisitions.

At this point, I am pretty sure I would not change my code to use the timeout.