ap--/python-seabreeze #147: Python cseabreeze works on Mac but not on Windows

hari-pad, 2022-01-19:
### spectrometer and system information ###

- model: HR4000
- operating system: Windows 10 64 bit
- python version: python 3.9.1
- python-seabreeze version: current master
- installed-via: pip

### current problem ###

I downloaded and installed python-seabreeze with the default cseabreeze backend, exactly following the instructions on the website. The install seems to have worked on my Macbook (M1 chip, Monterey 12.0.1), but not on my Windows. I then tried to install it with the pyseabreeze backend on my Windows, and that seems to work, but cannot list devices.

### steps to reproduce ###

1. I uninstalled python-seabreeze, cleared the cache.
2. Installed python-seabreeze with the default cseabreeze backend using pip.
3. Ran seabreeze_os_setup.exe. This failed to install some driver packages (see below). 
4. Restarted the computer. 
5. Tried to import Spectrometer, list_devices, at which point it says seabreeze can't load 'cseabreeze'.

6. Repeated step 1.
7. Installed python-seabreeze with the pyseabreeze backend using pip.
8. Repeated steps 3 and 4, with the same results.
9. Tried to import Spectrometer, list_devices - seems to work, no errors.
10. Tried to use list_devices - doesn't work.

### minimal code example and error ###

##Installing seabreeze:

```
pip install seabreeze ##works without any issues, so output not included here
seabreeze_os_setup
```
_Install windows drivers? [y/n] y
Launched admin shell
Install windows drivers? [y/n] y
Driver package added successfully.
Published name :            oem161.inf

Processing inf :            OOI_QEPRO.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_QEPRO_XP.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_SAS.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_Spark.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_Spark_XP.inf
Adding the driver package failed : The third-party INF does not contain digital signature information.

Processing inf :            OOI_STS.inf
Successfully installed the driver.
Driver package added successfully.
Published name :            oem162.inf

Processing inf :            OOI_STS_XP.inf
Successfully installed the driver.
Driver package added successfully.
Published name :            oem163.inf

Processing inf :            OOI_Torus.inf
Successfully installed the driver.
Driver package added successfully.
Published name :            oem164.inf

[... 408 lines cut ...]
    o = self.data[key]()
KeyError: (9303, 4114, 1, 6)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\THz-RED\Documents\GitHub\REMY_official\seabreeze_test.py", line 5, in <module>
    devices = list_devices()
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\spectrometers.py", line 57, in list_devices
    return api.list_devices()
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\api.py", line 115, in list_devices
    dev = _seabreeze_device_factory(usb_dev)
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\api.py", line 55, in _seabreeze_device_factory
    dev = _seabreeze_device_instance_registry[ident] = SeaBreezeDevice(device)
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\devices.py", line 345, in __init__
    self._serial_number = self.get_serial_number()
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\devices.py", line 412, in get_serial_number
    protocol = self._transport.protocol
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\transport.py", line 208, in protocol
    raise RuntimeError("no protocol instance available")
RuntimeError: no protocol instance available_

#####

I'd be grateful for any help you can offer with this! I am not partial towards the C or Python backend, I would just like to get it working on Windows. Great package, I appreciate all the work.

ap--, 2022-01-19:
Hi @hari-pad 

Can you try installing:
```
pip install seabreeze==1.3.0
```
and report back? Maybe there's an issue in the v2.0.0 release on windows.

Cheers,
Andreas 😃

hari-pad, 2022-01-19:
Dear @ap-- 

Thanks for the prompt response! Just tried that, and here's what I got - 

`pip install seabreeze==1.3.0`
Defaulting to user installation because normal site-packages is not writeable
Collecting seabreeze==1.3.0
  Downloading seabreeze-1.3.0.tar.gz (7.7 MB)
     |████████████████████████████████| 7.7 MB 6.4 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in c:\users\thz-red\appdata\roaming\python\python39\site-packages (from seabreeze==1.3.0) (1.22.1)
Requirement already satisfied: future in c:\users\thz-red\appdata\roaming\python\python39\site-packages (from seabreeze==1.3.0) (0.18.2)
Building wheels for collected packages: seabreeze
  Building wheel for seabreeze (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\program files\python39\python.exe' 'C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\pip\_vendor\pep517\in_process\_in_process.py' build_wheel 'C:\Users\THz-RED\AppData\Local\Temp\tmpb7bjt1qe'
       cwd: C:\Users\THz-RED\AppData\Local\Temp\pip-install-wv05senw\seabreeze_ed6c0fc5b8ff4bab9f96aa98f3c5995b
  Complete output (58 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.9
  creating build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\backends.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\compat.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\os_setup.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\spectrometers.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\_cli.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\_version.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\__init__.py -> build\lib.win-amd64-3.9\seabreeze
  creating build\lib.win-amd64-3.9\seabreeze\cseabreeze
  copying src\seabreeze\cseabreeze\__init__.py -> build\lib.win-amd64-3.9\seabreeze\cseabreeze
  creating build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\api.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\devices.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\exceptions.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\protocol.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\transport.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\__init__.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  creating build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\acquisitiondelay.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\continuousstrobe.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\databuffer.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\dhcpserver.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\eeprom.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\ethernetconfiguration.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\fastbuffer.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\fpga.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\gpio.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\i2cmaster.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\introspection.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\ipv4.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\irradcal.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\lightsource.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\multicast.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\networkconfiguration.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\nonlinearity.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\opticalbench.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\pixelbinning.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\rawusb.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\revision.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\shutter.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
[... cut ...] eze_wrapper.pyx
    tree = Parsing.p_module(s, pxd, full_module_name)
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for seabreeze
Failed to build seabreeze
ERROR: Could not build wheels for seabreeze, which is required to install pyproject.toml-based projects

I haven't encountered this error.

Best wishes,
Hari

ap--, 2022-01-19:
Interesting. I assumed I was shipping wheels for python3.9 for windows, but it looks like I wasn't. So let's try to fix it another way.

### First:

can you try doing the following:

```shell
pip install seabreeze==2.0.0
python --version --version
python -c "import seabreeze.cseabreeze._wrapper"
```

And report back

### And second:

Can you try installing Python 3.8 and with python 3.8 try `pip install seabreeze==1.3.0` and see if that works. That would allow us to rule out if it's a problem with the `2.0.0` version of seabreeze on windows.

hari-pad, 2022-01-19:
# First:

`pip install seabreeze==2.0.0
`
```
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: seabreeze==2.0.0 in c:\users\thz-red\appdata\roaming\python\python39\site-packages (2.0.0)
Requirement already satisfied: numpy>=1.21 in c:\users\thz-red\appdata\roaming\python\python39\site-packages (from seabreeze==2.0.0) (1.22.1)
```

`python --version --version`

`Python 3.9.1 (tags/v3.9.1:1e5d33e, Dec  7 2020, 17:08:21) [MSC v.1927 64 bit (AMD64)]
`
`python -c "import seabreeze.cseabreeze._wrapper"
`
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\cseabreeze\__init__.py", line 5, in <module>
    from seabreeze.cseabreeze._wrapper import SeaBreezeAcquisitionDelayFeature
ModuleNotFoundError: No module named 'seabreeze.cseabreeze._wrapper'
```

# Second:

Installed Python 3.8, chose it as the interpreter in Visual Studio Code, and ran the following.

`pip install seabreeze==1.3.0
`
```
Defaulting to user installation because normal site-packages is not writeable
Collecting seabreeze==1.3.0
  Using cached seabreeze-1.3.0.tar.gz (7.7 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: future in c:\users\thz-red\appdata\roaming\python\python39\site-packages (from seabreeze==1.3.0) (0.18.2)
Requirement already satisfied: numpy in c:\users\thz-red\appdata\roaming\python\python39\site-packages (from seabreeze==1.3.0) (1.22.1)
Building wheels for collected packages: seabreeze
  Building wheel for seabreeze (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\program files\python39\python.exe' 'C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\pip\_vendor\pep517\in_process\_in_process.py' build_wheel 'C:\Users\THz-RED\AppData\Local\Temp\tmp5rtpffz3'
       cwd: C:\Users\THz-RED\AppData\Local\Temp\pip-install-f41xvkah\seabreeze_660f19b9ef2a4acfbc657092200a0b26
  Complete output (58 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.9
  creating build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\backends.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\compat.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\os_setup.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\spectrometers.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\_cli.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\_version.py -> build\lib.win-amd64-3.9\seabreeze
  copying src\seabreeze\__init__.py -> build\lib.win-amd64-3.9\seabreeze
  creating build\lib.win-amd64-3.9\seabreeze\cseabreeze
  copying src\seabreeze\cseabreeze\__init__.py -> build\lib.win-amd64-3.9\seabreeze\cseabreeze
  creating build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\api.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\devices.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\exceptions.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\protocol.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\transport.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  copying src\seabreeze\pyseabreeze\__init__.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze
  creating build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\acquisitiondelay.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
[... 14 lines cut ...]
  copying src\seabreeze\pyseabreeze\features\networkconfiguration.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\nonlinearity.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\opticalbench.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\pixelbinning.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\rawusb.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\revision.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\shutter.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\spectrometer.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\spectrumprocessing.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\straylightcoefficients.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\strobelamp.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\temperature.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
  copying src\seabreeze\pyseabreeze\features\thermoelectric.py -> build\lib.win-amd64-3.9\seabreeze\pyseabreeze\features
[... cut ...] bc657092200a0b26\src\seabreeze\cseabreeze\c_seabreeze_wrapper.pyx
    tree = Parsing.p_module(s, pxd, full_module_name)
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for seabreeze
Failed to build seabreeze
ERROR: Could not build wheels for seabreeze, which is required to install pyproject.toml-based projects
```

ap--, 2022-01-19:
Okay, so the new wheels on windows are broken. I'll try to fix that.

But in the second test you still seem to be testing against Python3.9 `c:\program files\python39\python.exe`

hari-pad, 2022-01-19:
Apologies, I am still very much a Python rookie. I'm (apparently successfully) running Python 3.8 as the interpreter on VS Code but as you pointed out, it still seems to be using Python 3.9 - let me work on fixing this. 

Meanwhile I found something interesting has happened with the Python 3.9 - 

### Without the spectrometer plugged in - 

```
import seabreeze
seabreeze.use('pyseabreeze')
from seabreeze.spectrometers import list_devices, Spectrometer

devices = list_devices()
print(devices)
```

`[]`

This seems promising, it's the first time it worked. However - 

### With the spectrometer plugged in -

```
Traceback (most recent call last):
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\api.py", line 53, in _seabreeze_device_factory
    return _seabreeze_device_instance_registry[ident]
  File "C:\Program Files\Python39\lib\weakref.py", line 134, in __getitem__
    o = self.data[key]()
KeyError: (9303, 4114, 1, 9)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\THz-RED\Documents\GitHub\REMY_official\seabreeze_test.py", line 5, in <module>
    devices = list_devices()
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\spectrometers.py", line 57, in list_devices
    return api.list_devices()
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\api.py", line 115, in list_devices
    dev = _seabreeze_device_factory(usb_dev)
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\api.py", line 55, in _seabreeze_device_factory
    dev = _seabreeze_device_instance_registry[ident] = SeaBreezeDevice(device)
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\devices.py", line 345, in __init__
    self._serial_number = self.get_serial_number()
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\devices.py", line 412, in get_serial_number
    protocol = self._transport.protocol
  File "C:\Users\THz-RED\AppData\Roaming\Python\Python39\site-packages\seabreeze\pyseabreeze\transport.py", line 208, in protocol
    raise RuntimeError("no protocol instance available")
RuntimeError: no protocol instance available
```

I also tried rebooting the system and VS Code but get the same result as above.

I will report back as soon as I can figure out how to successfully install python-seabreeze 1.3.0 with Python 3.8, I appreciate your patience.

ap--, 2022-01-20:
Hi @hari-pad 

Your issue should be fixed in the new hotfix release `v2.0.1`.
Please try
```shell
pip install -U seabreeze
# this should install seabreeze==2.0.1
```
(On any of the supported python versions, so: 3.7, 3.8, 3.9, or 3.10)

If this resolves the issue feel free to close.
Let me know how things go.

Cheers,
Andreas 😃

hari-pad, 2022-01-20:
Andreas,

This seems to work! I am able to install it and then detect my instrument with no issues. 

Thank you so much for the help and the fix, I appreciate it. 

Best wishes,
Hari

closed 2022-01-20 by hari-pad