ap--/python-seabreeze #148: Version 2.0.1 crashes on read

drmcnelson, 2022-01-20:
### spectrometer and system information ###

- model: _(i.e. USB2000, STS, etc...)_

USB2000+

- operating system: _(i.e. Windows 7 64bit)_

Fedora 35

- python version: _(output of `python --version`)_

Python 3.10.1

- python-seabreeze version: _(i.e. 0.5.3 or 'current master')_

python seabreeze 2.0.1

- installed-via: _(conda, wheel, or python setup.py install)_

pip

### current problem ###

_detailed description of what doesn't work_

### steps to reproduce ###

  File "/usr/local/lib/python3.10/site-packages/seabreeze/spectrometers.py", line 212, in intensities
    out = self._dev.f.spectrometer.get_intensities()
  File "/usr/local/lib/python3.10/site-packages/seabreeze/pyseabreeze/features/spectrometer.py", line 147, in get_intensities
    tmp = self._get_spectrum_raw()
  File "/usr/local/lib/python3.10/site-packages/seabreeze/pyseabreeze/features/spectrometer.py", line 167, in _get_spectrum_raw
    tmp[:] = self.protocol.receive(
ValueError: invalid literal for int() with base 10: b'\x07\x00\xf6-H\x02<\x02-\x02N\x02&\x02E\x02.\x02I\x02V\x02a\x02Q\x02e\x02Z\x02h\x02l\x02q\x02g\x02\x8b\x02\xa3\x02\x96\x02\x92\x02\xa1\x02\x9e\x02\xab\x02\xa3\x02\x9c\x02\xc7\x02\xa9\x02\xcd\x02\xd2

1. run code example
2. ...

### minimal code example and error (very helpful if available) ###

```python
copy of minimal code example and error messages
```

ap--, 2022-01-20:
https://github.com/ap--/python-seabreeze/actions/runs/1725616126

There's new wheels ready for download on the current PR where I try to fix the described issue.

Could you download the wheels artifact from the link above, extract it, and install the corresponding wheel for your os and python version and test if it works?

Cheers,
Andreas 😃

drmcnelson, 2022-01-20:
I am not familiar with how to work with wheels.   Also, I recall that at 
least one of my systems was not using wheels.

drmcnelson, 2022-01-20:
Another bug, perhaps,  the following installation command line produces an error that option "--without-cseabreeze" is not recognized.

sudo pip install seabreeze[pyseabreeze] --no-use-pep517 --no-build-isolation --install-option="--without-cseabreeze"

ap--, 2022-01-21:
Once `v2.0.2` is out you should just be able to do:
(This pipeline has to finish to upload the new wheels: https://github.com/ap--/python-seabreeze/actions/runs/1728412220)

```shell
pip install -U seabreeze
```

and it should work.

Regarding the issue with
```
pip install seabreeze[pyseabreeze] --no-use-pep517 --no-build-isolation --install-option="--without-cseabreeze"
```
I would need more information about your current environment. please run `pip freeze` and report back with the versions of installed python libraries on your machine.

In case `v2.0.2` fixes the original issue, please close this issue 😃

drmcnelson, 2022-01-21:
Hi, here is the pip freeze.    If you can tell me how to download and install the wheel, I can give it a try.

[pip.freeze.txt](https://github.com/ap--/python-seabreeze/files/7916075/pip.freeze.txt)

ap--, 2022-01-21:
Sorry, could you run `pip freeze --all`?

And installing the wheel is not necessary. I made a new release, so just running `pip install -U seabreeze` on your machine should install `v2.0.2` with the newest fixes.

Cheers,
Andreas 😃

drmcnelson, 2022-01-21:
[pip.freeze.all.txt](https://github.com/ap--/python-seabreeze/files/7916540/pip.freeze.all.txt)

ap--, 2022-01-23:
Thanks for the info. I moved the discussion regarding the pip install problems to a new issue: #150 

In case version `v2.0.2` fixes the issue regarding crashes on `.read` please close this issue.

Cheers,
Andreas 😃

drmcnelson, 2022-01-23:
Will do, will try to test it later, working on a different measurement 
at the moment.  On that note, it would be great to be able to use your 
pyseabreeze with the adc1000usb

drmcnelson, 2022-01-25:
Version 2.0.2  seems to be a winner.   Installs easily and runs well.

Thank you

closed 2022-01-25 by drmcnelson

ap--, 2022-01-25:
Awesome. Great that it works!

Regarding the ADC1000USB, I'll open another issue for improving the docs on how to add new spectrometers and I'll start a draft pull-request sketching out what needs to be done to integrate the adc1000usb code.

Cheers,
Andreas :smiley:

PR #149 (2022-01-21): [pyseabreeze] fix regression after protocol unification
Should fix #148.
commit 302c20e1 (2022-01-21): [pyseabreeze] fix regression after protocol unification

Should fix #148.