ap--/python-seabreeze #150: Installing seabreeze using "--without-cseabreeze" errors

ap--, 2022-01-23:
Another bug, perhaps,  the following installation command line produces an error that option "--without-cseabreeze" is not recognized.

sudo pip install seabreeze[pyseabreeze] --no-use-pep517 --no-build-isolation --install-option="--without-cseabreeze"

_Originally posted by @drmcnelson in https://github.com/ap--/python-seabreeze/issues/148#issuecomment-1017999038_

ap--, 2022-01-23:
Okay, so this errors not when installing the seabreeze package, but one of it's dependencies.

See: https://github.com/pypa/pip/issues/1883

It seems `pip` will make this use case increasingly harder to support.
So my recommendation would be to just install the precompiled wheels from pypi via `pip install seabreeze[pyseabreeze]`
It's always supported to ignore the cseabreeze backend by using

```
import seabreeze
seabreeze.use("pyseabreeze")  # will force seabreeze to use pyseabreeze and ignore cseabreeze
...
```

If it's required to install seabreeze without the cseabreeze backend:

```sh
python -m pip install -U pip
python -m pip install setuptools setuptools_scm wheel "numpy>=1.21" "typing_extensions; python_version < '3.8'" "pyusb>=1.0.2"
python -m pip install --no-build-isolation --no-use-pep517 --use-deprecated=legacy-resolver "seabreeze" --install-option="--without-cseabreeze" 
```

Please let me know if this fixes the issue.
I will update the docs accordingly in case it does.

Cheers,
Andreas 😃

drmcnelson, 2022-01-23:
> See: pypa/pip#1883 <https://github.com/pypa/pip/issues/1883>

I would say the most usual use case is pyseabreaze without the 
cseabreeze backend.

Indeed I wonder, given that pyseabreeze seems to work so well without it 
and it seems to be so much easier to add features like the timeout on 
the python-only side, what is the point of the cseabreeze backend?  What 
does it add besides complication?

ap--, 2022-01-23:
> I would say the most usual use case is pyseabreaze without the cseabreeze backend.

A user survey I conducted around a year ago showed that the cseabreeze backend is the most commonly used by quite a margin.

> For one thing, I have yet to find the download for seabreeze on the oceaninsight web site. only the web page where they say it is under the MIT license but omit any clue one where to download it.

It's listed at the bottom of the README: https://github.com/ap--/python-seabreeze#related-repositories
Here's the direct link: https://sourceforge.net/projects/seabreeze/
Also it's vendored in this repository, with custom patches: https://github.com/ap--/python-seabreeze/tree/master/src/libseabreeze

> And,  given that pyseabreeze seems to work so well without cseabreeze, what is the point of it?  What does it add besides complication?

Compilation has not been an issue for the majority of the user base since binary wheels are provided for almost all python versions and platforms, via pypi and via conda. Which means that running:
```
pip install seabreeze
```
or
```
conda install -c conda-forge seabreeze
```
installs the precompiled `cseabreeze` backend by default. If specific Python versions or architectures do not provide wheels, I'm usually willing to add them to the CI/CD pipeline. (That's how initially support for RasperryPI Arm wheels was provided, until piwheels took over)

Moreover, the `cseabreeze` backend is still more feature complete compared to the `pyseabreeze` backend.
I would love to switch to the pure python implementation and drop the C backend, but a lack of hardware for testing, and a lack of user contributions (in case a user has access to a specific spectrometer) has made it impossible so far.

Since `python-seabreeze` is not actively supported by OceanInsight and I am not being paid for maintaining, further development, or porting to Python of more functionality of the cseabreeze backend, there's no strong incentive to drop a working backend.

drmcnelson, 2022-01-23:
> the |cseabreeze| backend is still more feature complete

Fair enough

> lack of hardware for testing, and a lack of user contributions (in 
> case a user has access to a specific spectrometer)

Yes, I might be one of those.  I have code for the adc100usb, but I 
don't know exactly what I need to do to add it your system.   If you 
have a short explicit list of instructions, starting with checkout from 
github, I might give it a try.

> vendored in this repository, with custom patches: 
> https://github.com/ap--/python-seabreeze/tree/master/src/libseabreeze

That addresses a few my issues with it.   My motto in software is thall 
shalt minimize external dependencies, and blessed are the simple for 
they shall receive swift error free compilation and fewer bugs.

So, I feel convinced, I might even try it.  Thank you

drmcnelson, 2022-01-25:
As of version 2.0.2,  pip install seabreeze[pyseabreeze] seems to work, not other switches were needed and there were no errors or warnings.

Very nice, thank you

ap--, 2022-01-25:
Great that it works :)

I'll keep this open until I updated the docs.

Cheers,
Andreas :smiley: