ap--/python-seabreeze #158: i cant turn the lamp on

chesse20, 2022-04-22:
### spectrometer and system information ###

USB 4000
windows 10

### current problem ###

i don't understand how to turn on the lamp, i tried following the api and it gets confusing to read. Can you add an example to the api page, i can't get data if the lamp is not on.

### steps to reproduce ###

1. run code example
2. get error

### minimal code example and error (very helpful if available) ###
my code and comments
```python
##in python console
## pip install seabreeze
## pip install matplotlib
from seabreeze.spectrometers import Spectrometer
import matplotlib.pyplot as plt
spec = Spectrometer.from_first_available()

###
#NOTE
#this code does not work unless you turn the lamp on!
#turn the lamp code
#
spec.f.SeaBreezeStrobeLampFeature.enable_lamp(1)
###
spec.integration_time_micros(20000)
print(spec.wavelengths())
plt.plot(spec.wavelengths(),spec.intensities())
plt.show()
```

    spec.f.SeaBreezeStrobeLampFeature.enable_lamp(1)
AttributeError: 'FeatureAccessHandler' object has no attribute 'SeaBreezeStrobeLampFeature'

ap--, 2022-04-22:
Hi @chesse20,

If your IDE supports code completion and you have installed seabreeze>=2.0.2, it should help with this, by suggesting the available attributes on the `spec.f` attribute. Please try the following:

```python
from seabreeze.spectrometers import Spectrometer
import matplotlib.pyplot as plt
spec = Spectrometer.from_first_available()

spec.f.strobe_lamp.enable_lamp(True)
spec.integration_time_micros(20000)
plt.plot(spec.wavelengths(), spec.intensities())
plt.show()
```

And let me know if it works.

Cheers,
Andreas :smiley:

chesse20, 2022-04-28:
this worked for me thank you!

closed 2022-04-28 by chesse20