ap--/python-seabreeze #162: UrlError when installing windows driver by os_setup

LeAmarillo, 2022-05-19:
### spectrometer and system information ###
- model: USB2000 but is not plugged into my pc during the installation
- operating system: Windows 10 64bit
- python version: Python 3.9.7
- python-seabreeze version: current master
- installed-via: _ conda

### current problem ###
Hi! I meet a weird problem. 

I cannot successfully run seabreeze_os_setup. Tried both run seabreeze_os_setup in Anaconda Powershell Prompt & run the os_setup.py directly in Spyder. 
The error information is like that:

Install windows drivers? [y/n] y
Downloading windows drivers from github
Error when installing drivers
Traceback (most recent call last):
  File "C:\Users\liane\anaconda3\lib\site-packages\seabreeze\os_setup.py", line 187, in windows_install_drivers
    drivers_zip_data = urlopen(url).read()
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 517, in open
    response = self._open(req, data)
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 539, in _open
    return self._call_chain(self.handle_open, 'unknown',
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 494, in _call_chain
    result = func(*args)
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 1417, in unknown_open
    raise URLError('unknown url type: %s' % type)
urllib.error.URLError: <urlopen error unknown url type: https>
Press [enter] to close.
[image]

On StackOverflow there was someone reporting similar error information. Some people solved the url open problem by installing openssl. I tried but it did not work.
[image]
Here you could see the openssl was installed successfully. I even tried to restart the computer but it did not help.

Also, I tried to change the url in the os_setup.py which is 
https://raw.githubusercontent.com/ap--/python-seabreeze/master/os_support
to 
http://raw.githubusercontent.com/ap--/python-seabreeze/master/os_support
and it did not work. Same error.

By the way, I tried to visit the url https://raw.githubusercontent.com/ap--/python-seabreeze/master/os_support  by browser but it only gave me a 404 error. Not sure if that matters.

### steps to reproduce ###

_Step by step instructions to reproduce the error. The more detailed the better, but please use some common sense:_

1. run conda install -c conda-forge seabreeze.
2. succeed
3. run seabreeze_os_setup
4. Install windows drivers? [y/n] -y
5. admin shell launched, asked me again: Install windows drivers? [y/n] -y
6. Error happened

### minimal code example and error (very helpful if available) ###

Install windows drivers? [y/n] y
Downloading windows drivers from github
Error when installing drivers
Traceback (most recent call last):
  File "C:\Users\liane\anaconda3\lib\site-packages\seabreeze\os_setup.py", line 187, in windows_install_drivers
    drivers_zip_data = urlopen(url).read()
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 517, in open
    response = self._open(req, data)
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 539, in _open
    return self._call_chain(self.handle_open, 'unknown',
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 494, in _call_chain
    result = func(*args)
  File "C:\Users\liane\anaconda3\lib\urllib\request.py", line 1417, in unknown_open
    raise URLError('unknown url type: %s' % type)
urllib.error.URLError: <urlopen error unknown url type: https>
Press [enter] to close.

ap--, 2022-05-19:
Hello,

an easy workaround, is to download the file manually https://github.com/ap--/python-seabreeze/raw/master/os_support/windows-driver-files.zip and run
```
seabreeze_os_setup path/to/your/downloaded/windows-driver-files.zip
```
in an admin shell

Let me know if that helps.
I will try to reproduce the issue when i have access to a windows machine.

Cheers,
Andreas

LeAmarillo, 2022-05-19:
Hi Andreas,

Thanks a lot for your quick response and it helps! Even though it told me there were some models not installed successfully(33/53 installed) but I checked the list that USB2000 should be okay. Thanks again!@ap--

ap--, 2022-05-20:
Glad I could help!
I'll leave this open for now so that I don't forget to try to reproduce.

Cheers,
Andreas 😃

Zhaofan2021, 2023-06-10:
I have also encountered this issue. Following the method you provided, download windows-driver-files.zip, rrun seabreeze_ Os_ Setup path/windows driver files. zip, successful. Thank you very much.