ap--/python-seabreeze #177: Problem with set_integration_time, the spectrometer doesn't wait for spectrum acquisition.

josue300earth, 2023-01-11:
### spectrometer and system information ###

- model: (USB2000+)
- operating system: _(Windows 10)
- python version: _(python 3.8.8)
- python-seabreeze version: _(2.0.3)
- installed-via: _(conda install)_

### current problem ###

_detailed description of what doesn't work_
My problem is simple I run this code: 

spec.integration_time_micros(10000000) # 10 seconds
wavelengths, intensities = spec.spectrum() 

however, when I establish the integration time and I acquire the spectrum, the program immediately takes a spectrum and shows it, not waiting until the spectrometer finishes the acquisition process. In a few words, the code "spec.spectrum()" doesn't block the spectrometer when it takes the spectrum. And the resulting spectrum is equal, no matter the integration time I use. 
I read other similar problems, but I don't use the "trigger mode" in my application, I just want only take a simple spectrum with different integration times. 

### minimal code example and error (very helpful if available) ###
This the code :

from seabreeze.spectrometers import Spectrometer, list_devices
import matplotlib.pyplot as plt
import time

devices = list_devices()
spec = Spectrometer(devices[0])

# set integration time
spec.integration_time_micros(10000000) # 10 seconds

wavelengths, intensities = spec.spectrum() #correct_dark_counts=True
plt.plot(wavelengths[10:],intensities[10:])

This code doesn't send error messages; it just doesn't work like it is supposed to. 

My doubt is, does python work with the spectrometer this way? Or I do something wrong. I didn't have problems with the installation o errors.

ap--, 2023-01-11:
Hi @josue300earth 

By default the spectrometer is constantly acquiring spectra. When you switch to a long integration time, the last spectrum taken at the short default integration time is still in the buffer.

If you run the following, you'll see that it'll take a 10 second spectrum:

```python
from seabreeze.spectrometers import Spectrometer, list_devices
import matplotlib.pyplot as plt

devices = list_devices()
spec = Spectrometer(devices[0])

# set integration time
spec.integration_time_micros(10000000) # 10 seconds
_ = spec.spectrum()  # throw away last spectrum...
wavelengths, intensities = spec.spectrum()
plt.plot(wavelengths[10:],intensities[10:])

```

Let me know if that fixes your issue.

Cheers,
Andreas :smiley:

josue300earth, 2023-01-12:
Hi Andreas, thank you for python code ad your quick answer. Well, here is my story; I did prove the code that you sent me, and it didn't work,  so I talked with other lab partners, and they lent me another spectrometer; it turns out that the spectrometer that I work has some damage because the new spectrometer works perfectly with previous code. I repeat, thank you. 

Regards Joshua.

ap--, 2023-01-12:
That's great! Happy to hear you found the cause.
I will close this issue now. Feel free to open another one in case there's anything else.

Cheers,
Andreas 😃

closed 2023-01-12 by ap--