ap--/python-seabreeze #185: Seabreeze and SR4 Spectrometers

fieterovich, 2023-03-27:
### spectrometer and system information ###

- model:  I don't really know, but it is the new _SR4 spectrometer, VIS-NIR_
- operating system: _linux ARM64_ (Jetson Xavier NX or Jetson AGX Orin)
- python version: _3.8_
- python-seabreeze version: _2.0.3_
- installed-via: _python in venv via pip3_

### current problem ###

Hi! I want to know if there is support for the new SR4 spectrometers, I tried to install seabreeze and it runs with an old Flame VIS-NIR spectrometer, as listed below:

`[<SeaBreezeDevice USB2000PLUS:FLMS00556>]`

but for the new SR4 one, I don't even find it listed when I try the commands devices = `sb.list_devices() `&
`print(devices)`

[image]

Is there any way to get this sorted? There is not too much info (as far as I could find) on how to run in linux ARM devices with python this specific spectrometer.

Best regards!
Francisco

ap--, 2023-03-29:
Hi @fieterovich 

Oceaninsight doesn't provide the developer documentation for their new spectrometer models anymore. And since I don't have the new hardware I can't help a lot with implementing support.

First thing you could do is run `lsusb` to determine the USB VendorID and ProductID for the new spectrometer model.

The next thing you could do is capture the usb traffic when the spectrometer communicates with the official ocean software. Google for "usb packet capture wireshark" to find instructions on how to set it up for your operating system. Once we have a packet capture of your spectrometer being initialized and retrieving one spectrum, we should have enough data to add support.

Cheers,
Andreas

fieterovich, 2023-04-04:
Hi Andreas, thank you for your answer!

The `lsusb` in a jetson lists: `Bus 001 Device 006: ID 0999:1002 Generic 4 Port USB 2.0 Hub`

And when I'm running Oceanview in windows I captured a file using Wireshark and USBPcap. 

Also, `Get-PnpDevice` in windows lists the spectrometer as: `OK         USBDevice       OceanHR4                                                                         USB\VID_...`

Let me know if I can give anything else from the spectrometer to you! Thank you for your time and your answer to my question.

[USB_capture.zip](https://github.com/ap--/python-seabreeze/files/11153375/USB_capture.zip)

ap--, 2023-04-04:
Awesome!

When I have some free time I will to look into it.
But from a quick look the packet capture is good.

I will try to get back to you within the next 10 days.

Cheers,
Andreas :smiley:

fieterovich, 2023-04-11:
Thank you Andreas!

This is what I did:

- Connected the SR4 Spectrometer
- Opened OceanView and pressed 'QuickView'
- Opened wireshark and selected 'USBPcap1'
- Right click and 'Start Capture'
- Entered the program, waited 30 seconds, went back to wireshark
- Stopped the capture with the red square in the GUI and saved the file.

Now I did the same, but with these modifications:

- When entering ocean view while capturing, selected quickview
- Selected the OceanView logo, and selected 'Reflectance'
- Followed the instructions (changed integration time, scans to average and boxcar width), created a new window of reflectance, by taking sample data with lights on and background reflectance with lights off.
- Stopped capturing and then saved the file

Here is the file attached. Let me know if this helps! Thanks again
Francisco

PS. if it is valuable to inform, the only USB connected to the laptop was the spectrometer. Did two files with the same procedure, but the USB capture 2 had some intervals where I got into internet and do some other staff, therefore a longer 'no action' time in the usb.

[USB_capture2.zip](https://github.com/ap--/python-seabreeze/files/11203203/USB_capture2.zip)

ap--, 2023-04-12:
Thank you for the clarification!
If you don't mind, we would need one more try:

Please do the following. Start with spectrometer disconnected, all oceanview software closed.
1. connect spectrometer
2. start the wireshark recording
3. open oceanview
4. record one spectrum
5. stop the recording

fieterovich, 2023-04-12:
Sure! Thank you very much. Attached is the file with the captured data.

[USB_capture4.zip](https://github.com/ap--/python-seabreeze/files/11214807/USB_capture4.zip)

ap--, 2023-04-12:
Hey @fieterovich 

In the pull request referenced above, there's some experimental code that you could try to check if rudimentary support for your spectrometer is working or not.

It seems a lot of the command set of the communication protocol was changed, and it's a bit hard to guess what the new correct commands are.

As always, the more detail you provide of what works and what doesn't, the more likely it is for me to guess what's wrong and to help.

Let me know how it goes,
Cheers,
Andreas :smiley:

fieterovich, 2023-04-12:
Thank you very much for the pull request and support @ap-- , I was trying to test it out, this is what I did in the Jetson:

- `source myenv/bin/activate`
- `git clone https://github.com/ap--/python-seabreeze.git`
- `cd python-seabreeze`
- `git fetch origin pull/189/head:pr-189`
- `git checkout pr-189`
- `pip install .`

Then I listed the usb, connected the spectrometer, and run a test code:

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ lsusb`
>Bus 002 Device 002: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 03f0:0d4a HP, Inc SK-2025 Keyboard
Bus 001 Device 004: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 003: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 002: ID 13d3:3549 IMC Networks Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ lsusb`
>Bus 002 Device 002: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 0999:1002 Generic 4-Port USB 2.0 Hub
Bus 001 Device 005: ID 03f0:0d4a HP, Inc SK-2025 Keyboard
Bus 001 Device 004: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 003: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 002: ID 13d3:3549 IMC Networks Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ python3 test_spectrometer.py `
>Traceback (most recent call last):
  File "test_spectrometer.py", line 8, in <module>
    spec = Spectrometer.from_first_available()
 File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 112, in from_first_available
    raise cls._backend.SeaBreezeError("No unopened device found.")
seabreeze.cseabreeze._wrapper.SeaBreezeError: No unopened device found.

The test code is: 

```
import numpy as np
from seabreeze.spectrometers import Spectrometer
import time
import os
from datetime import datetime

# Create a spectrometer object
spec = Spectrometer.from_first_available()
# Set integration time to 20 milliseconds
spec.integration_time_micros(20000)

last_ts_raw = datetime.now()

current_date = last_ts_raw.strftime("%Y-%m-%d-%H-%M-%S")
directory = os.path.join(os.getcwd(), current_date)
if not os.path.exists(directory):
    os.makedirs(directory)

while True:
    # Get wavelength and intensity data
    wavelengths = spec.wavelengths()
    intensities = spec.intensities()

    # Get the current time in milliseconds since the Epoch
    timestamp = int(time.time() * 1000)

    # Construct the filename with the current timestamp
    filename = f"spectrometer_data_{timestamp:013}.txt"

    # Open the file for writing
    with open(os.path.join(directory, filename), "w") as file:
        # Write data to file
        for wavelength, intensity in zip(wavelengths, intensities):
            file.write(f"{wavelength:.7f},{intensity:.7f}\n")

    # Wait for a short period of time before taking the next measurement
    time.sleep(0.01)
```

Let me know if I have to make anything different, I'm not sure if I was running the pull request or not, all I can say is that is not recognizing the device I think, and the spectrometer lights is only the red one turned on, when I'm using the spectrometer in windows, it blinks the green light besides the red one.

ap--, 2023-04-13:
In your test code at the very top before the imports you need to select a different backend for seabreeze:

```python
# add these two lines before the other seabreeze imports in your script
import seabreeze
seabreeze.use("pyseabreeze")
```

fieterovich, 2023-04-13:
Thanks, I tried with the `import seabreeze` and `seabreeze.use("pyseabreeze")`, and although is doing something, I'm still getting errors. This was the first output:

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ python test_spectrometer.py `
> Traceback (most recent call last):
  File "test_spectrometer.py", line 10, in <module>
    spec = Spectrometer.from_first_available()
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 108, in from_first_available
    for dev in list_devices():
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/spectrometers.py", line 56, in list_devices
    return api.list_devices()
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/api.py", line 119, in list_devices
    dev.open()
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/devices.py", line 373, in open
    self._transport.open_device(self._raw_device)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/transport.py", line 145, in open_device
    if pyusb_device.is_kernel_driver_active(0):
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/core.py", line 1107, in is_kernel_driver_active
    self._ctx.managed_open()
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

(I had a previous error, but it was that I didn't installed `pyusb` in the venv.)

When it said `access denied` I asked for help and somebody pointed me out that, because of the use of a `venv` I should not use `sudo` right away... Of course I tried anyway haha but it went outside the virtual environment and didn't find the libraries, so it was not a good solution. Doing a roundabout to the problem, a friend helped me to unlock the permissions of the usb port, by doing the following:

- `compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ lsusb`
> Bus 002 Device 002: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 0999:1002 Generic 4-Port USB 2.0 Hub
Bus 001 Device 005: ID 03f0:0d4a HP, Inc SK-2025 Keyboard
Bus 001 Device 004: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 003: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 002: ID 13d3:3549 IMC Networks Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Where `Bus 001 Device 007: ID 0999:1002 Generic 4-Port USB 2.0 Hub` is the spectrometer. We found after a while the usb port permissions in `/dev/bus/usb/` and listed the following:

- `compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ ll /dev/bus/usb/001/`
>total 0
drwxr-xr-x 2 root root    160 Apr 13 09:47 ./
drwxr-xr-x 4 root root     80 Dec 31  1969 ../
crw-rw-r-- 1 root root 189, 0 Mar  2 04:58 001
crw-rw-r-- 1 root root 189, 1 Mar  2 04:58 002
crw-rw-r-- 1 root root 189, 2 Mar  2 04:58 003
crw-rw-r-- 1 root root 189, 4 Mar  2 04:58 005
crw-rw-r-- 1 root root 189, 6 Apr 13 09:21 007
crw-rw-r-- 1 root root 189, 7 Apr 13 09:47 008

Then we ran `udevadm info --query=path --name=/dev/bus/usb/001/007` which had the output `/devices/platform/3610000.xhci/usb1/1-4/1-4.4`. Eventually, we ran:

-`compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ sudo chmod 777 /dev/bus/usb/001/007`
- `compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ ll /dev/bus/usb/001/`
>total 0
drwxr-xr-x 2 root root    160 Apr 13 09:47 ./
drwxr-xr-x 4 root root     80 Dec 31  1969 ../
crw-rw-r-- 1 root root 189, 0 Mar  2 04:58 001
crw-rw-r-- 1 root root 189, 1 Mar  2 04:58 002
crw-rw-r-- 1 root root 189, 2 Mar  2 04:58 003
crw-rw-r-- 1 root root 189, 4 Mar  2 04:58 005
crwxrwxrwx 1 root root 189, 6 Apr 13 09:55 007
crw-rw-r-- 1 root root 189, 7 Apr 13 09:47 008
[... 102 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

Let me know @ap-- if anything of this gives you an insight on how to overcome this issues. Thank you very much again for your time and patience with this topic.

Francisco

fieterovich, 2023-04-13:
Thank you for your time and patience to help me solve this issue, I try to do my best in documenting for you so appreciate the last comment :)

So the edit to the rules went well, nothing crashed over that.
About the code you sent me:

- In the first run, it gave the spectrometer serial number, but it did not succesfully ended. It crashed in the `print(dev.f.spectrometer.get_intensities())` line, as it follows:

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ python3 test2.py`
> SR400240
Traceback (most recent call last):
  File "test2.py", line 23, in <module>
    print(dev.f.spectrometer.get_intensities())
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/spectrometer.py", line 642, in get_intensities
    tmp = self._get_spectrum_raw()
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/spectrometer.py", line 638, in _get_spectrum_raw
    datastring = self.protocol.query(0x000_01C_00, timeout_ms=timeout)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 391, in query
    return self.receive(timeout_ms=timeout_ms)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 325, in receive
    response = self.transport.read(size=64, timeout_ms=timeout_ms)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/transport.py", line 200, in read
    ret: bytes = self._device.pyusb_device.read(
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 75] Overflow

- Then in the second attempt (after disconnecting and connecting the spec), I commented that line and crashed in the `get_wavelenghts()` one:

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ python3 test2.py`
>SR400240
Traceback (most recent call last):
  File "test2.py", line 26, in <module>
    print(dev.f.spectrometer.get_wavelengths())
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/spectrometer.py", line 666, in get_wavelengths
    data = self.protocol.query(0x000_011_00)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 391, in query
    return self.receive(timeout_ms=timeout_ms)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 325, in receive
    response = self.transport.read(size=64, timeout_ms=timeout_ms)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/transport.py", line 200, in read
    ret: bytes = self._device.pyusb_device.read(
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 75] Overflow

- Finally, I commented both wavelenghts and intensities lines, and the program ended succesfully:

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ python3 test2.py`
>SR400240
(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ 

At least is recognizing the serial number, which is something didn't happen before I think? Let me know your thoughts, thanks again!

ap--, 2023-04-14:
I made some small changes to the pull request. Please pull the changes and try the test code again.

fieterovich, 2023-04-14:
I tested it, and got the same output for the `get_wavelenghts` and `get_intensities` commands being commented one by one. 

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test$ python3 test2.py`
>SR400240
Traceback (most recent call last):
  File "test2.py", line 23, in <module>
    print(dev.f.spectrometer.get_intensities())
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/spectrometer.py", line 642, in get_intensities
    tmp = self._get_spectrum_raw()
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/features/spectrometer.py", line 638, in _get_spectrum_raw
    datastring = self.protocol.query(0x000_01C_00, timeout_ms=timeout)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 391, in query
    return self.receive(timeout_ms=timeout_ms)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/protocol.py", line 325, in receive
    response = self.transport.read(size=64, timeout_ms=timeout_ms)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/seabreeze/pyseabreeze/transport.py", line 200, in read
    ret: bytes = self._device.pyusb_device.read(
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/compas-orin-0/Desktop/SR4_test/myenv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 75] Overflow

These were the commands I used to pull the changes:

```
source myenv/bin/activate
cd python-seabreeze/
git fetch origin
```
>remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 8 (delta 6), reused 8 (delta 6), pack-reused 0
Unpacking objects: 100% (8/8), 727 bytes | 60.00 KiB/s, done.
From https://github.com/ap--/python-seabreeze
   9c12222..9665e6c  experimental-sr4-support -> origin/experimental-sr4-support

`git checkout pr-189`
>Already on 'pr-189'

Is it ok what I did?

ap--, 2023-04-14:
fetch alone doesn't update the local checkout.
you still need to 
```git merge --ff-only``` after the checkout command.

and if you haven't done an editable install, you need to pip install again afterwards.

fieterovich, 2023-04-14:
Ok, now this is working! 

Actually I did a clean install of the pull request because `git merge --ff-only` sent the error `fatal: No remote for the current branch.` and `git merge --ff-only origin/pr-189` sent the error `merge: origin/pr-189 - not something we can merge`.

I did a clean install on a new folder, and here are the results of the test code you sent:

`(myenv) compas-orin-0@compas-orin-0:~/Desktop/SR4_test2$ python3 test2.py`
>SR400240
[2262. 2293. 2315. ... 2492. 2525. 2555.]
[ 345.38217163  345.60322817  345.82427138 ... 1051.61272123 1051.77615338
 1051.93956725]

And my test code is working too! It is logging the spectrometer wavelenght and intensity. Thank you very much Andreas, for all the support and help. You're awesome! Let me know if you need anything else from me if you want to improve or test anything with this spectrometer, I would be very happy to collaborate with that. 

Cheers,
Francisco

closed 2023-04-15 by ap--

ap--, 2023-04-15:
Great :+1: 

A new release `v2.2.0` should be on pypi soon. You can then just `pip install seabreeze[pyseabreeze]` and your script should work.

I also pushed a script to the `dev` directory in this repo that let's you print the important information of the pcapng files in case you want to add more functionality.

Have a nice day!
Andreas

blozano824, 2023-04-28:
Hi! I am working with an SR2 and thought I would try to tweak the SR4 device class. I created a new class calls SR2 which is identical to the SR4 class in `devices.py` except with `model_name='SR2'` and `usb_product_id = 0x1001`. 

95% of the time, I receive the following error:
`usb.core.USBError: [Errno 13] Access denied (insufficient permissions)`. 
5% of the time, I am able to connect to the device and get the following information:
```
INFO : 2023-04-27 22:20:57,016 : __main__ : Info: {'model': 'SR2', 'pixels': 3648}
INFO : 2023-04-27 22:20:57,018 : __main__ : Wavelengths: [177.34085083 177.73592031 178.13094312 178.52591925 178.92084872 179.31573152 179.71056765 180.10535712 180.50009993 180.89479608]
INFO : 2023-04-27 22:20:57,018 : __main__ : Integration time: 12500
Traceback (most recent call last):
  File "/Users/brayan/Workspace/drivers/drivers/ocean/demo.py", line 92, in <module>
    main()
  File "/Users/brayan/Workspace/drivers/drivers/ocean/demo.py", line 87, in main
    loop.run_until_complete(run(flame))
  File "/usr/local/Cellar/python@3.7/3.7.16/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
    return future.result()
  File "/Users/brayan/Workspace/drivers/drivers/ocean/demo.py", line 21, in run
    spectrum = await flame.get_spectrum()
  File "/Users/brayan/Workspace/drivers/drivers/ocean/flame.py", line 165, in get_spectrum
    return await self._loop.run_in_executor(None, self._get_spectrum)
  File "/usr/local/Cellar/python@3.7/3.7.16/Frameworks/Python.framework/Versions/3.7/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/Users/brayan/Workspace/drivers/drivers/ocean/flame.py", line 100, in _get_spectrum
    return self._spec.intensities(correct_dark_counts=False, correct_nonlinearity=False)
  File "/Users/brayan/Workspace/drivers/venv/lib/python3.7/site-packages/seabreeze/spectrometers.py", line 212, in intensities
    out = self._dev.f.spectrometer.get_intensities()
  File "/Users/brayan/Workspace/drivers/venv/lib/python3.7/site-packages/seabreeze/pyseabreeze/features/spectrometer.py", line 691, in get_intensities
    struct.unpack("<" + "H" * self._spectrum_length, tmp[32:]),
struct.error: unpack requires a buffer of 7296 bytes
```

I am confused what could be the source of this behavior. The list of wavelengths is also very short and should range from 220 - 1050. I am working on a M2 Macbook using Python3.7. I am working on setting up Wireshark to capture the USB packets. Are there any other adjustments that need to be made to the `SeaBreezeDevice` subclass and the `SeaBreezeSpectrometerFeatureSR4` class?

ap--, 2023-04-28:
Hi @blozano824 

please open a new issue, and provide an example of the code that you're running and the modifications you made. Also please reproduce the issue with a plain example just using the python seabreeze module. It seems the traceback you've shown is using some wrapper code around seabreeze?

The more information you provide the easier it is for me to help.

Cheers,
Andreas

PR #189 (2023-04-15): Experimental sr4 support
completely untested...

might fix #185