ap--/python-seabreeze #200: Edge Trigger Issue Continues

waldowda, 2023-06-16:
### spectrometer and system information ###

- model: Flame-S / USB2000PLUS
- operating system: Win10 64bit
- python version: 3.8.10
- python-seabreeze version: 1.3.0 OR 2.4.0
- installed-via: conda or pip

### current problem ###

_detailed description of what doesn't work_

### steps to reproduce ###

We addressed (fixed?) the edge trigger issue back with closed issue #97 and again in #126. That seemed to function on and off in  1.3.0 as far as I can tell. I have new students now back on this project and now now setting trigger mode to '4' fails both in 1.3.0 and in the latest 2.4.0. 

I followed the normal steps to get Flame-S talking to py-seabreeze and then change trigger mode via:

```
spec.trigger_mode(4)
```

and it fails / errors but modes 0, 1, 2, and 3 all are able to be set. 

```
---------------------------------------------------------------------------
SeaBreezeError                            Traceback (most recent call last)
Cell In[6], line 1
----> 1 sec.trigger_mode(4)

File ~\Anaconda3\envs\SpecEchem2023\lib\site-packages\seabreeze\spectrometers.py:313, in Spectrometer.trigger_mode(self, mode)
    304 def trigger_mode(self, mode: int) -> None:
    305     """set the trigger mode of the device
    306 
    307     Parameters
   (...)
    311         value for the trigger mode you want to use.
    312     """
--> 313     self._dev.f.spectrometer.set_trigger_mode(mode)

File src\seabreeze\cseabreeze\c_seabreeze_wrapper.pyx:682, in seabreeze.cseabreeze._wrapper.SeaBreezeSpectrometerFeature.set_trigger_mode()

SeaBreezeError: Error: Invalid trigger mode
```

After the changes as referenced in #97 I used to be able set to trigger_mode to  '4' and it was triggered properly from my Camry potentiostat as far as I could tell. I checked devices.py and the changes made back in #97 were still in play. I did switch back from the Gamry driver to the Ocean Optics Driver as the Gamry driver seemed to get chosen by the OS.  I don't think I have changed anything I am aware of but mode 4 fails with the following error:

I am at a loss what else to try. Thanks, Dean

ap--, 2023-06-16:
Hi Dean,

It looks like you are using the wrong backend. Please try

```python
import seabreeze
seabreeze.use("pyseabreeze")
from seabreeze.spectrometers import Spectrometer

spec = Spectrometer.from_first_available()
spec.trigger_mode(4)

```

Cheers,
Andreas

waldowda, 2023-06-16:
Hi Andreas, As far as I know, I am using pyseabreeze not cseabreeze. I tried both earlier. I will try your suggested minimal code above in a few minutes. Thanks, Dean

waldowda, 2023-06-18:
Hi Andreas, I am not quite sure why but it is working with 1.3.0 now. I suspect it was an issue with my code I somehow fixed or it was the driver. I am not sure why but the Gamry Framework code still has a driver for ocean insight and that seems to always take over from the Ocean Optics driver. Also, I still have not got 2.4.0 working but I still have a few things to try. Anaconda doesn't seem to be releasing the spectrometer after using 1.3.0 despite restarting the kernel and stopping the 1.3.0 environment. I'll work on that and if I can't get the current version working I'll ask about that. I wonder if I will need to disconnect the usb cable and reconnect to release it. Anyway, I'll close this issue. Thanks! Dean

closed 2023-06-18 by waldowda