ap--/python-seabreeze #202: Flame-S in version 2.4.0 pyseabreeze and trigger mode

waldowda, 2023-06-19:
### spectrometer and system information ###

- model: Flame-S / USB2000PLUS
- operating system: Win10 64bit
- python version: 3.8.16
- python-seabreeze version: 2.4.0
- installed-via: pip

### current problem ###

Update on trigger issues on Flame-S moving to current release.

Flame-S (USB200PLUS) is recognized when using cseabreeze backend but not when using pyseabreeze backend. When cseabreeze is used, trigger mode 4 is not allowed. 

When using python 3.8.10, seabreeze version 1.3.0, and pyseabreeze, trigger mode 4 is allowed. I can continue using 1.3.0 although it might be best in the longer term to stay current with the latest stable release.

### steps to reproduce the issues###

I installed a new Anaconda env and setup seabreeze via pip. 

```
import seabreeze
seabreeze.use('cseabreeze')
#seabreeze.use('pyseabreeze')
print(seabreeze.__version__)
from seabreeze.spectrometers import list_devices, Spectrometer

devices = list_devices()

print ("Devices",devices)

spec = Spectrometer.from_first_available()
```

Output looks good but trigger mode 4 fails:

```python
2.4.0
Devices [<SeaBreezeDevice USB2000PLUS:FLMS16723>]
```
And...
```
spec.trigger_mode(4)
```
fails...
```
---------------------------------------------------------------------------
SeaBreezeError                            Traceback (most recent call last)
Cell In[3], line 1
----> 1 spec.trigger_mode(4)

File ~\Anaconda3\envs\SpecEchem2023\lib\site-packages\seabreeze\spectrometers.py:313, in Spectrometer.trigger_mode(self, mode)
    304 def trigger_mode(self, mode: int) -> None:
    305     """set the trigger mode of the device
    306 
    307     Parameters
   (...)
    311         value for the trigger mode you want to use.
    312     """
--> 313     self._dev.f.spectrometer.set_trigger_mode(mode)

File src\seabreeze\cseabreeze\c_seabreeze_wrapper.pyx:682, in seabreeze.cseabreeze._wrapper.SeaBreezeSpectrometerFeature.set_trigger_mode()

SeaBreezeError: Error: Invalid trigger mode
```

If I switch to pyseabreeze, I get the following results in the following error.

```
2.4.0
Devices []
---------------------------------------------------------------------------
SeaBreezeError                            Traceback (most recent call last)
Cell In[1], line 45
     41 devices = list_devices()
     43 print ("Devices",devices)
---> 45 spec = Spectrometer.from_first_available()

File ~\Anaconda3\envs\SpecEchem2023\lib\site-packages\seabreeze\spectrometers.py:112, in Spectrometer.from_first_available(cls)
    110         return cls(dev)
    111 else:
--> 112     raise cls._backend.SeaBreezeError("No unopened device found.")

SeaBreezeError: No unopened device found.
```

Again, I can continue to run using my 1.3.0 install though I am not sure what I functionally will be missing if anything as I think most changes related to other items.

ap--, 2023-07-16:
It's very strange that pyseabreeze doesn't recognize the spectrometer with `2.4.0` but does with `1.3.0`...

Could you post the commands used to create the `1.3.0` environment and the commands used to create the `2.4.0` environment?

waldowda, 2023-07-17:
Sure. It has been a while since trying that. I'll check and reply soon. BTW, I am using Anaconda for building python environments. Thanks