ap--/python-seabreeze #215: Install on raspberry pi

trautsned, 2023-10-22:
I am unable to install seabreeze on a raspberry pi. Using the lastest RPI OS (bookworm) gives the issue below. I have also tried RPI OS Bullseye but then get issues with numpy dependencies. Other options also give problems.

What's the easiest way to get seabreeze on an RPI? Should I continue with bookworm or try bullseye again? Are virtual environment ok? 32 or 64 bit?

Steps to reproduce:
install 32-bit bookworm on SD card and VNC to pi
create virtual environment and then try a seabreeze install
fails on missing libusb

Details
pi@applepi:~/Documents $ python3 -m venv env
pi@applepi:~/Documents $ source env/bin/activate
(env) pi@applepi:~/Documents $ pip -V
pip 23.0.1 from /home/pi/Documents/env/lib/python3.11/site-packages/pip (python 3.11)
(env) pi@applepi:~/Documents $ pip install seabreeze
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting seabreeze
  Downloading seabreeze-2.4.0.tar.gz (7.7 MB)
     7.7/7.7 MB 4.8 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  Ã Getting requirements to build wheel did not run successfully.
  â exit code: 1
  â°â> [24 lines of output]
      Traceback (most recent call last):
        File "/home/pi/Documents/env/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/pi/Documents/env/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/pi/Documents/env/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-thpjd559/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 355, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-thpjd559/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 325, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-thpjd559/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 507, in run_setup
          super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-thpjd559/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 341, in run_setup
          exec(code, locals())
        File "<string>", line 93, in <module>
        File "/tmp/pip-build-env-thpjd559/overlay/lib/python3.11/site-packages/pkgconfig/pkgconfig.py", line 248, in parse
          _raise_if_not_exists(package)
        File "/tmp/pip-build-env-thpjd559/overlay/lib/python3.11/site-packages/pkgconfig/pkgconfig.py", line 103, in _raise_if_not_exists
          raise PackageNotFoundError(package)
      pkgconfig.pkgconfig.PackageNotFoundError: libusb not found
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.

ap--, 2023-10-22:
It seems that piwheels is currently having problems catching up with building wheels for bookworm and aarch64. See notes at the top of this page: https://www.piwheels.org/

In the meantime:
If you look at the error in the traceback you posted, it says it's missing libusb. Installation of the development headers should make the pip install from the source distribution from pypi succeed.

```
sudo apt install libusb-dev
pip install seabreeze
```

Let me know if that solves the problem.

Once piwheels catches up with the build, binary wheels should become awailable and pip should not need to build the C-extension locally.

Cheers,
Andreas

trautsned, 2023-10-23:
libusb-dev installed without errors. However trying to install seabreaze gives an error (see below). I am not sure why it is trying to build a wheel since I thought they were built in pi wheels.

I do have seabreeze working in MS-Windows. It's awesome! I resurrected an old OceanOptics Jaz spectrophotometer and it gives results better than I expected. I wasn't able to get the OO software working in Windows 11 but your software works great. Fast and simple spectrophotometer.

I would be nice to have it working on a pi but I am happy enough using windows.

Here is the output from the pip install seabreeze. Let me know if there is other information that would help.

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting seabreeze
  Using cached seabreeze-2.4.0.tar.gz (7.7 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting numpy>=1.21
  Using cached https://www.piwheels.org/simple/numpy/numpy-1.26.1-cp311-cp311-linux_armv7l.whl (19.5 MB)
Building wheels for collected packages: seabreeze
  Building wheel for seabreeze (pyproject.toml): started
  Building wheel for seabreeze (pyproject.toml): finished with status 'error'
Failed to build seabreeze

There are lots of error lines. Here are the last few:
 line 249, in build_extension
          _build_ext.build_extension(self, ext)
        File "/tmp/pip-build-env-41bpxceu/overlay/lib/python3.11/site-packages/Cython/Distutils/build_ext.py", line 130, in build_extension
new_ext = cythonize(
             ^^^^^^^^^^
File "/tmp/pip-build-env-41bpxceu/overlay/lib/python3.11/site-packages/Cython/Build/Dependencies.py", line 1154, in cythonize
cythonize_one(*args)
File "/tmp/pip-build-env-41bpxceu/overlay/lib/python3.11/site-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: src/seabreeze/cseabreeze/c_seabreeze_wrapper.pyx

ap--, 2023-10-23:
So on piwheels the last python version for which wheels were compiled seems to be python3.9.

In addition: I need to make a new release, since cython seems to have broken the build process, too.

In the meantime you should be able to install directly from source via:

```
pip install git+https://github.com/ap--/python-seabreeze.git@main
```

I'll make a new release soon

trautsned, 2023-10-26:
Thanks. I get problems with installing numpy when I try to install seabreeze from source. I'll wait until numpy is up to date at piwheels. I am sure there are work arounds but the simplest solution is to wait a bit while everything catches up.

ap--, 2023-11-02:
Hi @trautsned,

the new release is available via pip and piwheels: https://www.piwheels.org/project/seabreeze/

closed 2023-11-02 by ap--