ap--/python-seabreeze #221: Drivers Dont have valid signatures

AlonsoUriarte14, 2023-11-02:
### spectrometer and system information ###

- model: _(, STS-VIS) _
- operating system: _(i.e. Windows 10 64bit)_
- python version: 3.10
- python-seabreeze version: latest (i downloaded the library today) 
- installed-via: _(pip install) 

### current problem ###

The computer is unable to recognize an unopened device. I ran te OS specific setup and found that only 31/53 of the drivers installed succesfully. 

### steps to reproduce ###

pip install seabreeze[pyseabreeze]

1. run code example

## Error
self.device = Spectrometer.from_first_available()
  File "C:\Users\dasus\AppData\Local\Programs\Python\Python310\lib\site-packages\seabreeze\spectrometers.py", line 112, in from_first_available
    raise cls._backend.SeaBreezeError("No unopened device found.")
seabreeze.pyseabreeze.exceptions.SeaBreezeError: No unopened device found.
```python

```

ap--, 2023-11-02:
Hello @AlonsoUriarte14 

The drivers won't be updated anymore. Those are the ones that shipped with seabreeze back then, and as far as I am aware Ocean Insight has not provided a newer set of drivers for public download.

But, that shouldn't keep this from working.

Can you confirm a few things:

- _please check the USB vendor ID and product ID:_
  Assuming your device is connected to the computer, go to "Device manager", find your device, 
  right click on it, select "Properties", go to "Details" tab, select "Hardware IDs" from the 
  drop-down, and you will find an entry of a form `HID\VID_046D&PID_C05A` (see: https://superuser.com/a/1106248)
- did you install `libusb`? see: https://github.com/libusb/libusb/wiki/Windows#user-content-how-to-use-libusb-on-windows

Cheers,
Andreas

AlonsoUriarte14, 2023-11-03:
[image]

in device mangager -> Ocean Optics STS (WinUSB) -> properties -> hardware ids-> i get two values: 
USB\VID_2457&PID_4000&REV_0051
USB\VID_2457&PID_4000

I installed libusb using Zadig just now but it didnt make a difference even after delting the drivers of device and reconeccting it.

ap--, 2023-11-03:
Okay, that looks good :)

can you post the output of:

```pycon
>>> import usb.core
>>> list(usb.core.find(find_all=True))
```

The pyseabreeze backend calls the above internally and tries to find a DEVICE with `2457:4000`. It will only be able to connect to your spectrometer if the above code returns a device that matches this.

AlonsoUriarte14, 2023-11-04:
[image]

It returns an empty list

ap--, 2023-11-04:
Can you try running your command prompt with administrator permissions and test the usb.core.find command again?

ap--, 2023-11-04:
And to be sure that your libusb installation is not the issue:

Please pip install the package `libusb-package` and run:

```python
import libusb_package

for dev in libusb_package.find(find_all=True):
    print(dev)
```

AlonsoUriarte14, 2023-11-06:
DEVICE ID 1022:1639 on Bus 003 Address 000 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x300 USB 3.0
 bDeviceClass           :    0x9 Hub
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x3
 bMaxPacketSize0        :    0x9 (9 bytes)
 idVendor               : 0x1022
 idProduct              : 0x1639
 bcdDevice              :  0x100 Device 1.0
 iManufacturer          :    0x0
 iProduct               :    0x0
 iSerialNumber          :    0x0
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 0 mA ====================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x1f (31 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0xc0 Self Powered
   bMaxPower            :    0x0 (0 mA)
    INTERFACE 0: Hub =======================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x9 Hub
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x1 (1 bytes)
       bInterval        :    0xc
DEVICE ID 1022:43ee on Bus 001 Address 000 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x300 USB 3.0
 bDeviceClass           :    0x9 Hub
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x3
 bMaxPacketSize0        :    0x9 (9 bytes)
 idVendor               : 0x1022
 idProduct              : 0x43ee
 bcdDevice              :  0x100 Device 1.0
 iManufacturer          :    0x0
 iProduct               :    0x0
 iSerialNumber          :    0x0
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 0 mA ====================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x1f (31 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0xc0 Self Powered
   bMaxPower            :    0x0 (0 mA)
    INTERFACE 0: Hub =======================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
[... 624 lines cut ...]
       bInterval        :    0x1
    INTERFACE 1, 5: Wireless Controller ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x5
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xe0 Wireless Controller
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x1
     iInterface         :    0x0
      ENDPOINT 0x3: Isochronous OUT ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :   0x31 (49 bytes)
       bInterval        :    0x1
      ENDPOINT 0x83: Isochronous IN ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :   0x31 (49 bytes)
       bInterval        :    0x1

ap--, 2023-11-06:
The fact that the `libusb-package` test work, indicates that you did not correctly install libusb.
Have you tried installing the correct dll manually in you windows installation?