ap--/python-seabreeze #224: Support for SR6 / HR6

tberezowski, 2023-11-27:
### spectrometer and system information ###

- model: SR6, HR6

### current problem ###

I noticed that SR2 and SR4 support were recently added. I would like to add support for SR6 and HR6 myself. Unfortunately, I cannot find a datasheet. These models are also not listed in the original [SeaBreeze](https://www.oceaninsight.com/globalassets/catalog-blocks-and-images/software-downloads-installers/javadocs-api/seabreeze/html/files.html). Does anyone have these datasheets or any information that would help me to add support for these devices?

ap--, 2023-11-28:
Hi @tberezowski 

Thank you for offering to contribute ❤️ 

Sadly there seem to be no developer datasheets publicly available anymore for the newer spectrometer models. But, if the SR6/HR6 use the same protocol as the ST, SR2, and SR4 it's possible that adding basic support is not too hard. 

I recommend reading this issue: #185
And having a look at these PRs:
- #220
- #192
- #189

If you create a packet capture file `.pcapng` you can use this script: https://github.com/ap--/python-seabreeze/blob/main/dev/pretty_print_obp_pcapng.py to pretty print some of the USB communication for debugging.

Let me know if you need further information,
Andreas 😃

knaugh, 2024-01-03:
I am also trying to get SR6 support working. I was able to add two classes and udev rule as described in #220 which allowed me to read the spectrometer. However I just copied the SR4 classes and changed their names, and didn't change any of the device parameters.
I do not have an OceanView license unfortunately, is there another way I can pull the info you need?

tberezowski, 2024-01-04:
@knaugh what I remember OceanView had some kind of a demo mode, but I am not sure if it is available in the recent version. Currently I havent start to work on this issue because my university is still in the process of buying the spectrometers... I should have them within one to two months and then I will post some update.

ap--, 2024-01-26:
SR6 support available in v2.6.0

Thanks again everyone ❤️

closed 2024-01-26 by ap--