ap--/python-seabreeze #226: USB2000 seems to no longer be usable due to fpga.py firmware version register check

dbhatman, 2023-12-08:
### spectrometer and system information ###

- model: USB2000
- operating system: Ubuntu 18
- python version: 3.11
- python-seabreeze version: 'current master'
- installed-via: conda

### current problem ###

Using a very old version of pyseabreeze I had successfully used USB2000 spectrometers, but now I am hitting an AssertionError in fpga.py when using `spectrometers.Spectrometer.from_first_available()`.

The error reported is `AssertionError: ['0x0', '0x46', '0x30']` being hit here https://github.com/ap--/python-seabreeze/blob/f40f81c0b3ebf5063b6ecdcf59858637a92bb9ec/src/seabreeze/pyseabreeze/features/fpga.py#L30 Maybe the USB2000 also does not adhere to the expected protocol as was noted here: https://github.com/ap--/python-seabreeze/issues/97#issuecomment-674567912

A USB2000+ (note I am using a non-plus but could not find a datasheet for it) datasheet seems to suggest the same "Return format" as the assertion expects (starting with the register's address)  but instead we are getting 0! Datasheet I could find https://ph208.edu.physics.uoc.gr/pdfs/OEM-Data-Sheet-USB2000-.pdf

### steps to reproduce ###

USB2000 plugged in and then

`spectrometers.Spectrometer.from_first_available()`

dbhatman, 2023-12-08:
Interestingly I am also seeing this from one of my FlameS sectrometers. Others are operating fine (ie are fixed by  https://github.com/ap--/python-seabreeze/issues/97#issuecomment-674567912)

ap--, 2023-12-10:
Hi @dbhatman 

Thanks for the report!

I don't think your USB2000 is a nonplus version. That code path is just run for the USB2000+ and FlameS if I'm not mistaken.

Can you please report the vendor and product id of the USB spectrometer?

Regarding the FlameS:
Could you try running your test code with the optimize flag so that assert statements are skipped?

```bash
python -O myscript.py
```

It seems your flame S is on major firmware version 3. It's possible that they changed the return code in those revisions.

And could you try to figure out what version of seabreeze broke the USB2000-nonplus for you?
If you could just pip install older versions to see when it broke, that would be super helpful.

Cheers,
Andreas

jlball, 2024-01-04:
Hi all,

Been running into the same issue with a USB2000+ (the Assertion error) and found that the most recent version where I didn't get this error is version 1.2.0.

ap--, 2024-01-06:
Hi @jlball,

Thank you for checking the versions and reporting back. Could you provide the error message when the assertion error is raised using the newest seabreeze version, and could you try running your test code with `python -O yourtestscript.py` to disable assertions.

If you can confirm that the code runs with assertions disabled, I can release a new version without the check.

Thanks,
Andreas