ap--/python-seabreeze #240: Ethernet Support for HDX

lovelesh-mis, 2024-04-08:
### spectrometer and system information ###

- model: OceanHDX
- operating system: Windows/ Linux 64-bit
- python version: 3.10
- python-seabreeze version: current master
- installed-via: python setup.py install

I am facing the same issue of USB timeout as in #133. I have tested with Flame-T and everything works fine but I get a timeout error with HDX. I am wondering if there is a mechanism to connect to HDX using Ethernet. pyseabreeze has `add_ipv4_device_location`  function but is not implemented.

Can you provide how to set started with this implementation? I feel we can bypass the pyusb drivers issue in Linux by implementing ethernet based drivers.

hperrey, 2024-05-29:
I am facing a similar issue and we are looking into implementing communication via Ethernet for an OceanHDX spectrometer. I am currently looking into how much effort it would be to do this through the `pyseabreeze` backend (versus a standalone solution) -- pointers into the right direction would be very welcome!

@lovelesh-mis, have you started with the implementation yet?

lovelesh-mis, 2024-05-29:
I was able to get it working to some using Ethernet with static IP. I used @Dargaud's [C Library for OceanHDX](https://gitlab.com/dargaud/C_OceanHDX) for reference. I designed my commands exactly as per his library. Socket library in windows and linux is platform agnostic and very reliable. Also, there is no driver issues as compared to USB. Wth my current setup, I can use Oceanview and my python code simultaneously.

I compared pyseabreeze commands to the C library mentioned above and they are exactly the same. So, I think one need to develop a mechanism to send the commands through an IP socket. I didnt have much idea about the pyseabreeze backend, so I went with standalone approach.
 Also, OveanView implements a basic mDNS protocol to discover new devices over the network. You can just run wireshark and grab those packets. All ocean devices are programmed to send their IDs as multicast packets.

hperrey, 2024-05-30:
Oh, thanks for the hint with Dargaud's library! That looks promising :)
So if I understand correctly, you created a Python wrapper around that library mimicking the seabreeze API?

Do you know whether the library supports buffered readout? That is a feature we would particularly like to use in the future, once all basic functionality is there.

I have started to implement IPv4 socket communication into `pyseabreeze` and will give that a shot today. It will only support static IPs for now and I hard-coded most in the beginning, but the `pyseabreeze` design would allow for an implementation of multicast to search for devices in the future. If my tests work, I will try to upstream this. Otherwise, I might go the same route as you did.

ap--, 2024-05-30:
Hello everyone,

I wasn't aware of that library, we should link to it in the README for reference. If someone could test if the USB implementation in dargauds library is working reliably on windows, we might be able to find the issue with pyusb that causes issues for seabreeze.

If there's interest in developing a ethernet backend for pyseabreeze, I can help with some guidance on how it would fit into the current module design. https://github.com/ap--/python-seabreeze/blob/f4365c66d7ff0296145955f4c5f0b19ea93ee583/src/seabreeze/pyseabreeze/types.py#L51-L131 a new Transport subclass would need to be implemented to handle the ethernet communication. (We should discuss in a new issue)
Also, that being said, it's probably easier to start with a minimal working example, and then take that and squeeze it into the correct structure for integration into python-seabreeze.

Cheers,
Andreas

hperrey, 2024-05-30:
Hej Andreas!

Thanks for the pointer -- I have made proof-of-concept implementation and would really appreciate your comments on that. It is far from done, but I would like to get your feedback on whether that approach is compatible with the structure of python-seabreeze.

Thanks! :)

ap--, 2024-06-07:
Closed via #243 thanks to amazing work by @hperrey ❤️ 

Will be available in `2.9.0`

closed 2024-06-07 by ap--

amoeez, 2024-10-15:
Thank you for adding the support but I haven't been able to connect to spectrometer using the code snippet given in readme. I also used the proof of concept by @hperrey in #234
When I run `spec = Spectrometer.from_serial_number("HDX12345")` it returns:

`OSError: [Errno 99] Cannot assign requested address`

The spectrometer works fine using the GUI via the ethernet. I have assigned static ip address.

- OS: Ubuntu 20.04
- Python: 3.10 (also tried with 3.12
- seabreeze: 2.9.2
- installed via: pip

Tb8854, 2025-11-14:
Just FYI for anyone attempting this. 

First get a newish version of the OceanView software. 
1. Plug in HDX via USB (C in my case). 
2. Go to device manager in OceanView, select automatically connect to devices and automatically connect to remote devices. (this set is in the manual) 
3. Find the device in schematic mode. Right click and click network setup. If it is blank,  it means you have a too old version of the OceanView software. 
4. In network setup enable everything, go through each subtab and click enable. Spam that apply settings/save settings botton. 
5. I used a usb to ethernet adapter. So, now, de-enable DHCP. I added a IPv4 address of 192.168.1.200 and a gateway of 10.18.2.1. Only now connect ethernet cable.  You can follow the manual here,. But here you have to cycle the device, turn off ocean view. Wait a few minutes. Plug in. 
6. In windows, go to settings -> Network connectivity -> Ethernet adaptor -> IP Settings -> IP assignment Manual, IPv4 address 192.168.1.101, subnet prefix - 24 and IPv4 gateway of 10.18.2.1 
8. Open OceanView, and disconnect from USB (the ethernet option should be there). Close OceanView. 
9. seabreeze.use("pyseabreeze",  network_adapter="192.168.1.101")

All working.