ap--/python-seabreeze #241: OceanHR4 connection issue

s910324, 2024-04-18:
### spectrometer and system information ###
- model : OceanHR4
- OS : Windows 11 64 Bit
- python version: 3.10.7
- python-seabreeze version: 2.7.0
- installed-via: `python pip install seabreeze` and `seabreeze_os_setup`

### current problem ###
1. Ocean HR4 cannnot be found using both cseabreeze and pyseabreeze backend
2. Both `WINUSB (v10.0.22621.2506)` and `WINUSB (v6.1.7600.16385)` driver were tested, but does no difference
3. Tried on two window machine, both have trobule finding HR4, but can recognize Ocean USB2000+ without any issue
4. Ocean HR4 can be found using OceanView 2.0.13 and works properly.
5. The HR4 system property window as below, it does not show `Manufacturer` correctly, I assume this might be the reason that it failed to show on the list
[image]

### steps to reproduce ###
`
import seabreeze
from seabreeze.spectrometers import list_devices, Spectrometer
print( list_devices())
`
result shows `[]` if only HR4 is connected

like to have some help, thank you.

ap--, 2024-05-29:
The HR4 is currently not supported. Contributions for adding support are welcome!

closed 2024-05-29 by ap--

MicheleCotrufo, 2024-06-06:
@s910324 We added support for HR4 in the last version (released few days ago) - any feedback would be great!