ap--/python-seabreeze #248: `pyseabreeze`: Lazy loading of `SeaBreezeError` results in warning

hperrey, 2024-06-18:
### spectrometer and system information ###

- model: HDX 
- operating system: CentOS Linux release 7.9.2009
- python version: Python 3.9.16 or 3.11.9
- python-seabreeze version: 2.9.0
- installed-via: conda

### current problem ###

When issuing `seabreeze.use()` after importing `SeaBreezeError`, the following warning will be shown:

```
 UserWarning: seabreeze.use has to be called before instantiating a spectrometer
```

### steps to reproduce ###

The following code will trigger the warning:

```python
import seabreeze
from seabreeze.spectrometers import SeaBreezeError
seabreeze.use("pyseabreeze", network_adapter="192.168.254.200")
```

On the other hand, just importing the `Spectrometer` works fine:
```python
import seabreeze
from seabreeze.spectrometers import Spectrometer
seabreeze.use("pyseabreeze", network_adapter="192.168.254.200")
```

I have not investigated the cause further, but from cross-reading the code, I expected that `SeaBreezeError` would be lazy-loaded as well?

ap--, 2024-06-18:
Hi @hperrey 

Thanks for reporting! I didn't realize that my previous implementation triggered the backend load when importing the SeaBreezeException.

I made a new PR with a fix for allowing SeaBreezeError to be loaded after deciding which backend you want with `seabreeze.use`.

Will provide a new release tomorrow.

Cheers,
Andreas 😃

ap--, 2024-06-19:
Okay, I over-complicated things. New, actually working, version on the way...

ap--, 2024-06-19:
Fix with a simpler solution.

v2.9.1 is broken and will be yanked.
v2.9.2 should be out soon

closed 2024-06-19 by ap--

hperrey, 2024-06-20:
Much appreciated, thank you for fixing this so quickly! :smile_cat:

PR #249 (2024-06-18): seabreeze.spectrometers: make SeaBreezeError import lazy
Closes #248 

A little dark magic to fix the lazy import warning on SeaBreezeError
PR #250 (2024-06-19): fix via common base class
Close #248