ap--/python-seabreeze #255: Problem finding OceanST in Python

Darvilium, 2024-07-24:
### spectrometer and system information ###

- model: OceanST
- operating system: Windows 10 64bit
- python version: Python 3.11.5
- python-seabreeze 2.9.2
- installed-via: conda -> conda install -c conda-forge seabreeze -> seabreeze_os_setup -> pip install seabreeze[pyseabreeze] -> added in windows/system32 the libusb-1.0.dll file

### current problem ###

After installation, I can find and use the spectrometer using OceanView, but not with python. I read, the spectrometer needs the libusbK driver to be seen in python, so I replaced the WinUSB driver with libusbK (v3.1.0.0) using Zadig. Now, when I list my USB devices in python I can see the spectrometer, but when I try to list and read out the spectrometer using the program below, I don't find it. Have I missed anything? Or is the OceanST simply not yet supported in seabreeze?

Program:

import seabreeze.spectrometers as sb

def check_spectrometers():
    devices = sb.list_devices()
    if not devices:
        print("No Ocean Optics spectrometers found")
    else:
        for device in devices:
            print(f"Spectrometer found: {device}")

if __name__ == "__main__":
    check_spectrometers()

Darvilium, 2024-07-24:
I forgot, it is a OceanST NIR spectrometer.

ap--, 2024-07-24:
Hi @Darvilium 

The ST is only supported via the pyseabreeze backend. Currently by default seabreeze uses the cseabreeze backend. I modified your code example below to use the pyseabreeze backend.

```python
import seabreeze
import seabreeze.spectrometers as sb

seabreeze.use("pyseabreeze")

def check_spectrometers():
    devices = sb.list_devices()
    if not devices:
        print("No Ocean Optics spectrometers found")
    else:
        for device in devices:
            print(f"Spectrometer found: {device}")

if __name__ == "__main__":
    check_spectrometers()
```

Let me know if that helps,
Andreas 😃

Darvilium, 2024-07-25:
Hey Andreas,

ah cool. Easy fix :D 

Yes, this was the problem. I added your line, and my programs are working.

Btw. I had contact via email with probably one of your collegues, his name is Norm, and he sent me a LabView Vi which uses OceanDirect. On LabView I got it then also working with OceanDirect. 
One of my collegues here in the company also previously wrote some code using OceanDirect and python, but it is not working for my machine. It claims, my API Version is 1.33.0. He left the company some time ago, so I cannot ask him. I append his code also to this post. Maybe you can also help me with this issue, please?
[Spec-NIR.txt](https://github.com/user-attachments/files/16371148/Spec-NIR.txt)

Thank you, 
Manuel

ap--, 2024-07-25:
Hi Manuel,

Happy to hear that it works now ❤️ 

> I had contact via email with probably one of your collegues, his name is Norm

I am not an OceanInsight employee, and never have been. This project started as a Python interface for an OceanOptics USB2000+ spectrometer I used during my PhD and it grew into a hobby project of mine. `python-seabreeze` is not officially supported by OceanInsight. I am not payed by OceanInsight.

That being said, I was in contact with incredibly nice OceanOptics employees that were very supportive of open-source.

> One of my collegues here in the company also previously wrote some code using OceanDirect and python, but it is not working for my machine. It claims, my API Version is 1.33.0. He left the company [...]

*Short answer:* I can't.

*Long answer:* I don't know if this is still the case, but when the OceanDirect Python SDK was introduced it required users to pay a significant yearly licensing fee, which I **strongly** disagree with. If I buy a spectrometer I should be able to have the means to read out data from that spectrometer without a subscription for a basic software interface. So you'll have to go through their support to get help with that.

Have a great day,
Andreas 😃

ap--, 2024-08-06:
Closing. Feel free to reopen or make a new issue in case there are additional issues.

closed 2024-08-06 by ap--