ap--/python-seabreeze #262: for some reason I can't get the intensity

darkc0rn, 2024-09-11:
### spectrometer and system information ###

- model: QE Pro
- operating system: Win 10, Win 11
- python version: 3.10.1
- python-seabreeze version: 2.7.0
- installed-via: pip

### current problem ###

I have a project on pyseabreeze 2.7.0. I would like to ask right away why this version is not in the releases, is there some kind of critical error?
So this project works fine from .py files, in 2 different laboratories, but when I try to export it to .exe, I get a list of spectrometers, open the connection, get the wavelength, but I can't get the intensity. What could this be related to?
If the wavelengths are not obtained via usb, but are taken from the list, then perhaps the problem is in it. That is, if I get the serial number of the spectrometer, maybe the library itself understands from this serial number what wavelength it can return. This is the only explanation why I can get the wavelength, but I can't get the intensity.
I saw one person had a problem with the version libusb.dll but if I had a similar problem, would I be able to get the intensity from the original .py file?
Translated by a translator, sorry in advance

ap--, 2024-09-11:
If the spectrometer returns wavelengths it means that the USB communication is working.

I've seen weird issues happen if the spectrometer is connected through a USB hub. Try different USB ports on your computer and report back.

Cheers, 
Andreas 😊

darkc0rn, 2024-09-11:
The program from the exe does not work in 2 laboratories, at the same time.py is working fine, but I'll try to change the port

darkc0rn, 2024-09-11:
The problem has been solved. As I understand from the source files, the library supports several methods of obtaining information. For some reason at startup.The py program could receive data without the pyusb library. And to get data from .exe, pyusb is needed. If you can explain why this is happening, I will be very grateful.

ap--, 2024-09-12:
but when I try to export it to .exe

What are you using to bundle the code and create the executable?

darkc0rn, 2024-09-12:
auto py to exe it generates a command for PyInstaller if I remember correctly
pyinstaller --noconfirm --onefile --windowed --name "MULTISPECTRO" --hidden-import "importlib.resources" "C:\Users\ilea\Desktop\milti spectro\Spectro-Thermometer\main.py"
the command will look like this if you remove third-party imports

ap--, 2024-09-13:
Hi @darkc0rn 

I think you have to report this in the pyinstaller repository. This doesn't seem to be an issue with python-seabreeze but rather with the way pyinstaller packages it.

Cheers,
Andreas 😊

ap--, 2024-10-04:
> The py program could receive data without the pyusb library. And to get data from .exe, pyusb is needed. If you can explain why this is happening, I will be very grateful.

My guess is that `seabreeze.cseabreeze._wrapper.*.pyd` is not packaged by pyinstaller, and then seabreeze falls back to the pyseabreeze backend, requiring pyusb.

closed 2024-10-04 by ap--