ap--/python-seabreeze #269: Ocean HR-6XR500-25 can't be connected

Hollen0318, 2024-10-28:
### spectrometer and system information ###

- model: Ocean HR-6XR500-25, Spectrometer 300 Lines Blazed at 500 nm; SLIT 25
- operating system: Apple M3 14.2 (23C64) MacOS Sonoma 14.2
- python version: Python 3.10.15
- python-seabreeze version: current main
- installed-via: pip install seabreeze

### current problem ###

The OceanView can detect and operate the device; Apple System Information can find the device; However, the code can't find the device either through first_available() or sb.Spectrometer.from_serial_number("HR600278") or list_devices() returns empty list.

### steps to reproduce ###

_Step by step instructions to reproduce the error. The more detailed the better, but please use some common sense:_

1. Plug the Mac computer to the Ocean HR
2. install miniconda for MacOS
3. Create a new environment named ocean with python version being 3.10 like conda create -n ocean python=3.10
4. Run pip install seabreeze
5. Open a notebook, or create a new python script
6. Write code listed below
import seabreeze.spectrometers as sb
spec = sb.Spectrometer.from_serial_number("HR600278")
7. Bug Shown Below
8. Open the Systems Information and the OceanView, take a screenshot

### minimal code example and error (very helpful if available) ###

```python
import seabreeze.spectrometers as sb
spec = sb.Spectrometer.from_serial_number("HR600278")
```
---------------------------------------------------------------------------
SeaBreezeError                            Traceback (most recent call last)
Cell In[14], [line 2](vscode-notebook-cell:?execution_count=14&line=2)
      [1](vscode-notebook-cell:?execution_count=14&line=1) import seabreeze.spectrometers as sb
----> [2](vscode-notebook-cell:?execution_count=14&line=2) spec = sb.Spectrometer.from_serial_number("HR600278")

File ~/miniconda3/envs/ocean/lib/python3.10/site-packages/seabreeze/spectrometers.py:181, in Spectrometer.from_serial_number(cls, serial)
    [179](https://file+.vscode-resource.vscode-cdn.net/Users/wrl/Downloads/OceanOptics%20-%20Spectrometer/Seabreeze/~/miniconda3/envs/ocean/lib/python3.10/site-packages/seabreeze/spectrometers.py:179)             return cls(dev)
    [180](https://file+.vscode-resource.vscode-cdn.net/Users/wrl/Downloads/OceanOptics%20-%20Spectrometer/Seabreeze/~/miniconda3/envs/ocean/lib/python3.10/site-packages/seabreeze/spectrometers.py:180) else:
--> [181](https://file+.vscode-resource.vscode-cdn.net/Users/wrl/Downloads/OceanOptics%20-%20Spectrometer/Seabreeze/~/miniconda3/envs/ocean/lib/python3.10/site-packages/seabreeze/spectrometers.py:181)     raise cls._backend.SeaBreezeError(
    [182](https://file+.vscode-resource.vscode-cdn.net/Users/wrl/Downloads/OceanOptics%20-%20Spectrometer/Seabreeze/~/miniconda3/envs/ocean/lib/python3.10/site-packages/seabreeze/spectrometers.py:182)         "No device attached with serial number '%s'." % serial
    [183](https://file+.vscode-resource.vscode-cdn.net/Users/wrl/Downloads/OceanOptics%20-%20Spectrometer/Seabreeze/~/miniconda3/envs/ocean/lib/python3.10/site-packages/seabreeze/spectrometers.py:183)     )
SeaBreezeError: No device attached with serial number 'HR600278'.

[image]
[image]
[image]

Hollen0318, 2024-10-28:
The ReadMe states that Pyseabreeze is not supported for the HR6 and recommends using Ethernet or CSeabreeze. Could you explain what CSeabreeze is? Also, could the issue be related to the USB port ID? I'm not entirely sure, but my USB Product ID is 0x1006, as shown in the system information screenshot.

ap--, 2024-10-29:
Supported in `v2.10.0` via:

```python
# pip install "seabreeze[pyseabreeze]>=2.10.0"

import seabreeze
seabreeze.use('pyseabreeze')

from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
print(spec.intensities())
```

closed 2024-10-29 by ap--